MarlinFirmware/MarlinDev #1: Do we need to watch 'Issues' in both places?

Roxy-3DPrintBoard, 2015-08-02:
Question:   Do we need to watch 'Issues' in both places?

closed 2015-08-02 by Roxy-3DPrintBoard

Wackerbarth, 2015-08-02:
Correct.  It will take a while to get things "sorted". The intention is that issues on "Marlin" will apply ONLY to Released (or tagged Testing pre-release) versions and NOT future improvements, etc. Those will be over here on the "MarlinDev"

thinkyhead, 2015-08-03:
I automatically see issues for both in my notifications. I expect that for many months to come issues will be getting posted to Marlin that we will want to move over to MarlinDev, and vice-versa. I suspect this may tend to obligate us to do more support and bug-fixing in past releases than we would like. On the other hand, it will be easier to point users to the latest release and mark older releases "obsolete." Being patient… We will see in time how well this new model works.

boelle, 2015-08-03:
well we can also do it the simple way and ask people to repost their issue at the right spot and close it 

and the new arrangement does not obligate us to more than we want to... 

what will be most interesting is to see if dev work picks up speed

thinkyhead, 2015-08-03:
I suppose we can close all open issues at MarlinFirmware/Marlin that refer to the old Development branch… or should we leave them open in both repos until they're resolved?

boelle, 2015-08-03:
hmm... if they are not resolved then move them here and close them.... my logic is that they then are in the right place

only reason why i have not moved more than i have is to have the issue list small... we can always move the rest when we are dry of issues and bored :-P

Wackerbarth, 2015-08-03:
@boelle -- Well, when I got up this morning and found 300+ messages -- Yikes!
Unfortunately, I don't feel that many of them remain relevant. I refer to the ones discussing various problems with bed leveling from a couple months ago.  That code has changed drastically and, assuming we think it is now working, we should just ask those users to re-test.

boelle, 2015-08-03:
if you where just paid 0.01$ for each of those :-D

but yes we can also just close them and ask people to repost here if they still have the issue with the latest dev snapshot

Wackerbarth, 2015-08-03:
Actually, I was hoping that we would soon get to a Release Candidate and get those "Testers" over on Marlin, leaving MarlinDev for things like experimental features and 1.2.x, etc.

Sniffle, 2015-08-03:
Im in the process of testing measured height change vs offset height to
make sure they are very close if not exactly the same on a makerfarm prusa
variant with a rambo board.  Ill report back on it hopefully by the end of
the day.

boelle, 2015-08-03:
well i'm up for that... ie testing...

that said it will not be this week.... tina has gone to bed as she needs to be up at 2am.. she is going on hols with her mom for the next week and they need to be at airport before the devil had his morning coffee

while she is away i need to replace the preheater in the car with a new one... and then complete detail the car inside and out... and just to push my luck i have to print 2 spice jars each day also

but after that i'm pretty much up for testing code

thinkyhead, 2015-08-03:
> leaving MarlinDev for things like experimental features

Well, we still have to patch up Marlin prior to release. Should we not be using MarlinDev for that?

Wackerbarth, 2015-08-03:
@thinkyhead -- Scott, I draw the line at "Release Candidate". The problem that we have today is that Bo seems to be treating MarlinDev as if we were already past that point.  That is certainly getting in the way of getting to the RC stage.

thinkyhead, 2015-08-04:
Well, I'm all for getting a 1.1.0-RC1 soon. If we do it really soon we'll certainly get loads of feedback on leveling problems. But, I'd like to make sure we don't destroy a bunch of glass printbeds in the process…