MarlinFirmware/MarlinDev #2: Self awareness/conversation about goals and what do we believe is currently ready to move for a release tag?

Sniffle, 2015-08-02:
So, the difference between the current dev firmware and the current release, unless things have changed, are massive.  What are some features/changes that are currently implemented that while maybe not 100% complete are stable enough to move to the integration or release branch?

Im thinking the current bed leveling/compensation needs testing to confirm correct offset vs measured offset.  While it does work and M851 is usable to change the offset to work so that  correct layer height can be established(ive verified this but havent properly measured my offset in quite some time) and it is consistent, i dont know that the offset matches closely with the measured offset currently.

Display changes, to my knowledge they arent completed currently but are stable.  I am currently running the development on my printer without issues to the RRDFGD.

Can we get a few others to confirm features/stable updates/bug fixes that are currently stable that can be rolled into ths integration branch

And of course the ultimate goal of marlin is a self aware non network enabled version of skynet :-P thus the self aware in the title and just an inventory check on the devel branch because there are some things that probably should be rolled into integration for wider testing to reduce the huge gap between current release and devel.

thinkyhead, 2015-08-03:
@Sniffle If you try the latest MarlinDev code, perhaps you can confirm whether leveling is fixed or not for your configuration. (With `M851` set to the real z probe offset, of course.)

Sniffle, 2015-08-03:
Are we going to put a pause on merges to abl stuff so that we have a
dedicated snapshot to test so we can get a good readout on what if anything
is wrong? I know there was a merge a few min ago... I have the current
development version before this new repo was created that ive been running
just still have a few things to test.

Sniffle, 2015-08-03:
@thinkyhead for some reason sanity check is preventing me from compiling even with an unedited configuration.h due to automatic versioning conflicts... gonna see if i can work out where this is occuring and fix it...

Wackerbarth, 2015-08-03:
@Sniffle -- Can you meet me in the Hangout https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia
I'm the one behind the auto versioning and Scott just made some changes to SanityChecks.

It shouldn't be too hard to sort it out.

Sniffle, 2015-08-04:
@thinkyhead OK, had to do a little work around to compile with windows due to a bug specific to windows with the automatic version generation.  

Just tested the cartesian implementation of ABL, it's off vs measured distance and is inconsistent in when trying to correct it via M851.

per my usual testing routine of:
G28
G29
G1 X150 Y150 Z0 F4000

Measured Z offset was -6.8 using a .381mm feeler gauge, after settign up offset and running my usual test routine, Z was below the bed exactly 1mm again raising the bed in .1 increments until the feeler gauge passed under the nozzle.  I adjusted via M851 to Z-5.8  after running the routine I was exactly 1mm above the bed.  So in an attempt to make it work, I split the difference, this time going for an offset of Z-6.3  after the routine, I was exactly .5mm below the bed.  Something isn't adding up properly, as of right now, I still want to test a little bit more but so far my tests are showing we are really close but slightly off the mark.

Sniffle, 2015-08-04:
after further testing at Z-6.8 it is consistently doing the same thing hitting the bed and sliding as it tries to go below the bed(my Z axis rests on nuts so it doesnt hard crash, I'm going to try adjusting the Z height to consistently be the correct height over the bed

eboston, 2015-08-04:
I have been seeing these references to M851 but looking at the RepRapWiki list of codes, that one is not listed.  I know that the list is probably not current and doesn't list it.  But my question is how is M851 different from the M212 command?  I have a printrbot and the M212 command is used for setting the Z height.  Sounds a bit like what the M851 command does.  Are they the same?  Do they basically perform the same function, but in different ways?

Sniffle, 2015-08-04:
M851 is a custom Mcode in marlin that specifically sets the Z probe offset or changes the Z probe offset without reflashing firmware

@thinkyhead I did more testing, I got it to consistently hit the proper height above the bed with Z-5.8 I'll receck the measured distance to make sure i didnt screw up measuring

Sniffle, 2015-08-04:
@thinkyhead ok, after being a little less rushed this morning, I can confirm that cartesian bed leveling with a prusa i3 variant the measured offset and the actual offset are the same, and that from multiple locations and heights on the printer it reliably will run through the G28/G29 and move to the same height over the bed repeatedly.

Imy testing process was to move the Z axis to a random height, turn off the motors, move the hotend around to a new position, and go through my usual routine, at a Z offset of -5.8 I was able to reliably be unable to pass the feeler gauge under the hotend nozzle but by raising it by .1 the feeler gauge would then pass.

Delta leveling might still need to be tested but for me at least this current revision of leveling is reliable enough to print with.

thinkyhead, 2015-08-08:
@Sniffle I'm very glad to hear that! Was the earlier measured Z offset of -6.8 just off by 1mm?

To answer the original topic title, I think we should start testing delta bed leveling and when that's working reliably we can go forward with RC1. I'd really like to get bed leveling well-documented at that time too, using text, videos, images, diagrams, and whatever else we need to make all the steps clear.

Wackerbarth, 2015-08-08:
@thinkyhead -- Scott, I think that you are expecting too much. IMHO, the current bed leveling will NEVER work, in a practical sense, for delta printers. It simply doesn't have enough "variables" to adjust for the observed errors. You cannot fit the surface of a sphere with only a plane within the tolerances that we need to print successfully. (I'm not claiming that the surface that we are trying to fit is actually spherical, but I am using the term as a generic description of a non-planar surface)

There have been extensions implemented in older versions of Marlin which address this by various strategies.  For example, one strategy is to store the probe matrix and interpolate corrections from it in real time (an irregular bed surface).  Another approach is to make adjustable some additional mechanical description parameters to better allow the user to "tune" his machine.

But these changes have not been ported into the present code base. And I think that it is the wrong effort to do so. Instead, I would prefer to see us release the current code FOR NON-DELTA ONLY.
We can indicate that the fundamental mechanism for delta printers is present and that reasonable results may be obtainable by manually "tweaking" the parameters compiled into the firmware.

What I have done is to use the modified older version to "learn" appropriate parameter values for MY machine and ported those values into the new Marlin as a patch to by personal Configuration file.

We need to implement a better "delta kinematic model" into the NEXT (1.2 ?) release after cleaning up some things so that we are doing so in a cleaner manner rather than adding more and more "warts" to the present code.

fmalpartida, 2015-08-08:
I fully agree here. We should start thinks that the current code base is good. I need to close the loop this weekend, but it is looking in very good shape to have a good RC.

thinkyhead, 2015-08-10:
I haven't been keeping up with the Delta hardware development, so can you give me a quick primer as to why leveling with deltas is unusual - apart from the expectation that the delta base should always be level? Is it enough to do teensy adjustments to the endstop positions to achieve the same thing?

Wackerbarth, 2015-08-10:
The issue with leveling for deltas has little to do with the actual bed leveling, but rather expecting the procedure to compensate for other construction/calibration issues. There is some code from very old versions of Marlin which handled irregular beds (by interpolating between stored reference points), but this code has not been brought forward to anything "current".  Similarly, there is some code which attempts to handle a more complex model (such as providing an adjustment for the length of the arms and the position of the towers. However, in addition to being based on "old" code, it's iterative parameter fitting fails to converge on a number of printers because, again, although better, it does not have a sufficient parameter space to match the actual construction of those machines.

Although it can be argued that the proper correction is a mechanical adjustment, such adjustments are either difficult, or impossible (without replacing significant parts of the machine, and then without guarantee that the replacement will be any better).

A better model can certainly help. But we are limited in both implementation time and computational power.

Wackerbarth, 2015-08-10:
@clefranc's issues are now about the leveling, per se, but rather problems in defining the proper combination of pins and options to get the appropriate travel with and without his probe being active.

I feel that this is a "documentation" issue, but one very difficult to handle because of the convoluted code that has evolved. It is also very difficult to document something when no one seems to understand just how the machine was designed to handle various configurations.

Sniffle, 2015-08-10:
I would suggest looking at the original delta branch to get an better
understanding of the deltaeveling because it is delta specific without all
of the bloat of other versions making it convoluted.  Ill link to it in the
morning if you guys dont know where to find it.  Also smoothie while being
able to co puter properly the code should be similar to gain an
understanding of what it needs to be.

I run smoothie on my kossel mini... The code imementation is quite
impressive.

Wackerbarth, 2015-08-10:
@Sniffle -- Todd, I fully understand what the code NEEDS to do.  The problem is that I am stuck with what others will allow to be done to the present code base.  The current algorithm is INADEQUATE and, rather than tacking on yet another "wart", we need first to clean up the code.

Sniffle, 2015-08-10:
Ahh... That was my misunderstanding... And i do agree :-)

closed 2016-05-04 by jbrazio