MarlinFirmware/MarlinDev #3: SERVO*_PIN, NUM_SERVOS and HAS_SERVOS

Wackerbarth, 2015-08-02:
@AnHardt @Natealus @thinkyhead 
This replaces #2580 from the "Marlin" project

From one of the pins* files, I see the following

```
//
//These Servo pins are for when they are defined. Tested for usage with bed leveling
//on a Delta with 1 servo. Running through the Z servo endstop in code. 
//Physical wire attachment was done on EXT1 on the GND, 5V, and D47 pins.
//
 #undef SERVO0_PIN
 #undef SERVO1_PIN
 #undef SERVO2_PIN
 #undef SERVO3_PIN

 #if HAS_SERVOS
   #define SERVO0_PIN       47
   #if NUM_SERVOS > 1
     #define SERVO1_PIN     -1
     #if NUM_SERVOS > 2
       #define SERVO2_PIN   -1
       #if NUM_SERVOS > 3
         #define SERVO3_PIN -1
       #endif
     #endif
   #endif
 #endif
```

This seems backwards to me.

The pins ARE on the board (in this case, only one).  It is a user option whether, or not, it is connected.

So, I think that ALL of the #if and #endif in the above code fragment should be removed from pins_*

"HAS_SERVOS" be determined elsewhere (NUM_SERVOS >0) and the Sanity check, for EVERYONE, insist that, for example, 
```
#if NUM_SERVOS > 0 && SERVO0_PIN <1
  #error No pin has been assigned to Servo
#endif
```

Natealus, 2015-08-02:
Oh I agree it would be much simpler but when I reworked the pins for the Azteeg X3 Pro, profiles were linked together and some of the RAMPS pins just didn't line up with the setup on my 4 extruder system...turned a heater into always on state and no way to kill it. The only solution at the time was to #undef the pins to make sure they didn't get locked on RAMPS earlier in the hierarchy and not accept the X3 Pro pins like it should.

Albeit...I'm not really a coder yet. I still need to learn a full programming language so there's probably a more elegant way to do that. Don't beat me!!! lol :)

Wackerbarth, 2015-08-02:
@Natealus -- Nate, What I am referencing is nothing for which you are responsible. I am arguing that the "template" that you are following is improperly defined.

What happens if you just drop the #if clauses and retain the four #defines ?

Natealus, 2015-08-02:
Yeah I can try just defining the pins and taking out the #if phrases

Natealus, 2015-08-02:
Correct you are sir, I removed the #if statements and just have

 #undef SERVO0_PIN
 #undef SERVO1_PIN
 #undef SERVO2_PIN
 #undef SERVO3_PIN

 #define SERVO0_PIN        47
 #define SERVO1_PIN        -1
 #define SERVO2_PIN        -1
 #define SERVO3_PIN        -1

And it does work again with those commands from the previous issue. I'll push the changes unless something else seemed off in pins_AZTEEG_X3_PRO.h?

Natealus, 2015-08-02:
Pull request #9 has the fix. :)

thinkyhead, 2015-08-03:
#11 takes #9 to its logical conclusion.

Natealus, 2015-08-03:
Hehe I was starting the other pin files but you're faster at it lol.

closed 2015-08-03 by Wackerbarth

thinkyhead, 2015-08-03:
@Natealus True! My fingers are constantly breaking the sound barrier as I type.

Natealus, 2015-08-03:
@thinkyhead I think it's also cause I'm trying to run it through my head many times to make sure my changes don't break something cause I'm not as proficient with code as you lol.