MarlinFirmware/MarlinDev #10: Missing Travis test

AnHardt, 2015-08-03:
@boelle 
Please activate the Travis test.

boelle, 2015-08-03:
i dont know how

boelle, 2015-08-03:
found out

we need a travis ylm file but i guess its allready there

will enable during today

Den mandag den 3. august 2015 skrev Bo Herrmannsen <bo.herrmannsen@gmail.com

boelle, 2015-08-03:
morning :-D

should be done now

boelle, 2015-08-03:
hmm... instructions says:

"Trigger your first build with a git push"

now sure how and if i can do that from the web interface

AnHardt, 2015-08-03:
Thanks a lot.
The next PR will show if it works.

boelle, 2015-08-03:
make a PR with a txt file saying marlin is awesome firmware ?

AnHardt, 2015-08-03:
No. Will do something more meaningful this morning.

boelle, 2015-08-03:
closing this one as the first PR after travis was enabled now runs

closed 2015-08-03 by boelle