MarlinFirmware/MarlinDev #12: A configuration option to enable debugging outputs

Natealus, 2015-08-03:
It's often that @thinkyhead has to remerge new updates into his g29-debug branch so we can get more verbose output on the probing process. 

Could we make conditional debugging modes for G29 and other processes that often go without any output as a configuration_adv.h option? That way it's built in and can be enabled at anytime to find the roots of issues easier.

Natealus, 2015-08-03:
@thinkyhead @AnHardt  I think this would be a great thing to implement cause on the current dev code here...ABL is doing some really weird stuff again. It won't even do one line of probes with out a short move and probing the wrong spots and moving on and doing the same on the next line till it gets too close for the probe and starts grinding the servo.

Wackerbarth, 2015-08-05:
Implemented

closed 2015-08-05 by Wackerbarth