MarlinFirmware/MarlinDev #14: Issues to move over?

boelle, 2015-08-03:
Hi all

I got the idea of moving over some of the PR's so that they are where they should be

which ones should that be?

just give me a list of # numbers

boelle, 2015-08-03:
@AnHardt which issues do you think without doubt should be moved over?

AnHardt, 2015-08-03:
I wouldn't copy every message to the new repository. A simple link to the old thread, in the first message of the new issue her, and a "please continue at: link to the new thread", at the end of the old thread, is sufficient. (but i has to be the complete link - not just the #number).
What issues? Hard to say. I think we should do it one by one, when the time comes to make a meaningful statement.

boelle, 2015-08-03:
well that is also going to be the way.... i tried an issue mover site and its limited to issues with 35 comments or less

so far i have just copied those related to G29 as these needs to be fixed first.... we should really only allow those first time arround as they will never get done otherwise

i have been on a lot of hangouts and seen enough talk about a lot of things.... but i'm tired to that the most basic never has been touched

sorry for the rant.... but i think its  justified

closed 2015-08-03 by boelle

AnHardt, 2015-08-03:
@boelle 
I claim the opposite. For a lot of different setups it's working now (again).
There are special cases where a lot of uncommon conditions come together, like max-z-endstop, probe not min-z-endstop, 0,0 not in a corner of the bed, servo probe triggered when not deployed, ... , where we have problems - especially when all case met at once. I then the 'tester', with this setup, vanishes to an other branch - what shall we do?
Addressing the problems one by one.
___Systematically___ testing more and more cases.

(My printer mutated to a test-monster with a lot of things ___I___ do not need to print.
Extra: MAX6675, AD595, now 7 end-stops plus a probe, servo, fan-extension for RAMPS, 2 types of displays, ...
But neither my time nor my money is unlimited.)

boelle, 2015-08-03:
well i'm just about to be ready to test again... i just wanted a solid base to start from and something to fall back on if a code snapshot should not work for me