MarlinFirmware/MarlinDev #15: Dev version and bed levelling - wrong Z after G29 (several mm above bed)

boelle, 2015-08-03:
_From @harerama on March 24, 2015 8:33_

I just upgraded to the latest dev version. I noticed one change that broke my init G-code.
I had the following: 
  G28
  G1 Z10.0 F{travel_speed} ;move the platform down 10mm
  G29
Now, with the new version, the G1 move is NOT substracted from the final Z height, which results a wrong first layer Z (in this case, my printer started at Z=10+layer height which messed up the print).

The same init sequence was working perfectly previously.
I just need to remove the G1 line to fix the problem, and I am fine with it, but you may want to lay the stress on this functional change.

As far as I remember I copied a starting sequence from some webpage, so there may be many users in the same situation.

Regards

_Copied from original issue: MarlinFirmware/Marlin#1682_

boelle, 2015-08-03:
_From @harerama on March 24, 2015 8:43_

Actually, it seems to be something in the G29 command and not in the G1 in between - I'll add further details

boelle, 2015-08-03:
_From @thinkyhead on March 24, 2015 13:23_

@harerama Reset your homing offsets and see if that helps.

boelle, 2015-08-03:
_From @harerama on March 24, 2015 13:52_

I tried M502, if that is what you mean. My home offsets are all set to 0.
If I send just G28, everything works fine (but no levelling of course), then if I add a G29, my print starts about 1cm above the bed...
If you have a working start G-code sequence that includes auto bed levelling,I can try it (I have a Prusa i3 with Ramps 1.4).
My bed is almost perfectly flat, so I can avoid the G29 completely, but there must be a way to have it work.

boelle, 2015-08-03:
_From @thinkyhead on March 24, 2015 14:1_

Which of the bed leveling options are you using?

boelle, 2015-08-03:
> If you have a working start G-code sequence that includes auto bed levelling,I can try it (I have a Prusa i3 with amps 1.4).

_From @brainscan on March 24, 2015 14:19_

After you tried M502 did you save the settings? Think it's M500. I had a similar problem a long time ago but my start sequence is just G28, G29. Sorry if you have tried this, just seems like the same as my old issue and this was the fix.

boelle, 2015-08-03:
_From @harerama on March 24, 2015 14:43_

I will try again tonight - I am using grid levelling

boelle, 2015-08-03:
_From @Sniffle on March 24, 2015 15:45_

I just uploaded the firmware with the latest commits, the issue seems to me that the new raise before retracting probe isn't being taken into account for the Z height and z probe offset.

boelle, 2015-08-03:
_From @harerama on March 24, 2015 16:50_

@Sniffle Does it mean that you can reproduce the problem ?

boelle, 2015-08-03:
_From @Sniffle on March 24, 2015 16:53_

every time i go to print, the printer heats, runs through g28, then g29 and then begins printing in the air ~5mm which is what my rais before retracting is set to.

edit:  I'm going to add my z raise after probing to my offset to see if this corrects and proves the issue.  I'll report back in a couple hours.

boelle, 2015-08-03:
_From @Sniffle on March 24, 2015 17:44_

At this point hte gap is reliable, I am working towards my new adjusted offset and it's looking like the gap is 4mm, where that number comes from i don't know.  5 which was my raise after probing was too much and caused a minor crash, i went from that to double my original offset, which was still a crash but less to make the hotend raise.  so now i am down to 4mm + my original offset to make it print.

boelle, 2015-08-03:
_From @epatel on March 25, 2015 0:0_

Looking at this, if the mesh bed leveling could be at fault here. 
The default search height down to the bed is 4 mm (`MESH_HOME_SEARCH_Z `).

Are you using `MESH_BED_LEVELING`?

Note! It is not possible to compile the mesh + auto bed leveling, and a `G29` by itself does not activate the mesh compensation. To activate mesh compensation one need to run through probing all points, either via the display (if `MANUAL_BED_LEVELING ` was enabled) or a number of `G29 S2`.

boelle, 2015-08-03:
_From @alexborro on March 25, 2015 0:5_

Is there any document about how to use the Mesh Leveling? I wanna try it
tomorrow..

boelle, 2015-08-03:
_From @Sniffle on March 25, 2015 0:20_

I finally got the new arbitrary of set setup and now I can reliably print
again.

My offset went from 4.2 to 6.5,  how this happened I have no clue but math
has gone wrong somewhere by 2.3mm

boelle, 2015-08-03:
_From @epatel on March 25, 2015 0:31_

@alexborro I think I described it shortly in the pull request. Maybe got lost in all the text.

*The bed level option is added last in the Prepare menu. When selecting it the hotend will travel to all points and one use the encoder to move it up/down. Move it down until the usual paper thickness from the bed and press the encoder to "enter" that value, keep doing that until all points have been entered. The mesh can be saved in the EEPROM so storing them is good. A G28 homing will show if the mesh is active or not. If it does not travel to the first measure point (for sync position with mesh) the mesh is not active.*

https://github.com/MarlinFirmware/Marlin/pull/1619

Later I also added `G29`, with which one can do this without a display. 

`G29` or `G29 S0` Print mesh info

`G29 S1` Initiate probing, will do a homing + travel to first probe point. Now use Printrun or what you use to slowly lower the hotend until it touches the bed, i.e. with a paper between to feel when it's close enough.

`G29 S2` Save current height for the current point, then travel to next probe point. Repeat the manual lowering until touches. Do this until all points has been probed.

I use a paper and do the *goto next point* when the paper feel loose, which gets the compensation a little too loose so I compensate that in Slic3r with "Z offset" (in Printer Settings) set to `-0.05`. 

When probing and `MESH_HOME_SEARCH_Z` set at 4 the Z endstop is considered being at 4 mm and it is possible to search downward (yes, this will press on the Z endstop unless it is an opto) to 0. It might be possible to have the endstop below the bed, but I think the traveling between the points need to change as it will travel on the endstop level between the probe points, so an extra lift might be needed.

boelle, 2015-08-03:
> the endstop below the bed, but I think G28 might need to change as the
> initial movement to the first probe point could be bad so an extra lift

_From @alexborro on March 25, 2015 0:35_

Got it, but how do I use my Z Probe to do an automatic probing?

boelle, 2015-08-03:
_From @epatel on March 25, 2015 0:43_

@alexborro I don't have that setup so I have not added that, sorry. 

But I think it shouldn't be to hard to add, either taking code from the auto bed leveling parts or optionally add the mesh there.

boelle, 2015-08-03:
> But I think it shouldn't be to hard to add, wither taking code from the

_From @alexborro on March 25, 2015 0:46_

I will try to implement it tomorrow.

boelle, 2015-08-03:
_From @epatel on March 25, 2015 0:52_

@alexborro Cool! Please let me know how it works, and feel free to ask I'll help if I can.

boelle, 2015-08-03:
_From @thinkyhead on March 25, 2015 8:25_

@Sniffle If you've figured out the printing in the air problem, please contribute to this conversation! https://github.com/MarlinFirmware/Marlin/issues/1507

Or… have you just found a workaround?

It sounds like a fairly simple issue, in a way, if it is persisting for you. Somehow the Z coordinate is getting lost by the time the probing routine ends, so the firmware probably thinks Z = 0 when printing starts.

boelle, 2015-08-03:
_From @harerama on March 25, 2015 8:54_

Here is the log of the terminal:
  - G28: if I print after that everything is ok (but no bed levelling)
  - then G29
  - then I manually go down Z for the nozzle to almost touch the bed (true manual homing)
=> the Z value is @ -2.7, so I would have printed 2.7 mm above the bed

I don't know where this 2.7 value comes from...

I am using grid levelling (the default one)
_______________
Send: M501
Recv: echo:V17 stored settings retrieved (366 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z3990.00 E758.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X250.00 Y250.00 Z1.75 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P1000.00 R1000.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P15.13 I1.43 D40.17
Recv: echo:Filament settings: Disabled
Recv: echo:Z-Probe Offset (mm):
Recv: echo:   M851 Z-7.30
Recv: ok
[...]
Send: M114
Recv: X:87.00 Y:148.99 Z:7.28 E:951.41 Count X: 87.00 Y:149.00 Z:7.22
Recv: ok
[...]
Send: G28
Recv: ok
[...]
Send: M114
Recv: X:169.00 Y:101.00 Z:7.30 E:951.41 Count X: 169.00 Y:101.00 Z:7.30
Recv: ok
[...]
Send: G29 V4
Recv: G29 Auto Bed Leveling
Recv: Bed X: 20.00 Y: 20.00 Z: 7.28
Recv: Bed X: 130.00 Y: 20.00 Z: 7.20
Recv: Bed X: 130.00 Y: 150.00 Z: 7.06
Recv: Bed X: 20.00 Y: 150.00 Z: 7.14
Recv: Eqn coefficients: a: -0.00073138 b: -0.00106613 d: 7.31540012
Recv: Mean of sampled points: 7.16992521
Recv: 
Recv: Bed Height Topography:
Recv: +-----------+
Recv: |...Back....|
Recv: |Left..Right|
Recv: |...Front...|
Recv: +-----------+
Recv:  -0.03058 -0.10802
Recv:  +0.02757 +0.11103
Recv: 
Recv: planeNormal x: 0.000731 y: 0.001066 z: 1.000000
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: 1.000000 0.000000 -0.000731
Recv: -0.000001 0.999999 -0.001066
Recv: 0.000731 0.001066 0.999999
Recv: ok
Recv: echo:endstops hit:  Z:7.14
[...]
Send: M114
Recv: X:86.99 Y:148.99 Z:7.35 E:951.41 Count X: 87.00 Y:149.00 Z:7.13
Recv: ok
[...]
Send: G1 X100 Y100 F6000
Recv: ok
[...]
Send: G1 Z0
Recv: ok
[...]
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G91
Recv: ok
Send: G1 Z1.0000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G91
Recv: ok
Send: G1 Z0.1000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: M114
Recv: X:100.00 Y:100.00 Z:-2.70 E:951.41 Count X: 100.00 Y:100.00 Z:-2.88
Recv: ok

boelle, 2015-08-03:
_From @harerama on March 25, 2015 9:2_

Here is my full Configuration.h - in case it helps

    #ifndef CONFIGURATION_H
    #define CONFIGURATION_H

    #include "boards.h"

    //===========================================================================
    //============================= Getting Started =============================
    //===========================================================================
    /*
    Here are some standard links for getting your machine calibrated:
     * http://reprap.org/wiki/Calibration
     * http://youtu.be/wAL9d7FgInk
     * http://calculator.josefprusa.cz
     * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
     * http://www.thingiverse.com/thing:5573
     * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
     * http://www.thingiverse.com/thing:298812
    */

    // This configuration file contains the basic settings.
    // Advanced settings can be found in Configuration_adv.h
    // BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

    //===========================================================================
    //============================= DELTA Printer ===============================
    //===========================================================================
    // For a Delta printer replace the configuration files with the files in the
    // example_configurations/delta directory.
    //

    //===========================================================================
    //============================= SCARA Printer ===============================
    //===========================================================================
    // For a Delta printer replace the configuration files with the files in the
    // example_configurations/SCARA directory.
    //

    // User-specified version info of this build to display in [Pronterface, etc] terminal window during
    // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
    // build by the user have been successfully uploaded into firmware.
    #define STRING_VERSION "1.0.2"
    #define STRING_URL "reprap.org"
    #define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
    #define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
    #define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
    //#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

    // SERIAL_PORT selects which serial port should be used for communication with the host.
    // This allows the connection of wireless adapters (for instance) to non-default port pins.
    // Serial port 0 is still used by the Arduino bootloader regardless of this setting.
    #define SERIAL_PORT 0

    // This determines the communication speed of the printer
    #define BAUDRATE 115200

    // This enables the serial port associated to the Bluetooth interface
    //#define BTENABLED              // Enable BT interface on AT90USB devices

    // The following define selects which electronics board you have.
    // Please choose the name from boards.h that matches your setup
    #ifndef MOTHERBOARD
      #define MOTHERBOARD BOARD_RAMPS_13_EFB
    #endif

    // Define this to set a custom name for your generic Mendel,
    // #define CUSTOM_MENDEL_NAME "This Mendel"
[... 761 lines cut ...]
     * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
     * 301 - Rambo  - uses Analog input 3
     * Note may require analog pins to be defined for different motherboards
     **********************************************************************/
    // Uncomment below to enable
    //#define FILAMENT_SENSOR

    #define FILAMENT_SENSOR_EXTRUDER_NUM    0  //The number of the extruder that has the filament sensor (0,1,2)
    #define MEASUREMENT_DELAY_CM            14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

    #define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
    #define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
    #define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
    #define MAX_MEASUREMENT_DELAY           20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

    //defines used in the code
    #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

    //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
    //#define FILAMENT_LCD_DISPLAY

    #include "Configuration_adv.h"
    #include "thermistortables.h"

    #endif //__CONFIGURATION_H

boelle, 2015-08-03:
_From @thinkyhead on March 25, 2015 9:5_

???

    Recv: echo: M851 Z-7.30

???

__7.3 + 2.7 = 10.0__

boelle, 2015-08-03:
_From @thinkyhead on March 25, 2015 9:26_

Mulling this over… When `G29` begins, my guess is that it just assumes you did `G28` and that the Z axis hasn't been moved. So maybe doing a `G1` before `G29` isn't truly "supported." But it should be… Your log indicates that the firmware seems to be aware of the Z coordinate. If that's the case, when starting a print, the firmware should be moving to the starting coordinate for the print. It should move Z from its known position to 0 for the first layer.

So, all in all very weird.

boelle, 2015-08-03:
_From @harerama on March 25, 2015 9:36_

@thinkyhead: too early for me to spot easy math...

I removed the G1 between G28 and G29, so this is not the problem.
The sequence for the log above is:
 G28
 G29
 G1 X100 Y100 F6000
 G1 Z0 (the nozzle is 2.7 mm above the bed)
 several G1 to touch the bed (so down 2.7 mm)

boelle, 2015-08-03:
_From @SkyFlyer2 on March 25, 2015 9:54_

I'm also have this problem. This issue occur after adding command "do_blocking_move..." into retract probe function.

That all, what i know.

boelle, 2015-08-03:
_From @thinkyhead on March 25, 2015 9:54_

So again, very odd. It ends up at `Z_RAISE_AFTER_PROBING` - `Z_PROBE_OFFSET_FROM_EXTRUDER` = ~2.7mm above the bed in the case where you don't move the bed down.

boelle, 2015-08-03:
_From @scotty1024 on March 25, 2015 12:28_

I can add that this may not be a recently added bug. I have a Printrbot Full Metal Simple running their branch of firmware and out of the blue one day in Feb it started printing in air after the G29. No firmware change, it's running the factory code it came with. I switched from CURA to Simplify3D for its slicing but couldn't really see anywhere I got the startup wrong and going back to CURA didn't get me back to working. I'm thinking EEPROM calibration setting? That's all I've "changed" and do so frequently to keep calibration tight.

boelle, 2015-08-03:
_From @harerama on March 25, 2015 13:0_

I am also using EEPROM.

Everything was working good until I upgraded Marlin to dev version because I had trouble getting EEPROM to work in my previous version.
This may be a hint though, and I will try when I have time to deactivate EEPROM and run G29.

boelle, 2015-08-03:
_From @harerama on March 25, 2015 18:21_

Same problem without EEPROM support.
The problem seems to come from the last raise after last probe point - which is not taken into account...

boelle, 2015-08-03:
_From @SkyFlyer2 on March 25, 2015 18:35_

Yes, this is the case. The last climb of the probe (do_blocking_move ...) is not taken into account and marlin thinks that there was no lift.

It looks as if we raised Z-axis by hands.

But how to fix it?

boelle, 2015-08-03:
_From @epatel on March 25, 2015 18:41_

@harerama Could you perhaps checkout tag `1.0.2` and see if that behaves like this? Then we would know if it was introduced before or after that point. If after one could binary search a little among the merges into Development until we find which commit caused this. Before? Yeah then maybe use tag `1.0.1`. Well, this can be a bit tedious I know. git has a command for this but I think maybe manual could work here http://git-scm.com/docs/git-bisect

boelle, 2015-08-03:
_From @harerama on March 25, 2015 19:5_

Found the bug in Marlin_main.cpp, in function set_bed_level_equation_lsq (@line 1103)
Delete the following lines:

    // put the bed at 0 so we don't go below it.
    current_position[Z_AXIS] = zprobe_zoffset; // in the lsq we reach here after raising the extruder due to the loop structure

I don't understand the purpose of this line, nor the comment.

Again, this WAS working with 1.0.2 at some point.

boelle, 2015-08-03:
_From @epatel on March 25, 2015 19:12_

The commit https://github.com/MarlinFirmware/Marlin/commit/05932e445883c4fdf5620f90d503a3745639ffbc
The first comment also describes an unintended scenario I guess?
Maybe line 258 (current commit) where `float zprobe_zoffset;` should have ` = -Z_PROBE_OFFSET_FROM_EXTRUDER` just as a precaution?

boelle, 2015-08-03:
_From @thinkyhead on March 26, 2015 3:8_

@harerama Good catch. I wasn't looking anywhere near that function. Clearly that line doesn't belong, but maybe it was intended to follow the `plan_set_position` line…? I will remove it if no one else has made a PR yet.

@epatel I see that you're correct. The `zprobe_zoffset` value is never initialized unless EEPROM is in use. I'm confused and annoyed by the way that `gcode_SET_Z_PROBE_OFFSET` and the Z offset interact, however. The custom M code requires a value between -15 and -5 but then it sets the offset to the negative if the input, so it will be between +5 and +15. So yes,  `float zprobe_zoffset = -Z_PROBE_OFFSET_FROM_EXTRUDER` is correct, but I just find it weird that it insists on being the opposite sign to everything else. So I'm going to see about making that more consistent as well.

boelle, 2015-08-03:
_From @thinkyhead on March 26, 2015 3:12_

I could use some feedback on the lines that read:

    #elif defined(Z_PROBE_ALLEN_KEY)
      engage_z_probe();

Should this be replaced by:

    #elif defined(Z_PROBE_ALLEN_KEY) || defined(SERVO_LEVELING)

…? It seems like this is correct, because the `engage/retract z_probe` functions customize their behavior based on both of these flags.

closed 2015-08-03 by boelle

thinkyhead, 2015-08-03:
@boelle It makes me sad to see you having to copy a bunch of old threads like this. I think it's fine to just write a small blurb linking to the original issue, with references to the original participants. Then we will get it in our notifications and just refer to the old thread.

boelle, 2015-08-03:
i did not copy them :-D

there are tools for that...

but the only decent tool had a limit of issues with 35 comments or less...

was easy to use thou... just point to issue as a link and tell where to stick it and hit move

boelle, 2015-08-03:
btw.. only moved the most important ones so far

less clutter etc...

thinkyhead, 2015-08-03:
If you have an automated tool, that's very helpful. Too bad about the 35 comment limitation. I wonder why 35 is the cut-off.

boelle, 2015-08-03:
it a bug... the tool was: https://github-issue-mover.appspot.com/

boelle, 2015-08-03:
if you look at their git repro some say its a config thing... but without telling how to set it up

maybe their git source code can tell the secrect.... but well not anything i will waste time on... linking like i did took the same amount of time