MarlinFirmware/MarlinDev #19: normal advance direction change

boelle, 2015-08-03:
_From @bkubicek on May 22, 2012 6:33_

I am right now trying spending my pre-worktime trying to understand the "original" advance, so i can help out. I am not sure if this is a problem, but the Direction setting of the main ISR happens, independent if Advance is active or not. Not 100% sure however if this is allright, since I have not finshed to understand everything; but if you have a move that extrudes and then retracts, might it not be the case that there are still e_steps unperformed, and the main ISR changes the direction to early?

_Copied from original issue: MarlinFirmware/Marlin#176_

boelle, 2015-08-03:
_From @bkubicek on May 22, 2012 6:38_

sorry, my bad. I have overread something.

thinkyhead, 2015-08-11:
We have a lot of work to do to fix `ADVANCE`. Right now I (personally) consider it broken.

closed 2016-05-04 by jbrazio