MarlinFirmware/MarlinDev #20: G29 must require homing of all axes

boelle, 2015-08-03:
_From @gaussgss on March 31, 2015 18:54_

When Z axis isn't homed and bed auto-leveling requested, `G29` produces wrong results.

Tested on commit be593b600b8b9b427656dd27be7fc277aec0a8a5

_Copied from original issue: MarlinFirmware/Marlin#1760_

boelle, 2015-08-03:
_From @thinkyhead on April 1, 2015 2:37_

@gaussgss Currently the `G29` command doesn't care if Z is at a known position, just that X and Y have been homed. It probably should prevent using `G29` unless Z has also been homed (especially on deltas).

boelle, 2015-08-03:
_From @Ziggy2013 on April 1, 2015 8:57_

It's not clear to me why Z must be homed before a G29.  The G29 itself needs to establish where the Z home is a first step before probing. And the point the G29 uses to establish Z Home could be different depending on the type of probing (eg 3 point, 9 point etc)

boelle, 2015-08-03:
_From @thinkyhead on April 1, 2015 9:37_

@Ziggy2013 Well, that 's what `G29` is failing to do – establishing a known Z position before probing. If it doesn't get it from a `G28` homing command, then it must establish it when first starting to probe the bed, and I'm not sure it does that.

boelle, 2015-08-03:
_From @Wurstnase on April 1, 2015 9:44_

I don't think that's a good idea to implement the home z in the g29. On Repetier this makes also some trouble. For the transformation you need a fix point from where you can calculate it. If you don't home z and make a g29, the probe will not know where it is.

You can't measure the length of your living room from everywhere. You will go to one wall/edge first.

boelle, 2015-08-03:
_From @nophead on April 1, 2015 9:54_

I home to the top and use my probe to find where the bed is, not as a
bottom end stop. I think Marlin is broken because it treats the probe as a
bottom end stop, so whenever it hits it resets the Z home point. At some
point this needs to be fixed so it uses the height information from the
plane equation, which is more accurate.

If you don't a bottom endstop then the probe needs to act as one in G28 but
not G29. I feel yet another horrible global flag being born.

boelle, 2015-08-03:
_From @Ziggy2013 on April 1, 2015 18:7_

IMHO the G29 should first establish a start "reference" Z = 0 point using the probe.  Where that reference point is on the X/Y plane needs to be set based on the type of probing to be done.  eg a 3 point probe may require a different start Z reference point to a 9 point probe.

If G28 is also going to set that Z "reference" point as well, then the same X/Y reference point as used by the G29 probing needs to be used.  

Need to also keep in mind that not all auto leveling implementations use a separate Z probe.  For example, some implementations use the extruder tip itself to find the Z reference point.  Others use a dockable Z probe sled. In these implementations should G28 always try to find the Z reference point using the extruder or the sled?  

In the older version of Marlin I am using with auto leveling set the G28 only homes X and Y.  The G29 first raises the extruder/probe, moves to the centre of the bed, lowers the probe until it touches the bed to establish a Z reference point.  After that the G29 does its 9 point probing.  This all works very well. I am not sure what adding Z homing to G28 is trying to fix?
.

boelle, 2015-08-03:
_From @thinkyhead on April 2, 2015 0:34_

We have a PR pending that establishes a separate `Z_PROBE_PIN` so that the probe can be treated differently from an endstop. I suspected that the logic was being screwed up by the resetting of the Z position every time the probe switch is triggered. I will continue to document and do a full write-up of the program flow during bed leveling, and this should help to reveal where changes need to be made so that probes and endstops can get along together.

Note that in addition to probes that use the nozzle as the probe, there are some that use an NC switch with a floating X gantry, and these could conceivably re-use the Z endstop by applying the right (opposite) logic. (e.g., if the Z axis goes "below" the z endstop it breaks the circuit. Only works with an NO switch as the Z min endstop.)

boelle, 2015-08-03:
_From @ProjectEnder on April 2, 2015 20:57_

I know with the Delta you need to G28 sending G29. i would have to double check but i dont think if DELTA is defined that the received G29 runs a G28 before. this might be a good thing to add for the DELTA mode as for those you always need to home the unit first.

boelle, 2015-08-03:
_From @TechMasterJoe on April 3, 2015 11:7_

I honestly think that g29 needs to implement a integrated ver of safe home z before starting where it touches off the bed center and stores that info as a reference just for auto bed leveling. As a plain Z home can be done anywhere with the side-effect of over- or under- comp, with auto bed leveling it's like tossing in a random value for i in a PID never made sense to me.

Honestly think we need a fixed optical endstop that's on the Z axis independent of the bed as a hard reference that we set to a sheet of paper off the center of the bed. I know this sounds like more work, but hey, with a hard-set home the topographical bed map can even tell you direction and how many turns each bed screw is off-by, making bed leveling a breeze.

boelle, 2015-08-03:
_From @thinkyhead on April 4, 2015 1:31_

I think that if you issue a `G29` and the axes haven't been homed, it should just call `gcode_G28` directly instead of issuing a message that you need to home first. I think that would be a reasonable _expected behavior_ for this use case.

boelle, 2015-08-03:
_From @TechMasterJoe on April 4, 2015 3:29_

agreed

boelle, 2015-08-03:
_From @Ziggy2013 on April 6, 2015 9:59_

Please keep in mind that the `M206` and `G92` gcode functions also need to be handled if `G28`'s are proposed to be done implicitly with a `G29`.

boelle, 2015-08-03:
_From @ProjectEnder on April 10, 2015 21:14_

I was just looking over the code, to see if there would be a conflict with the `M206` and `G92` by Adding a `G28` to the `G29` command. As far as i can tell (still trying to study the code enough to know where everything is)  `M206` and `G92` do not call, there should be no problem adding a command to call `G28` before running running `G29`.

boelle, 2015-08-03:
_From @thinkyhead on April 11, 2015 0:44_

I think the issue is, if you do a `G29` expecting values that you intentionally set with a `G92` (or if offsets were changed using `M206`) then a `G28` could mess that up. Sometimes `G92` can do the job of a `G28` (on such machines as the QuadRap which have no end-stops). But I think that when you do `G92` the axes you set are flagged as "known" so there should be no XY homing if those axes are set (and marked known) by any method. Homing will only occur when axes are "unknown."

boelle, 2015-08-03:
_From @Ziggy2013 on April 11, 2015 2:50_

More generally the issue is the G29 needs known X = 0 and Y =0 reference
points to be able to probe the bed and calculate the bed plane using the
measured Z probing points. The G29 also needs to measure/calculate an
assumed Z = 0 reference point.  The known X = 0, Y = 0, Z = 0 reference
points then need to be the same during the print otherwise the calculated
plane will not be usable when adjusting the Z to match the bed plane.

In other words G28 homing followed by G92 / M206 offset adjustments should all be ok PRIOR to the
G29 being completed but not after.

thinkyhead, 2015-08-10:
What's the consensus? Do we want to require `G28 XYZ` homing before `G29`? I can imagine that `G29` without `G28` (of all axes) is a pretty rare thing. And, when you start a print, typically `G28` and `G29` are built-in. But, the raw fact is that `G29` can operate without a known Z – establishing a known Z is a really a separate thing.

Finally, I should note that we have also been discussing preserving the results of `G29` (and saving them to EEPROM as part of the custom configured values) so that the tilt of the bed is known. In that case, we would leave the correction matrix in place through `G28` instead of clearing it ("your bed is flat") as we do now. Then the machine would always move with correction — except when homing, since you want the axes to move in straight lines as they seek out end-stops (and not when babystepping, since that's direct-to-stepper).

Wackerbarth, 2015-08-10:
In the scenario that you describe, G29 without G28 (Or as in G28 G29 G29) would provide a measure of just how flat the bed appears to be since the latter G29 would already be corrected for bed offset and tilt.

I'm not convinced that "homing" cannot work with correction. (It seems as you are describing homing as some automated version of baby stepping ).

But, in any case, I vote for "G28 is not required as long as the machine knows, sufficiently well, where it is."  In particular, I'm thinking of those printers that have very slow Z travel and use their probe to establish the bed location.

Ziggy2013, 2015-08-10:
I suggest you think this through from the perspective of a regular 3d Printer user.  Not all of them will understand (or want to understand) all the theory behind G28's and G29's.

Many times I see 3D Printer users with start up gcode which is wrong because they haven't realised why the sequence is important. 

The G28/G29 sequence required needs to be very clear and consistent. And if not followed warnings should be issued.

Nandox7, 2015-08-10:
Why not even trigger a G28 every time a G29 is issued?
Before you run the G29 you want to make sure that the printer knows where it is so that all the probing 
actions execute within the expected area. 
Plus the time the G28 takes is a fraction of what the G29 takes, so the added time is rather minimal.
This would reduce the need to always have the two code added to the gCode and reduce duplication of code that need to be added in the firmware so less checks as the code will always know that a homing is done before it probes the bed.

So several advantages and only one disadvantage that isn't that big, time.

Wackerbarth, 2015-08-10:
@Nandox7 -- Before you make general conclusions, please make sure that you have really considered ALL of the cases. There are some printer configurations that CANNOT perform the default G28 immediately before every G29. And depending on just what actions result from a G28, there are cases where something different is needed. Rather than attempting to handle every case directly in the firmware, it makes much better sense for us to simply provide the components and expect the user to select the ones that are appropriate for his configuration.

Nandox7, 2015-08-11:
@Wackerbarth how much those printers represent in the overall total configuration types?
Can those be changed?
My suggestion was in the way to try and simplify the process.

Ziggy2013, 2015-08-11:
If there are particular printers which require something a bit outside the ordinary wrt G28/29, then use the configuration.h parameters to define the different behaviour.   

The "standard" behaviour should not be compromised to accommodate all printers and vice versa.

Wackerbarth, 2015-08-11:
Simplify for whom. Certainly not those who have to maintain the firmware or those who have any printer other than the most popular.  All of the decent host drivers already have a provision for the inclusion of constant text additions at certain critical points in the print cycle.

Any printer supported by a vendor will have the manufacturer supplied phrases.

I'm much more inclined to go in the opposite direction, removing all of the fancy sequences from the core firmware and then providing micro-code sequences as a part of the configuration. (Much like the RISC philosophy in CPU design)

Wackerbarth, 2015-08-11:
@Ziggy2013 -- Yes, but that "standard" doesn't require that it be reduced to some single operation especially designed for the currently favored model. It's much better to have instructions that don't do too much and expect everyone to use the (appropriately selected) sequence that fits their needs.

Even with the same hardware components, various users will have different definitions for the "proper" behavior. So we need to maintain the ability to customize that behavior without rewriting major pieces of the firmware.

thinkyhead, 2015-08-11:
The "design spec" for `G29` only gives this much, leaving the rest open:

- `G29` shall acquire the tilt of the bed and enable tilt compensation.

Right now `G29` only requires `G28` or `G92` for X and Y to be performed before it will operate, and it throws an error if you don't have a "known" XY position. It leaves it open for you to use `G92` instead of `G28` if you want, and maybe there are some good uses for that.

My personal feeling is that it's fine to just throw the error, and not to home on your behalf. But it would be interesting to see how it behaves if you alter it to home XY on your behalf. (Exception: On a delta `G28` always homes XYZ together.) The end result won't change for those who already have `G28` in front of `G29`. It would only affect those rare cases where `G28` was left out of a GCode file, or if you try to do `G29` from the console without `G28` first.

Ziggy2013, 2015-08-11:
@Wakerbarth

>Ziggy -- Yes, but that "standard" doesn't require that it be reduced to some single operation especially designed for the currently favored model. It's much better to have instructions that don't do too much and expect everyone to use the (appropriately selected) sequence that fits their needs.

Even with the same hardware components, various users will have different definitions for the "proper" behavior. So we need to maintain the ability to customize that behavior without rewriting major pieces of the firmware.<

This is no different to what I am suggesting.

Has anyone looked at the gcode definitions on http://reprap.org/wiki/gcode recently?  There are many new gcodes in this G28/29/30/31/32 space which describe the homing and Z probe behaviour. Is Marlin going to align with these or ignore them and  try to do something different?

thinkyhead, 2015-08-11:
@Ziggy2013 
- `G30` is implemented.
- `G31` isn't very useful right now, because we don't have support for variable proximity probes. But perhaps in the future.
- `G32` and `M557` (level with arbitrary points) aren't in our sites at this time. But it's an interesting feature. I presume it's meant for leveling based on a section of a large bed.
- `M558` (set probe type) seems helpful to make switching probes easier without re-flashing firmware. And, implies that handling for multiple probe types (and behaviors) could all be compiled in at once. That would take some re-working. But, we are building on the generalized probing stuff, and this might be easier in future.

Ziggy2013, 2015-08-12:
@thinkyhead 

Note also the comments against a number of these RepRap gcodes eg

>G29 The printer must be homed with G28 before G29.
>G29.1 Set the offset of the Z probe head. The offset will be subtracted from all probe moves.  ie the Z probe offset in the general case is positive

And there is also M565 which appears to duplicate G29.1

I guess my point with this is - If there are good reasons for not aligning with these RepRap gcodes then ok, But otherwise why not align?

thinkyhead, 2015-08-12:
G29.1 and G29.2 are unique to MachineKit. I think they should use something else and let `G29` be limited to actually starting the leveling procedure. No current hosts or slicers care about G29.1 or G29.2 I'm sure.

closed 2016-05-04 by jbrazio