MarlinFirmware/MarlinDev #25: Z Axis not moving up between probes or after probing completes on G29.

boelle, 2015-08-03:
_From @croadfeldt on March 29, 2015 11:31_

Might just be my config, but I'm not seeing the Z axis move up between probing points.

I'm using the grid mode, Z raises to 15 before initial probe, moves into position 1, deploys Z probe, lowers to the contact point, makes contact then lifts probe, moves to position 2, deploys probe which contacts immediately, raises probe and repeats for remaining 2 probe points. The only time it raises the Z axis is before the initial probe and it does not probe each position twice. Again this maybe a config issue, so here is what I think are the relevant config items. Also I added a serial dump to probe_pt to output z_before each time it's called.

    #define X_PROBE_OFFSET_FROM_EXTRUDER -22     // Probe on: -left  +right
    #define Y_PROBE_OFFSET_FROM_EXTRUDER -19     // Probe on: -front +behind
    #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.20  // -below (always!)

    #define Z_RAISE_BEFORE_HOMING 15
    #define Z_RAISE_BEFORE_PROBING 15
    #define Z_RAISE_BETWEEN_PROBINGS 10
    #define Z_RAISE_AFTER_PROBING 5 

Here's the output of G29 V4

    Marlin1.0.2
    echo: Last Updated: Mar 29 2015 06:28:19 | Author: (none, default config)
    Compiled: Mar 29 2015
    echo: Free Memory: 4770  PlannerBufferBytes: 1232
    echo:Hardcoded Default Settings Loaded
    echo:Steps per unit:
    echo:  M92 X80.00 Y80.00 Z4000.00 E106.00
    echo:Maximum feedrates (mm/s):
    echo:  M203 X300.00 Y300.00 Z2.00 E25.00
    echo:Maximum Acceleration (mm/s2):
    echo:  M201 X2000 Y2000 Z100 E10000
    echo:Accelerations: P=printing, R=retract and T=travel
    echo:  M204 P2000.00 R3000.00 T2000.00
    echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
    echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
    echo:Home offset (mm):
    echo:  M206 X0.00 Y0.00 Z0.00
    echo:PID settings:
    echo:   M301 P34.18 I3.07 D95.19
    echo:Filament settings: Disabled
    echo:Z-Probe Offset (mm):
    echo:   M851 Z-12.20
    ok
    G29 Auto Bed Leveling
     z_before: 15.000
    Bed X: 35.000 Y: 20.000 Z: 11.883
     z_before: 15.000
    Bed X: 170.000 Y: 20.000 Z: 11.418
     z_before: 15.000
    Bed X: 35.000 Y: 170.000 Z: 10.980
     z_before: 15.000
    Bed X: 170.000 Y: 170.000 Z: 10.525
    Eqn coefficients: a: -0.00340279 b: -0.00598749 d: 12.11915874
    Mean of sampled points: 11.20156192
     
    Bed Height Topography: 
    +-----------+
    |...Back....|
    |Left..Right|
    |...Front...|
    +-----------+
     -0.22181 -0.67631
     +0.68119 +0.21694

    planeNormal x: 0.003403 y: 0.005987 z: 1.000000
     

    Bed Level Correction Matrix:
    +0.999994 +0.000000 -0.003403 
    -0.000020 +0.999982 -0.005987 
    +0.003403 +0.005987 +0.999976 
    ok
    echo:endstops hit:  Z:10.53

I'm getting acquainted with the code, but will take some time to do so. Hopefully I can figure out if this is a code or config issue before I do.

_Copied from original issue: MarlinFirmware/Marlin#1730_

boelle, 2015-08-03:
_From @croadfeldt on March 29, 2015 11:34_

One more note, I see the same behavior on release 1.0.2 as well as my fork, which is current Marlin/dev with the Z_PROBE_AND_ENDSTOP pull request I have outstanding. I mention that as I wanted to make sure it was my fork that was broken.

boelle, 2015-08-03:
_From @thinkyhead on March 30, 2015 3:42_

@croadfeldt I'm looking at your `Z_PROBE_AND_ENDSTOP` PR and it looks good, will continue to comment on that there.

Meanwhile, have you experimented with these settings also? Sometimes I find that the Z value is too low, or the z divisor too high.

    #define X_HOME_RETRACT_MM 5
    #define Y_HOME_RETRACT_MM 5
    #define Z_HOME_RETRACT_MM 2
    #define HOMING_BUMP_DIVISOR {10, 10, 20}  // Re-Bump Speed Divisor (Divides the Homing Feedrate)

I don't necessarily think this is connected with your issue, but definitely want to make sure switches aren't being triggered twice.

Regarding endstop issues, I've been talking to someone else about whether the code should "disable endstops" in between the probes (if it doesn't already) to ensure that switches can't be thrown or cause endstop limits to be invoked. But that again would seem to have no bearing on your probe not probing each point properly.

In general, the raise-between-probe code has been a weird problem for a while. Recently the code to raise and lower the Z axis was moved into the function that raises and lowers the probe, which makes sense. Yet sometimes the behavior is strange, not waiting for movement, and in your case, not even initiating movement…

Basically, we need to insert lots of `SERIAL_PRINT` commands into this process and look at all the variables as it goes through the motions.

Not only is _this_ behavior still lingering, we still have a problem where at the end of probing the Z coordinate is off. And just from a casual reading, now I'm not sure that `G28` should add `home_offset[]` when coordinates are given (and certainly not _only_ when a coordinate is given), as we just stopped `G92` from adding the offset. So… we need to clarify the meaning of `home_offset` so that it applies in the proper cases, and document that clearly, otherwise things like this only become a source of confusion later.

boelle, 2015-08-03:
_From @croadfeldt on March 30, 2015 7:15_

@thinkyhead Thanks appreciate it. I'm still getting acquainted with the code, as I do I will see if I can help out there. Will start with lots of serial output as you mentioned, the biggest issue is the lack of information here I think, in regards to what is happening inside the black box...

As for the config changes, going to give them a shot now, but as you said, not sure they will affect anything. It does possibly eliminate a potential cause though, which is always a good thing. If I find a solution, will let you know.

boelle, 2015-08-03:
_From @thinkyhead on March 30, 2015 7:24_

@croadfeldt Careful. Running `G29` code in head repeatedly considered dangerous. :boom:

boelle, 2015-08-03:
_From @croadfeldt on March 30, 2015 7:29_

Yeah, I found that out already. :) Thankfully the reset button is easy to get to and so far the probe appears to be fairly robust and has bounced back everytime so far... Fingers crossed.

That said, the Z Min endstop has saved me from head crashes when the probe has failed to deploy. That is the main reason I wanted to split out the probe.

boelle, 2015-08-03:
_From @thinkyhead on March 30, 2015 7:59_

#1740 tries some new tricks, passing the same z to `retract_z_probe` that gets passed to `probe_pt` in the first place. Maybe that will help, or…?

boelle, 2015-08-03:
_From @oysteinkrog on March 30, 2015 19:8_

I see the same issue.

boelle, 2015-08-03:
_From @thinkyhead on March 31, 2015 3:27_

@oysteinkrog Post your `Configuration.h` and `Configuration_adv.h` someplace (like a gist) and I will use them for doing test builds and messing with the z probe code.

boelle, 2015-08-03:
_From @croadfeldt on March 31, 2015 3:43_

https://gist.github.com/croadfeldt/8e81729f8735272e70bd

boelle, 2015-08-03:
_From @oysteinkrog on March 31, 2015 5:57_

https://github.com/oysteinkrog/Marlin/commits/config_i3

boelle, 2015-08-03:
_From @thinkyhead on March 31, 2015 10:40_

I believe #1755 will finally fix all of our raise-between-probing issues. The probe was not being raised for `ProbeStay`, only for `ProbeRetract`.

boelle, 2015-08-03:
_From @Natealus on April 6, 2015 19:9_

Didn't quite get it all the way but it is doing much better. :1 :) Still all that weird crap with `Z_RAISE_AFTER_PROBING` adding Z and throwing the calculations way off. I've just disabled that line for the time being.

boelle, 2015-08-03:
_From @thinkyhead on April 7, 2015 2:10_

Getting closer, yet so far… I want to make sure fixing one thing doesn't suddenly cause issues for something else, so more documenting and debugging yet to do…

boelle, 2015-08-03:
_From @Natealus on April 7, 2015 5:33_

Yeah. On recent pulls it still doesn't deploy the probe servo on a plain G29. Just G29 E

boelle, 2015-08-03:
_From @thinkyhead on April 7, 2015 22:40_

@Natealus I just made a commit that might fix the issue of the probe not deploying without E. See if it helps. #1847

boelle, 2015-08-03:
_From @Natealus on April 9, 2015 5:42_

I've recently reconfigured on latest Dev build and it seems to still be doing it. With `G29 E`, I haven't gotten very good success with the measurements to flatness conversion. I think it has to do with my probe being -62mm Y from the hotend and I can't compile with that setting any higher than -39mm Y so I can't say that's the codes fault.

But remember how we were discussing the movements between probings? When its going across on X it pulls up straight and moves over straight. But when it moves back to first X position and moves back a row Y...it does that bowl curve whereas it's only calling one segment for the whole move. Even 20 segments per second appears virtually straight let alone 58-200 segments per second. That odd single segment move "May" be affecting measurements from the machine's expected virtual position to its actual position. Didn't someone determine that the `prepare_move_raw` was only calling 1 segment whereas `prepare_move` creates a straight move with the set delta segments?

boelle, 2015-08-03:
_From @Natealus on April 9, 2015 6:8_

And another quick question...should the probe offsets be to the center of the effector? Or should they be in relation to the first nozzle which has an `EXTRUDER_OFFSET` of -9.2, -9.2? Does the bed leveling routine have anything at all to do with `EXTRUDER_OFFSET_XY`?

boelle, 2015-08-03:
_From @thinkyhead on April 9, 2015 10:48_

@Natealus The probe is intended to be offset from the nozzle. So, if you moved the nozzle to the center of the bed, then you would move by the (negative) probe-offsets to move the probe to the center of the bed.

boelle, 2015-08-03:
_From @Natealus on April 13, 2015 21:20_

Yeah I was just curious if when it's doing it's calculations that the code is actually taking that into account or if its calculating from the center of the effector? I'm modifying my physical setup with the probe to bring it closer to the hotends and after I fix that I will do some more effective testing but it was...wayyyyy off on my previous print tests after G29 E. I've since disabled that ABL system and allowed my previous settings on height and radius to take over. Slight calibration here and there on the endstops but normal printing still is working very well. 

Except for this MakerGeek's brand of PLA glow in the dark. I've had two fails with it cause it slipped the extruder gear. Looks like it has bulges in some of the filament. Not surprised actually, that's exactly the problem with their blue ABS I had before. I've used several Jet filaments ABS and PLA and have not had a single damn issue with them. There's a new filament manufacturer in Salt Lake City that I have to try since they're close.

boelle, 2015-08-03:
_From @thinkyhead on April 14, 2015 1:14_

@Natealus Sorry for all the ABL issues – a combination of changes and the addition of Delta bed leveling from the kossel project have thrown off the tracking of the Z position (mainly).

For those who want to help out, I've got a debug build with lots of extra logging (which I'm keeping updated with the latest code) at https://github.com/thinkyhead/Marlin/archive/debug_leveling.zip

boelle, 2015-08-03:
_From @Natealus on April 14, 2015 2:8_

I'll give it a try after I get my new part to position the Z-Probe better. I think its likely cause it was so far away and I couldn't set the true offset without a compiling error. I found a good position but I need to move some things on the part so its not Too close to the hotend and melts lol.

boelle, 2015-08-03:
_From @johnny13 on April 18, 2015 19:48_

This is my config file. I am also seeing this issue. it just drags my servo arm along to the next point. [https://drive.google.com/file/d/0B7RA4a8zXgZYNk0yT09mNnVmSVE/view?usp=sharing](https://drive.google.com/file/d/0B7RA4a8zXgZYNk0yT09mNnVmSVE/view?usp=sharing)
  
I will try your debug leveling version out and report back.

boelle, 2015-08-03:
_From @thinkyhead on April 19, 2015 23:51_

Thanks @johnny13 ! I'll take all the feedback I can get. This should be fixed up and much more solid soon. I just need to go through these debug logs tonight, follow along with the code, and then I will post small incremental patches for everyone to test out.

thinkyhead, 2015-08-10:
How's this looking now? https://github.com/thinkyhead/Marlin/tree/debug_G29

dEDshort, 2015-08-17:
Have this same problem with an inductor z probe. Had to install z min endstop switch to prevent nozzle from scratching aluminum bed. Needs to raise in between probes for accurate measurement on bed.
The main reason added this comment theres a 404 error on thinkyheads debug_G29 link. Need to know where the latest code to upload and try would be. If you could help Thanks

Wackerbarth, 2015-08-17:
@dEDshort -- Try the version that I have under PR#54.  It is totally up-to-date. There are a few differences from Scott's approach. I converted his debug output into a feature that you need to enable in your configuration file so that we can have it as a permanent part of the source tree rather than a separate branch.

Also, so that you can actually print with the resulting firmware, I converted it to a debug option.
In order to get the output, you need to issue a "M111 32" command (and can turn it off with "M111 0".

Changes to the Configuration.h are in the  // @section bedlevel near line 443. Some options have changed. Be sure to check your configuration.

nitewing76, 2015-08-19:
Whenever my probe was not raising and being scraped along the bed, I always fixed the issue with a fresh "install" (meaning I would save the config files as *_old.h, overwrite everything, then reload my specific settings into the config files). I assume this works because of a stray character somewhere or a corrupt file?

As I recall, in Release 1.0.2 the Z_Probe does not raise raise before retracting the Z_probe...causing the corners of a mechanical switch to scrape the bed (adding stress to the switch, arm, and servo mount). I cannot remember the behavior in the RC, but since the time before @thinkyhead correct the servo jitter in the Dev version...the only time the probe retracts during G29 is at the beginning and the end. 

Can someone please provide a link to somewhere that describes the mathematics of ABL in detail? I have completed Calculus II and an introductory course on matrices and MatLab...and my neighbor has a Ph.D. in mathematics, so I can always bother him if I need more assistance. lol

Sniffle, 2015-08-19:
You need to specify what marlin you are using also what printer type
because as of right now the current marlindev master branch works
properly.  I have it running on a prusa i3 variant now without any issues.

nitewing76, 2015-08-21:
For this topic, I have no problems with the Z-probe while using ABL_PROBE_PT.

thinkyhead, 2015-08-31:
@nitewing76 Shall we close this for now? As for documentation on the bed leveling, the basic approach (as I recall) is known as the "least squares solution." Try Googling that to get an overview of the idea, and then ask around here some more about the specific implementation.

nitewing76, 2015-09-05:
I have heard of this before...but I cannot remember which class it was in. All of my textbooks are in PDF, so it will not be difficult to search for. Thank you very much! 

Just skimming over this thread to refresh my memory.....I think you can close it. because either the problem has been fixed or is being discussed elsewhere in more detail.

closed 2015-09-05 by Wackerbarth