MarlinFirmware/MarlinDev #26: Dev version and bed levelling - wrong Z after G29 (several mm above bed)

boelle, 2015-08-03:
Org link: https://github.com/MarlinFirmware/Marlin/issues/1682

thinkyhead, 2015-08-12:
@harerama @brainscan @Sniffle @epatel @alexborro @Roxy-3DPrintBoard @Ziggy2013 @oysteinkrog @alvinavr @lrpirlet 

Does this bug still exist? We have more PRs related to leveling pending in this repo, so watch this space…

Sniffle, 2015-08-12:
Im currently using the dev tree to print with, ill need to accurately
measure again to verify but my last test showed it being accurate.

kagouraki, 2015-09-24:
i have the same issue. is there a fix ?

epatel99, 2015-09-24:
I can only help if this is related to the MESH/MANUAL bed levelling feature.

Wackerbarth, 2015-09-24:
@kagouraki -- EXACTLY which version of the code are you using? If it isn't the RC latest RC branch in the Marlin repository, please verify that the issue exists there and provide you configuration so that we can understand the context.

kagouraki, 2015-09-24:
this is my config : 
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 dev"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// @section machine

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
// :[0,1,2,3,4,5,6,7]
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// :[2400,9600,19200,38400,57600,115200,250000]
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_SANGUINOLOLU_12
#endif

[... 703 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.0 //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         3.3 //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.9 //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20   //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

epatel99, 2015-09-24:
Looks like you are using the Mesh/Manual bed levelling feature :) Have you read this? 
https://github.com/MarlinFirmware/MarlinDev/blob/9a093b2e8f/Documentation/MeshBedLeveling.md

kagouraki, 2015-09-24:
i have read this and i have send you my config. What need i do ?

epatel99, 2015-09-24:
Please explain the problem.

Wackerbarth, 2015-09-24:
@kagouraki -- You failed to specify which source you are using. The configuration file appears to be from an older version of the code.

kagouraki, 2015-09-24:
i have loaded the firmware. I have gone to prepare->level bed. I have gone through the procedure. after this if i go to prepare->home axis it homes and then goes to x20 y20. if i try to print the first layer is 0.15 mm. In the screen it says 0.15 but the nozzle is several mm above the bed. This is my problem. 
@Wackerbarth  please give me the right source and i will reupload. Thank you

drbogger, 2015-09-24:
That's very similar to the problem I had with it, except my LCD said z was at 4mm, when it was actually at 0..

kagouraki, 2015-09-24:
how did you solve it ?

Wackerbarth, 2015-09-24:
Gentlemen:  If you are not using the RC branch from the Marlin repository, I suggest that you move to that version. And open an issue against that branch if your problems persist.

kagouraki, 2015-09-24:
can you provide a link ? i am confused with all the versions... sorry.

drbogger, 2015-09-24:
Yeah, I was planning on it, I didn't know it even existed until 15 minutes ago. I'll get on it in a bit.

epatel99, 2015-09-24:
@drbogger I made this specially for you :) https://github.com/MarlinFirmware/MarlinDev/pull/199 don't know if you got to test it? It moves to `[0,0,0]` instead of `[20,20,n]` as before.

drbogger, 2015-09-24:
Haven't had the chance to yet, but hopefully I'll have some time today.

Has this fix been implemented in the RC branch yet?

Also, which RC branch is the latest code? There's "Release, RC, RCBugFix"... I don't know which one is the correct one to use?

kagouraki, 2015-09-24:
i have downloaded this version https://github.com/MarlinFirmware/Marlin .
There is no Manual bed leveling section in the config... 
can anyone give me a version that has manual bed leveling option and has not the problem i mentioned above ?

Wackerbarth, 2015-09-24:
Release is a post-"candidate" tagged release. (1.0.2-1 at the moment). You probably don'r want that.
RC is the branch you want to test.

RCBugFix is an open branch to provide a place for anyone to propose Pull requests to the RC branch. We need it only to get around some workflow issues in github. As I incorporate patches, one version of RCBugFix is terminated and a new one, with the same name, is created so that there is an appropriate place for submissions.

epatel99, 2015-09-24:
@drbogger Good question. I put the PR to MarlinDev...Now, after the RC move MarlinDev seem to be turned up side down, and I'll wait before trying again. Could not find any (simple) instructions how the new build system works.

drbogger, 2015-09-24:
@epatel99 Could you submit the PR to the RC Branch? Whichever one is the correct one?? lol

epatel99, 2015-09-24:
@drbogger I put the changes on the dev branch https://github.com/MarlinFirmware/MarlinDev/tree/dev ...but I see now according to the readme they should have been made on master. But the readme also says "The dev branch contains the most up-to-date Marlin code.". My bad. I'll see what I can do. I don't have too much time to play the diff and merge game.

dagnall53, 2016-01-01:
I have just set up what i think was latest "RC3" release Onto my shapeoko based printer and had a similar issue. 

(#define DETAILED_BUILD_VERSION "1.1.0-RC3 From Archive"
#define STRING_DISTRIBUTION_DATE "2015-12-01 12:00")

The print head z offset after zeroing was way off ( approx 1cm higher) than what the Z_PROBE_OFFSET_FROM_EXTRUDER was set to ) . I also think I saw (sorry have not repeated this) different offsets in z heights after G29 and G30 (ie inconsistent behaviour). 

HOWEVER in exploring a possible solution, I tried using the (new?)  "Home offset" and set "M206 X0.00 Y0.00 Z10.00"  This has "sorted" the issue and now G29 G30 and G28 all accurately "zero" my z axis. I'm using eprom, so this comes up every time I start.
I would like to know why this " M206 Z10" needed setting, but hope it helps others. 
Also, what is "CUSTOM_M_CODE_SET_Z_PROBE_OFFSET" ??? might this be interfering with the settings? 

Dagnall
 
I have copied my #defines below in case they help: 

#define AUTO_BED_LEVELING_FEATURE 
#define DEBUG_LEVELING_FEATURE
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define AUTO_BED_LEVELING_GRID
// Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER -11     // Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 2    // Z probe to nozzle Y offset: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -1.40  // Z probe to nozzle Z offset: -below (always!)

  #define Z_RAISE_BEFORE_HOMING 15       // (in mm) Raise Z axis before homing (G28) for Z probe clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case.

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.

  #define Z_RAISE_BEFORE_PROBING 15   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 15    // How much the Z axis will be raised after the last probing point.
#define Z_SAFE_HOMING

#define CUSTOM_M_CODES
#if ENABLED(CUSTOM_M_CODES)
  #if ENABLED(AUTO_BED_LEVELING_FEATURE)
    #define CUSTOM_M_CODE_SET_Z_PROBE_OFFSET 851
    #define Z_PROBE_OFFSET_RANGE_MIN -20
    #define Z_PROBE_OFFSET_RANGE_MAX 20
  #endif
#endif

thinkyhead, 2016-04-18:
@dagnall53 We've fixed a lot of issues in `RCBugFix` and that will be the basis for our next update to MarlinDev. So if you don't have any problem with `RCBugFix` then we can close this issue and call it fixed!

closed 2016-05-04 by jbrazio