MarlinFirmware/MarlinDev #28: Homing & Leveling Bugs

boelle, 2015-08-03:
Org Link: https://github.com/MarlinFirmware/Marlin/issues/2040

paulusjacobus, 2015-08-03:
@boelle i have added an additional key to the G28 command that skips the clearing of the rotation matrix so you don't have to do an ABL for each subsequent print. I have been using it for a week and found it very useful and time saving feature. Would that feature be a candidate for extending Marlin?
Usage G28 S (S for skip)

boelle, 2015-08-03:
100% it can be usefull... why clear the matrix just because i want to establish what my current Z height is ?  :-D

Wackerbarth, 2015-08-03:
@paulusjacobus -- I think that the concept is very useful. However, I suggest that, we store the parameters for the rotation matrix in EPROM and that it be the default.

If the matrix is to be cleared and reset, that can be the responsibility of the G29 command.

boelle, 2015-08-03:
the copy of marlin i got from @Roxy-3DPrintBoard does excat that.... so a power cycle will not wipe the rotation matrix.

My only issue with G28 from that copy of marlin is that G28 places Z 4mm to heigh up... but Z before home setting in configuration.h is set to same number so i think its a easy one to fix

thinkyhead, 2015-08-03:
- What was the original justification for having `G28` throw away the matrix?
- Is it possible to know the total size of the EEPROM so we can write this info after the config info? (If not, we can just get use total data length stored after the "Vxx" version number, and write it above that.)

thinkyhead, 2015-08-03:
I'm renaming this issue because it's already off-topic.

paulusjacobus, 2015-08-04:
Cool happy to have a look into storing the rotation matrix into EEPROM (next weekend). My ABL correction seems very stable after powering up (incl. heating up the bed). Also Scott's comment "original justification for having G28 throw away the matrix?" made me to reconsider the G28 skip parameter. Maybe it shouldn't clear the matrix at all...

Roxy-3DPrintBoard, 2015-08-04:
> @thinkyhead  said
> What was the original justification for having G28 throw away the matrix?

This is just the way it always has been.  Who ever did the original G29 code for Marlin Auto Bed Leveling chose to clear the matrix there so everything would be back to normal after a G28.   It was just convenient place to do it.

> Is it possible to know the total size of the EEPROM so we can write this info at the end? (If not, we can just get use total data length stored after the "Vxx" version number, and write it above that.)

Its location really doesn't matter.  It just has to be in the same place for the read as it is for the write.

> How automated will this loading and saving the matrix to EEPROM be? Remember that EEPROM has a limited number of write cycles, then it's fried.

The way I implemented it, it gets stored at the same time a M500 stores the other settings.   When Marlin resets it reads the Correction Matrix at the same time it reads everything else from the EEPROM.    The user would only do a G29 when things get too far out of kilter.   Probably the user does a test print to verify the alignment is good, and then doess a M500 to store the new matrix.  

The EEPROM has at least 10,000 guaranteed write cycles.   But I think the new Arduino's are actually quoting 100,000 write cycles.   The EEPROM will be fine.

> Cool happy to have a look into storing the rotation matrix into EEPROM (next weekend). My ABL correction seems very stable after powering up (incl. heating up the bed). Also Scott's comment [...]

The work is already done.  It just hasn't been merged into the main Marlin tree yet.   Please check out:  

http://3dprintboard.com/showthread.php?8064-Saving-G29-Bed-Level-Correction-Matrix-to-EEPROM

I don't think we want extra complexity with a Skip parameter to G29.   I think if you run a G29 you are doing it for the purpose of generating a new Correction Matrix.   Instead, I think you want G28 not to clear the Correction matrix.   That would seem to work for all of the normal use cases.   For people that always to a G28 followed by a G29, it appears to work the same as it does now.  For people that just want to do a G28, they will continue to be using their last Correction matrix (whether it was loaded from EEPROM or generated a while ago via a G29).

Wackerbarth, 2015-08-04:
@Roxy-3DPrintBoard -- Do you have this code up on Github? I'd like to try merging it with the current "master".

One comment about the entire concept: The G29 matrix is just a special case of having a general set of parameters which determine the kinematic transformation from Object space to Machine (actuator) space.

As you know, a general 3D linear transformation uses 9 coefficients. But, because we assume that the scaling (Object distance to Machine steps) is a known set of constants, there are only 6 degrees of freedom. Additionally, 3 of those are related to the "zero" position. For the bed correction, we need to store the bed tilt (only 2 parameters). I think that the last degree of freedom is consumed by the locking of the Z-axis of the object to the machine.  So, in the general case, we need store only 5 parameters, including the offsets of the end stops.

However, if we maintain generality, using common code, we could store additional parameters representing more complex kinematics such as a delta (which I believe needs 11 parameters).

It would also open up the possibility of adding a compensation for the hyperbolic warp that appears when two supposedly parallel actuators actually have some mis-alignment.

In this context, I will point out that having G29 always reset to "factory" may not be the desired strategy.
If I already have a good idea of the values of the parameters, using the previous values as an initial guess can greatly speed up any iterative solution.

Therefore, I suggest that we might modify M500 - M503 to selectively work on portions of the configuration space. Then "G28 ; M502 •; G29" might be equivalent to, and therefore replace, the current behavior.

Roxy-3DPrintBoard, 2015-08-04:
> Do you have this code up on Github? I'd like to try merging it with the current "master".

No, it is not in a branch or fork.   But I can send you the ConfigurationStore.cpp file if you want.   Or you can just follow the direction in that post and in 10 minutes have the feature added to your fork.

I understand what you are saying about the  different coefficients.   Some of them are not really needed.   But there was plenty of EEPROM and it was easier (and cleaner) to just save and restore the entire matrix.   I guess if EEPROM space were to get tight that strategy would have to be rethought.

Wackerbarth, 2015-08-04:
I'm not concerned about the way this was done as long as it is considered a TEMPORARY solution.
The description was to suggest to anyone, such as yourself, who is rewriting parts of the code that there might be a more general approach.

I hate to see systems full of "special purpose" hacks when the result could have been accomplished in a more general way.

And I hate, even more, being expected to support those hacks as a legacy.

Roxy-3DPrintBoard, 2015-08-04:
Perhaps Temporary is the wrong word.   An evolving solution might be better wording.  To be a fully evolved solution it would need to encompass everything you mentioned up above.   But that is very difficult to do.   

I really don't have strong feelings on this 'feature'.   I've never used it other than to test it and make sure it acted reasonable.   I do a G29 at the start of every print.    But a lot of people don't like doing G29's so it does help them even though it is not a complete solution that handles every type of issue.

paulusjacobus, 2015-08-04:
@Roxy-3DPrintBoard great that the work already has been done but as @Wackerbarth commented, does it take only the necessary parameters? 
My customised "G28 S" seems a good approach to you. I'll do a pull request so it's up to the community to adopt or reject it. So far it is very handy when you have to print a big batch of prints (like car models).
@Wackerbarth I really like your analysis of the storage. I don't think G29 does a factory reset but G28 does. I'm pedantic here, G29 always needs G28 hence factory reset is implied which bring me to the principle of componentised G commands that are independent (no condition for do the command). G29 is such a way not an independent command since it cannot run without a G28. Should G29 include the homing sequence? Then it becomes more like a routine...
Also I noticed that I always need to home the z axis in order to do a successful ABL. BTW I use @thinkyhead 's G29_Debug code which works well for me.

Wackerbarth, 2015-08-04:
@paulusjacobus --No, I think you have it backwards. The default needs to be to keep the previous matrix. G28 performs an alignment to assure that one or more axes are in their proper known position relative to the machine frame. It's function is not to "reset" the machine. We already have M502 to do that. 

Depending on its implementation, G29 may not require that the Z-axis start in a known position.

You should not be designing the input language on the basis of some present implementation.
For now, if the matrix needs to be reset, we can let G29 do the reset. But some future G29 might not require that. So leave it to G29 to know what it needs.

paulusjacobus, 2015-08-05:
> It's function is not to "reset" the machine.
Cool that means the rotation matrix clearing code should be removed entirely.
>You should not be designing the input language on the basis of some present implementation.
That's exactly the problem. You nailed that there since we don't have clear requirements and a design. So I merely guessing what was intended. I frequently read the G code list on the Wiki but that list is very limited. The NIST standard offers also little help and  is very CNC oriented. G29 does not even exist...
We should (and I will help) extend the ABL documentation on the Marlin Wiki.

Roxy-3DPrintBoard, 2015-08-05:
> Cool that means the rotation matrix clearing code should be removed entirely.

No...  There is already support in the matrix class to create a unity matrix.   So, in the M502 where you are asking the printer to return to default (configuration.h) settings, it just plain makes sense to do the one line of code that sets the matrix to unity.    And, at the top of the G29 it set it to unity but for no reason other than an error in parameters could cause the G29 code to abort.   It makes sense to have the matrix at unity if you can't actually do a G29 with the requested parameters.   (This one could be argued).    

The point I'm making is there are times the matrix should be set to unity.   Just not as a result of the G28 homing operation.

paulusjacobus, 2015-08-05:
@Roxy-3DPrintBoard I think we need to get the use cases listed so it becomes more clear what the G codes 28 and 29 will need to support that will give us a better way to discuss the features. I'll make an attempt:
1) G28, G29 use case power up printer and I want to calibrate the printer bed: unity to corrected matrix
2) G28 use case: home after I printed a model, need to print more models with same correction matrix applied
3) G29 use case small corrections needed: I printed a long time and have some drift in the bed.(additional corrections)
4) G29 failed to do the ABL sequence (i.e. my servo got misaligned or the glass plate was shifted and the probe misses the glass plate): revert back to unity or revert back to previous correction?
G28 home, power down the printer (finished)
5) M502: Power up the printer and load the correction matrix (the printer bed is stable and corrections seem to be sufficient for quality prints/no calibration required)
6) G28, G29, M500: Do a calibration and save it to EEPROM for future use

Any other use cases?

Roxy-3DPrintBoard, 2015-08-05:
I think that is a good start for the different scenarios.  I have two corrections:

> 5) M502: Power up the printer and load the correction matrix (the printer bed is stable and corrections seem to be sufficient for quality prints/no calibration required)

Doing a M502 reverts the printer to the default configuration.   This should be identical to having the EEPROM turned off and the printer is freshly reset.    M502 should cause the printer's current settings to be as if they came from the Configuration.h settings.   And I think it can be argued the correction matrix should be set to unity at the end of this command.   

> 6) G28, G29, M500: Do a calibration and save it to EEPROM for future use.

I think this is plausible and correct.   But I suspect there is one more very similar use case:

G28, G29, test print, M500: Do a calibration, test it, and then save it to EEPROM for future use.

Incidently, the code published up above at that link does conform to all of those use cases.   @Wackerbarth Can you expand the use cases or provide more detail to handle the more complex 'corrections' you were envisioning?   It would be interesting to see how a more complete compensation for wrap and alignment mesh with the current thinking.

Wackerbarth, 2015-08-05:
The "factory" matrix is a unity matrix.
Any reference to "read EPROM" would return the factory settings if there is no EPROM.

"Power reset" should always do a M501 as its first operation.

M502 returns the factory (unity) matrix. That is what the current implementation of G29 expects.
However, a new G29 SHOULD accept any current matrix and calculate the changed result.
If I had a "perfect" printer, but with a slanted bed and perform
```
M502
G28
G29
M500
```
I should have everything ready to go.

If I run G29 again, it would give all readings of 0.00 and produce the same matrix that I am already using.

If I run G29 and get different results because my printer isn't perfect and changes over time,
I have the option to save the new matrix (M500), discard it and revert to the previous one (M501) or revert to the factory setting (M502).

But the default case is "run what you brung"

As to other more complex transforms just replace "matrix" with "transformation parameters". A 3x3 linear transform is just a special case of the more general transformation concept.

As for more complex transforms, consider this sequence
```
p = Z
p = D*z
p = D*z + E
p = A*x +C*y +D*z +E
p = A*x +B*x*y +C*y +D*z +E
```
They represent
1) The unity transform
2) A scaling (as in mm to steps)
3) With an offset (correcting for the end-stop location)
4) A slanted bed (The rails are not parallel to the bed)
5) A warped bed (Caused by having two rails which are not parallel to each other)

But, in general we have
```
p = f(A, B, C, ..., x, y, z)
```
where f is any continuous function.
I've only shown a single dimension. We actually deal with vectors.
Thus, the P vector would be the positions of all the actuators, the FRAME coordinates
x,y,z are the location in OBJECT space, X for short
A,B,C is some vector of transformation parameters, A for short
f is the transformation, or F for the set of transforms
So, in general
```
P = F(A,X)
X = F'(A,P)
```
Namely within the limits of interest, there is an inverse mapping

Mathematically, I'm sure that there are a few more conditions, but it has been decades since I worried about the criteria necessary to describe the limits on the mapping.

Note that, in theory, we might even have a transform to compensate for a bowed Acme rod that generates a wave rather than a straight line.

But that is generally beyond the scope of our goal.
However, we are interested in the transformation that applies to DELTA printers.

Roxy-3DPrintBoard, 2015-08-05:
Wow!   I have some questions, but it is going to take a little bit of time to get my thoughts organized.

paulusjacobus, 2015-08-05:
@Roxy-3DPrintBoard the code you have produced so far seems like the best starting point since it covers a lot of scenarios..
>M502 should cause the printer's current settings to be as if they came from the Configuration.h settings &
>correction matrix should be set to unity

Yes I agree that was the intended use of M502, "return to the default factory settings". I don't think matrix corrections are part of the default factory settings.

>it is going to take a little bit of time to get my thoughts organized

Yes I feel a bit overwhelmed and need to analyse it a bit further.

Roxy-3DPrintBoard, 2015-08-06:
> Yes I agree that was the intended use of M502, "return to the default factory settings". I don't think matrix corrections are part of the default factory settings.

Well, perhaps not intentionally.  But I think you can argue it from the opposite perspective to get to the right answer.   If you are restoring factory settings you probably don't want a non-unity correction matrix to be active.   If you are restoring factory settings, you do want everything in a deterministic state and not have random values affecting things.

Wackerbarth, 2015-08-06:
I think we all agree that M502 SHOULD set the matrix to unity (assuming the present state vector implementation).
From my perspective, that is a part of the "factory design". I don't know of anyone who DESIGNS their device with intentional skew, etc.

But, if we look at it simply from a "state vector" perspective, the state could have the designed end-stop offsets and gear ratios built into its standard transform matrix.

So, eventually, we will want to call on the Kinematics module to reset, dump, and load whatever in needs to store in order to perform coordinate frame transformations.

thinkyhead, 2015-08-08:
> at the top of the G29 it sets it to unity but for no reason other than an error in parameters could cause the G29 code to abort

No, it sets it to unity because if it didn't it would be moving with compensation already during the bed leveling, and coordinates would all be screwy.

> I think we all agree that M502 SHOULD set the matrix to unity

Yes, in the sense that the "default" is for ABL to be turned off. But I think we also need some other GCode that will set it back to unity, besides `G28`. And note: when you turn off ABL, you need to remember to set the current (model) position to be the same as the current machine position (stepper position).

The argument for _not resetting the matrix_ with `M502` is: The machine is ready to go, movement is all good with compensation, so why should `M502` require you to follow it up with another `G29` to get back to that happy state?

Wackerbarth, 2015-08-08:
The argument for not resetting the matrix with M502 is -- I disagree. M502 is an appropriate code for resetting parameters to "factory".  The factory value for the ABL matrix is the unity matrix. PERIOD. If I assembled a perfect mechanical system, then "out-of-the-box" the G29 probe values would all be 0.00.

What you are saying is that you want the reset PART of the configuration parameters.  I agree that making that an option would be appropriate.  But we don't need yet-another G-code for the purpose. We should overload the current one with some "code_seen" flags. In the absence of those flags, we would reset all sections of the operational parameters.

Roxy-3DPrintBoard, 2015-08-08:
> >at the top of the G29 it sets it to unity but for no reason other than an error in parameters could cause the G29 code to abort

> No, it sets it to unity because if it didn't it would be moving with compensation already during the bed leveling, and coordinates would all be screwy.

Good point.   You are correct.

Wackerbarth, 2015-08-08:
@Roxy-3DPrintBoard -- But the "coordinates would all be screwy" really applies only if they are "unreasonable".  Where to probe and how to convert those readings into a new matrix are really two separate issues. It is only important that you properly maintain your readings in a known coordinate system and compute your corrections based on the system that you are using at that time. Then you must apply to adjustment properly.

The present methodology is computing a correction directly from object space to frame space in one step. It is just as valid to work in some "adjusted" space as long as you realize that the correction that you are computing is from that space to the frame space. Then, instead of replacing the correction matrix with the computed correction, you concatenate the new correction with the previously used transformation.
Think of the current "replacing" as first concatenating with an identity transformation.

AnHardt, 2015-08-08:
For now it seems to be sufficient to store the actual rotation matrix at the beginning of a G28, set the  identity, do the homing and restore the rotation matrix.

While diving a bit into the delta code i now understand Richard better and better. Even the rotation matrix is nether calculated nor used in the delta code. What a mess.

Roxy-3DPrintBoard, 2015-08-08:
> But the "coordinates would all be screwy" really applies only if they are "unreasonable". Where to probe and how to convert those readings into a new matrix are really two separate issues. It is [...]

Yes.  I understand your point.  I've been thinking some more about this.  Other than the fact the X & Y would get offset a small amount from where the probe was told to go, the printer does know what the uncorrected coordinates are.  That is what it tracks.  When it tells the probe to go down, and it touches something, it is going to be the uncorrected coordinates that get read.     

So, I think G29 should not be 'correcting' and offsetting the X & Y locations of where it probes, probably, it doesn't really matter much.

Wackerbarth, 2015-08-08:
If all you are doing is planar bed leveling, it is easy to show (mathematically) that it doesn't matter WHERE you probe, but only that you know the correct distance between the probing points. Probing with a linear offset just adds a constant to every reading. When you take the differences to determine the slope, that constant, whatever it may be, gets cancelled out.

thinkyhead, 2015-08-12:
> I think G29 should not be 'correcting' and offsetting the X & Y locations of where it probes

That's correct. During `G29` the movements in `XY` should be uncorrected, and not also move `Z`. If matrix correction was left enabled across `G29` invocations, the effect of doing a `G28\nG29\nG29` would be strange indeed. The second time through, the result would be that the bed is (just about) perfectly level. That's just due to the way it's implemented. If the existing correction was taken into effect, then it would just add onto the existing correction, which –if there was perfect fidelity in probes– would give "no correction" after the first `G29.` It's just simpler right now to clear the correction matrix at the start of `G29` and use uncorrected (model=machine) coordinates rather than rewrite these routines to always use the stepper (machine) positions (because model≠machine).

filipgoc, 2016-02-24:
For the record, I would love this to proceed all the way to saving the matrix in EEPROM. Even though it seems the feature went to sleep. 

---
@paulusjacobus do you somewhere have the code for throwing away of the matrix resetting in G28 you used (still use)? I can't find it in your fork, and, even the latest dev marlin has no changes along these lines. 

Issue is I tried to just comment our the whole reset matrix statement, but it messes up my G28! After G28 and G29, subsequent G28 reverses some homing directions so it actually can't home. Curious if you did something to counter this problem... Thanks!

(Posting this publicly here in case more people want to do this on their own and find this post.)

AnHardt, 2016-02-25:
It's sleeping because to few people are interested in.

Until the bed is hot again the next G29 (grid, 3-point) is always done.
M140 ...
M104 ...
G28
G29
M109 ...
M190 ...

Homing with a uninitialized matrix is crap. Save the matrix somewhere else - set unity matrix - do G28 - restore the matrix.
Example code in https://github.com/AnHardt/Marlin/pull/26 (untested!)

paulusjacobus, 2016-02-25:
> Example code in AnHardt/Marlin#26

@FilipGoc and @AnHardt

I agree, from my experience saving it to eeprom and reload it would only
work for a very stable and predictable bed. My Mendel is just all over the
place and I need to re-do the bed probing for each print. Yes, it takes
time and I have learned to live with it.

paulusjacobus, 2016-02-25:
Note: I actually warm up the bed first and then do the autoleveling to
prevent warping after the auto bed leveling

AnHardt, 2016-02-25:
One reason more to not store the matrix in the EEPROM. ;-)
My matrix changes a bit when i'm removing very sticky parts from the bed.

thinkyhead, 2016-03-02:
If anyone is interested in implementing such a feature yourselves, take a look at the way that `MESH_BED_LEVELING` is saving its data in `configuration_store.cpp`. The `MESH_BED_LEVELING` implementation has the advantage that is preserves its Z probe values. Note that mesh compensation can also be turned on and off with `M420`, unlike the auto bed leveling, and when you do so the current coordinates change.

filipgoc, 2016-03-02:
@thinkyhead very useful. thanks.

closed 2016-05-04 by jbrazio