MarlinFirmware/MarlinDev #29: Wrong Z homing position

boelle, 2015-08-03:
Org Link: https://github.com/MarlinFirmware/Marlin/issues/1761

thinkyhead, 2015-08-10:
Based on August 8 code:
https://github.com/thinkyhead/MarlinDev/tree/debug_G29

Wackerbarth, 2015-08-11:
@thinkyhead -- My PR #54 wraps your additions in a debug selector to enable or disable their output at run time. It then wraps each fragment with a compile time feature selection so that it can be omitted from the firmware.

I anticipate that "tracing" these values is going to be a much needed capability for some time as we attempt to deploy the code in the field.  Therefore, to avoid the effort required to maintain it as a patch to the ever-changing development, I suggest that we go on and accept PR#54'and encourage those encountering difficulty to enable the feature.

thinkyhead, 2015-08-12:
@Wackerbarth I suppose. The reason I don't like it as a standard part of the code-base is because:

- it adds a mess of long strings;
- people will need to recompile anyway to enable it;
- `M111` becomes redundant if you're already recompiling to get the feature; and
- users are best-off disabling and re-flashing without it once all their leveling woes abate.

If you still very much want to add this as an `M111` option, I've added a few more helpful debug messages that you might want to add into your branch – to see when the probe/endstop is being enabled/disabled and eliminate the switch being off as the cause of a failed probe.

Wackerbarth, 2015-08-12:
@thinkyhead -- I think that you are missing the entire point of including it in the code base.
1) Yes, if someone wants it, they will need to enable the feature and compile their firmware.
However, having the code in the code base, even though it is normally disabled allows us to keep it "current" and make necessary changes to the debugging code at the same time that we make changes to the executable code. (It helps to reduce the "stale patch" issue that is bound to occur because you haven't updated your patch to keep it in sync with the code.

2) The M111 option is there because I don't want to keep swapping firmware just to do some debug run.
I have enough FLASH ROM to include the code, but I don't want to refresh just to be able to actually use the printer.

When the leveling issues are fully resolved, then I would be happy to see it removed (A very easy operation)  However, as long as the community is having leveling issues, there will be a need to compare the behavior on one machine to that on another. So, even if I think that my leveling issues are solved, I may need to enable the output for a test of some condition that is affecting others.

closed 2016-05-04 by jbrazio