MarlinFirmware/MarlinDev #32: Is this code usable for printing?

paulnd, 2015-08-03:
I'm grateful for the work that going into resolving the ABL bugs but is this dev code useable for printing? With usual caveats re development code etc of course

boelle, 2015-08-03:
it should be.... but not sure... it can fail. only a test will tell :-D

Sniffle, 2015-08-04:
@paulnd ABL for cartesian printers is currently working, I just finished doing extensive repeatability testing on it.  That being said any other problems you may run into i can't speak to.  OH! there is currently an SD init bug that if you don't have an SD card in your display it WILL cause the printer to slow to a crawl due to overhead

Wackerbarth, 2015-08-04:
I just push a fix for the SD card problem.

Sniffle, 2015-08-04:
With that being said the answer is for the most part yes :-P

paulnd, 2015-08-04:
Thanks, I'll give it a go

closed 2016-05-04 by jbrazio