MarlinFirmware/MarlinDev #35: Two observed stepper signal discrepancies during diagonal moves

boelle, 2015-08-03:
org link: https://github.com/MarlinFirmware/Marlin/issues/2370

maicrodrop, 2015-08-05:
Small update:  Resetting TCNT1 to 0, if it's already passed OCR1A, works so far; been printing and testing with it.  Have also tested setting TCNT1 to (0.75*OCR1A) which works as well, though haven't tested it as thoroughly yet; this would be preferable, as it minimizes the dead time between ISR calls.

I've also been investigating the 1.5.6>1.5.7 compiler change further.  I added code to the ISR, to use the fastio.h code to write a pin HIGH at the beginning and LOW at the end of the ISR.  While this does change the absolute execution time of the ISR, compiling the same changed code in 1.5.6 and 1.5.7 at least gives a (theoretical) good relative comparison.  I found that an individual STEP pulse takes ~3.5µS in 1.5.5, and ~4.5µS in 1.5.7.  The entire ISR (measured between when the "ISR_signal" pin [defined in pins.h, so WRITE is used in the normal way] goes HIGH and then when it goes LOW) takes ~86µS in 1.5.5, and ~108µS in 1.5.7.  Still doesn't narrow down _why_, but at least shows that there is a definite execution time difference.

The four images below show the 'scope shots listed above; there are four-step-set captures compiled in 1.5.5 and 1.5.7, and single step zoom in both as well.  Yellow is Channel 1/X STEP; Blue is Channel2/Y STEP; Purple is Channel3/"ISR_signal", in these shots HIGH while the ISR is running (set HIGH when it starts, then set LOW at the very end).
[image]
[image]
[image]
[image]

One final note on testing; I did also experiment with having the stepper interrupt write the timer and compare values to the serial port if TCNT1 ran past OCR1A; I saw quite a few instances of the timer being at 212, when it should have triggered the interrupt when it was 203, showing that the overrun is pretty small (I believe those numbers pretty closely match µS, so the ISR is overrunning when it should get triggered again by ~9µS, in those instances).  Again, this test does effect the ISR execution time, but at least does provide _some_ insight.

As always, let me know if you'd like more information, if I wasn't clear somewhere, or if you notice any egregious logic errors.

thinkyhead, 2015-08-11:
Interesting results! I wonder if the compile flags are different between 1.5.5 and 1.5.7. Maybe the better performance of 1.5.5 can be restored by using a different optimization level. Or, perhaps this is due to library changes that we can't affect.

Wackerbarth, 2015-08-11:
@thinkyhead -- Actually, if we go to IDE 1.6 (current), we CAN control things and use it to build from the core/ library/ compiler of our choice.

The Arduino IDE now includes a library distribution system to automatically offer upgraded components.
By specifying our own boards and cores, and even downloading archives of the older compilers, we can selectively mix and match each of the parts.

oysteinkrog, 2015-08-16:
This sounds similar to an issue I had last year:
https://plus.google.com/+%C3%98ysteinKrog/posts/FPpTKnrfE27
The reason I kept scrolling up/down in the menus is that I suspected that the LCD cpu load would help trigger it.

guyzmo, 2015-08-20:
Hi, 

for the note I went into the same issue on latest marlin with a board running at 1/16th microstepping (657 steps/mm for E drive).

I found out the extruder drive was stuttering starting at a feedrate of 913mm/s, and I could notice it as well on high speed X/Y moves, but I did not take data for those.

The gap is consistent being 32ms for every feedrate:

[image]

And the gap is happening when when the pulsing period is reaching 200µs:

[image]

I have added a global that takes the minimum calculated timer value, and when the gap is happening, I had a timer value of 202.

Because of the 32ms period we felt it was likely to be a race condition between `OCR1A` and `OVF`, as 1/2000000*65535=32ms. So a google search on *OCR1A race condition* led to the makergear forum post and then here.

The `if (TCNT1 >= OCR1A) TCNT1 = 0;` at the end of the `ISR` works great (with a negligeable delay visible on the scope), though I wonder if there wouldn't be room of optimization to avoid running into this issue.

HTH

Wurstnase, 2015-08-20:
I've done something similar in my Due-Fork.

```
uint16_t counter = TCNT1 + 10; // 10 or whatever you want for a safe spot
if (counter >= OCR1A) OCR1A = counter;
```

Wurstnase, 2015-08-28:
Look like that ```OCR1A = (OCR1A < (TCNT1 +16)) ? (TCNT1 + 16) : OCR1A;``` solve this issue.
http://forums.reprap.org/read.php?397,542552,549268#msg-549268

guyzmo, 2015-08-30:
@Wurstnase do you have any rationale behind the `TCNT+16`? 
And will we catch it faster when that happens, than with the `TCNT1=0` fix?

I'm not really convinced this hack is a better one than the previous fix…

Wurstnase, 2015-08-30:
> @Wurstnase do you have any rationale behind the TCNT+16? 
Not really. 8 μs just looks good for it.
>And will we catch it faster when that happens, than with the TCNT1=0 fix?
Yes. With TCNT=0 you reset it and it starts again.

thinkyhead, 2015-09-01:
> 8μs just looks good for it.
@Wurstnase Naive question: Is it possible to use `F_CPU` in some way instead of `16` explicitly?

Wurstnase, 2015-09-01:
I think we need fixed ticks. Not time.

thinkyhead, 2015-09-03:
@Wurstnase What is meant by "fixed ticks"? I could use a primer to understand what is being done in your solution, and even to understand what `OCR1A` and `TCNT1` mean. Can you explain to me as if I were a small child? (Because in this matter, I truly am…)

Wurstnase, 2015-09-03:
https://github.com/MarlinFirmware/Marlin/issues/2525#issuecomment-136880938

thinkyhead, 2015-09-04:
Haha, thanks! I just saw that other comment too. :tongue:

guyzmo, 2015-09-10:
After trying out both fixes, I pushed the `E` axis to `M203 E999999` and tried to see how it goes as I'm going stupidly fast. As a reminder I used to have an issue at around `F913`, and I used to have the issue double at around `F2200`.

Now it's all fine at `F2000`, `F3000`… up until reaching `F6000` where I had the following issue:

[image]

So it's doing a mini pause between each 4 pulses. Sounds like the original problem, so I did zoom in:

[image]

and the gap is consistently 24ms. I then looked at when the issue starts, and it is at `F5980` it misbehaves at higher speeds, like the following at `F10000`:

[image]

Obviously, it's less an issue than the original one — as it's unlikely people is going to drive extruders at that speed — but it still feels somehow related, even though it's more likely to be an issue with the way `calc_timer()` is doing its job. What concerns me is that it's not really the kind of issues I was expecting by running the steppers way too fast (I was more expecting irregular stepping).

I'm having a look at that, and I'll try to find a proper solution, but I'd be happy to hear your thoughts about it. I might open a new issue on the topic if you believe both issues are not related.

BTW, my steps/mm are `657` on the extruder.

Wurstnase, 2015-09-11:
You can't really go higher than 40.000 steps/s. With 657 steps/mm this is ~60mm/s or 3650mm/min.

closed 2016-05-04 by jbrazio