MarlinFirmware/MarlinDev #37: Error Recovery Problem

Wackerbarth, 2015-08-03:
Because of positional errors, at the end of bed leveling, the attempt to retract the probe failed.

> 12:38:26 PM: N831 M111 S6 *77
< 12:38:26 PM: FIRMWARE_NAME:Marlin 1.1 rkw-163-g1869e90 SOURCE_CODE_URL:https://github.com/Wackerbarth/MarlinDev/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Kossel Pro EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
> 12:38:26 PM: N832 M111 S4 *76
> 12:38:38 PM: N844 G28 X0.00 Y0.00 Z0.00 *94
> 12:38:46 PM: N853 G29 *60
< 12:39:09 PM: -4.09 -4.23 -4.18 -3.94 -4.26 -4.19 -3.67 
< 12:39:09 PM: -3.97 -4.01 -4.05 -4.07 -4.03 -3.74 -3.42 
< 12:39:09 PM: -3.86 -3.89 -3.92 -3.87 -3.80 -3.59 -3.39 
< 12:39:09 PM: -3.66 -3.76 -3.81 -3.77 -3.76 -3.68 -3.38 
< 12:39:09 PM: -3.47 -3.49 -3.51 -3.54 -3.44 -3.31 -3.19 
< 12:39:09 PM: -3.27 -3.28 -3.28 -3.34 -3.21 -3.06 -2.91 
< 12:39:09 PM: -3.08 -3.08 -3.06 -2.93 -2.98 -2.88 -2.68 
< 12:39:14 PM: Error:Z-Probe failed to retract!
< 12:39:14 PM: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
> 12:40:07 PM: N906 M999 *5
< 12:40:07 PM: Resend: 855
< 12:40:08 PM: Error:Line Number is not Last Line Number+1, Last Line: 854
< 12:40:08 PM: Resend: 855
< 12:40:09 PM: Error:Line Number is not Last Line Number+1, Last Line: 854
< 12:40:09 PM: Resend: 855

However, this issue is that I issued the M999 command to resume.
But, it failed because the line numbers are now out-of-sync between the printer and the host driving it.

thinkyhead, 2015-08-04:
> N906 M999 *5

Marlin is way off if it thinks the last line is 854, but it's supposed to be 905. Could it be a host issue?

Wackerbarth, 2015-08-04:
I don't know if you consider it a "host" issue, or not. While Marlin was not responding, the host (Repeater-Host Mac) continued to send queries which went unanswered. RS-274 indicates that N-codes are optional labels and are not REQUIRED to be in any order.

In any case, I think that following the M999 "reset", ANY line number, or at least from the line number associated with the M999, SHOULD be acceptable.

thinkyhead, 2015-08-05:
The `Resend: 855` message is specifically asking the host to resume sending lines starting after the last line it actually received. The host should respond by sending `N855 G1 X12.345 Y123.45`. The log doesn't seem to show what the host is actually sending. I thought perhaps the `N906 M999 *5` command was updating the last line number in Marlin, but if that were the case it wouldn't keep asking for line 855. Do you have any idea what the host is sending each time Marlin responds with `Error:Line Number is not Last Line Number+1, Last Line: 854`?

Wackerbarth, 2015-08-05:
I don't know for sure. But I suspect that the host does not have the capability to go back in time.
The intervening lines were just M105 status requests. It doesn't care that they got dropped.
I think that M999 SHOULD be updating the last line number in Marlin (to 906)

At least to me, M999 is a "System Reset". At that point, past history should go out the window.

thinkyhead, 2015-08-09:
> I think that M999 SHOULD be updating the last line number in Marlin (to 906)

Probably! Does that now happen after #60?

Wackerbarth, 2015-08-09:
Yes, it's fixed.  forgot to close this issue.

closed 2015-08-09 by Wackerbarth