MarlinFirmware/MarlinDev #41: Discussions about cleaning up G28 and G29 code

Roxy-3DPrintBoard, 2015-08-03:
I’m getting ready to try to clean up the G28 and G29 code.   I have seen a number of opinions on how it should be done.  The one thing all of the opinions have in common is they believe the current state of the G28 and G29 code is very ugly.    The purpose of this thread is to get the best thinking out on the table about how this clean-up operation should proceed. 
One idea was to move the Auto Bed Leveling code into libraries.   That might make sense, but much more discussion is needed before I could do that.  For example, would there be multiple libraries?  One for each type of Auto Bed Leveling?    Or would it be one large library with different global names for the different types of Auto Bed Leveling?   And perhaps even more importantly, how would this library get built?   Is this part of the main Marlin build environment and does it get built automatically if some component in the library gets changed?     If anybody thinks this is the right way to proceed, please answer these questions and help us all understand the path forward if we were to go this way!
The solution I’m thinking makes sense (just because it could be done quickly and easily) is to move the G28 and G29 code to a separate file for each type of Auto Bed Leveling.   I’m thinking it would make sense to have a file named ABL_Delta, ABL_Mesh, ABL_3-Point, ABL_Grid, and maybe eventually ABL_Scara and ABL_Core-XY.    The process I’m thinking makes sense is a master file with all of the G28, G29 and support routines gets constructed and those routines get removed from Marlin_main.cpp.   The first step would be to get this included into the compilation of Marlin_main.cpp and everything works exactly as it always has.    
Once that can be done, this file gets replicated and renamed for each flavor of Auto Bed Leveling.   An #ifdef surrounds the entire contents of the file so nothing gets included unless that type of Auto Bed Leveling is turned on.    And then anything within the file that is not associated with what that file is trying to accomplish gets deleted along with the corresponding #ifdef’s that control those features.  (i.e. The ABL_Grid file would have all the #ifdef’s for 3-Point, Delta, Mesh and Scara deleted.)
Ultimately, if we can keep enough commonality in the support routines like probe_pt() and run_z_probe() we can move them back into the main line code.  But initially, they would reside with the Auto Bed Leveling code that supports the particular flavor.  I’m not sure on this point, once again some discussion would be appreciated.   The reason I’m leaning this way is I think different types of Auto Bed Leveling could benefit from having their own flavor of point probing and homing.  
If it makes sense to use libraries as the final solution, doing the separation into separate files will probably still be a step in that direction and be helpful in the final solution.
Opinions, and thoughtful comments are appreciated!

Wackerbarth, 2015-08-03:
@Roxy-3DPrintBoard -- I think that you are on the right track.  Marlin_main NEEDS to have major chunks moved out into a few separate files. G28 and G29 are certainly good candidates. The method that you describe of creating one big cut and then subdividing it is a good approach.

So, the first step needs to be "drawing a ring around that chunk" and describing just how code on one side of the boundary "talks" to code on the other side.

One comment, near the end you suggest moving support routines back into the main code.  Instead, I suggest that you just make them a shared component, still outside the main code, servicing each of your flavors of leveling.

Roxy-3DPrintBoard, 2015-08-03:
> One comment, near the end you suggest moving support routines back into the main code. Instead, I suggest that you just make them a shared component, still outside the main code, servicing each of your flavors of leveling.

Yes, I agree.   And in fact, besides G28 & G29, run_z_probe() is used by G30 and M48.   It has to be available.   So maybe it makes sense to separate the probe support routines out into a generic module first.   And if a particular flavor of Auto Bed Leveling needs its own way of doing things, it can turn off the generic code and implement its own version within its own file?

Sniffle, 2015-08-03:
Im not 100% confident with this language, but can it be approached from the
OO perspective and have modules that input/output etc, along with having
subroutines that can be called from that object so that things like the
servo routines can still be accessed from othe Gcodes because the
instantiated object for ABL can be given a generic name that can still be
called by other gcodes as well as the abl routine.  Which would solve both
issues.  It would just be a matter of having the abl routine instantiated
in the proper place.

Roxy-3DPrintBoard, 2015-08-03:
I'm sure that is possible.  And maybe even desirable.   It would be good to hear other people's comments.    The whole reason I suggested the above approach is it goes a long way to simplifying the code but keeps everything identical to what is there.    

One of the big problems keeping the Auto Bed Leveling from getting fixed for each flavor is the fact that everything is tangled up.   If each flavor could be worked on and altered without affecting anything else, it would be easier to keep things stable.

And in fact, if an Object Orientated approach does make sense, I think my initial reaction would be trying to do it for a single flavor is going to be much easier than trying to get a grand, unified class to work.   If that single flavor could be converted successfully, probably most of the hooks in the rest of the code would be close to correct and make it easier to start converting the other flavors over to Object Orientated.   Please feel free to disagree.   I haven't done much Object Orientated programming.

Sniffle, 2015-08-03:
Im used to java programming so its my automatic go to which is why i
suggested it...

Having a cartesian_abl class with methods that are passed parameters via a
method call inside the class would make this a much easier implementation
because the servos could be initialized inside of the class(if desired) and
called via the class when needed.  Along with all of the other pros and
cons of doing OO such as proper scope(which is basically forced) and with
proper scope comes the ability to control what is accessible from the class
by outside routines so that whatever happens inside the class isnt futzed
with by errant code outside the class

closed 2016-05-04 by jbrazio