MarlinFirmware/MarlinDev #43: SD: Init caught in endless loop when init fails

Sniffle, 2015-08-04:
as the description says SD:init goes into an endless loop when an SD card is not inserted.  

I inserted an SD card to test if the loop continued and it does not.  

This is on the rep rap discount full graphics controller

Wackerbarth, 2015-08-04:
I've run into the same thing.  I narrowed it down to Aug 2 to the interval between 8d3b74c and eb65c57 where all of the SDCARDDETECT stuff was changed.

With no card in my BrainwavePro board, everything seemed fine until then.
But after the change, I get a constant stream of alerts and the system seems to have slowed down because of the extra overhead.

Sniffle, 2015-08-04:
I also experiences the slow down from the extra overhead

Wackerbarth, 2015-08-04:
Fixed!

closed 2015-08-04 by Wackerbarth

thinkyhead, 2015-08-08:
Sorry, I neglected to test this with the SD card removed.