MarlinFirmware/MarlinDev #44: More parse-able M115

thinkyhead, 2015-08-04:
Currently the output of `M115` looks like this:

    FIRMWARE_NAME:Marlin 1.1 abc-123-de4db33f SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel RepRap EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

This is hard to parse. It would help to add a delimiter, such as a semicolon, between parts. This results in…

    FIRMWARE_NAME:Marlin 1.1 abc-123-de4db33f; SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin/; PROTOCOL_VERSION:1.0; MACHINE_TYPE:Mendel RepRap; EXTRUDER_COUNT:1; UUID:00000000-0000-0000-0000-000000000000;

Wackerbarth, 2015-08-04:
And along the same line, how do we want to present the version number?
Is it "1.1.0 RC3-de4db33f" or "1.1.0RC3-de4db33f"?  If we don't allow whitespace (" ") within any of the fields, then that character serves as the delimiter.
It's almost trivial to generate the line in any desired format. But we should (MUST) have a standard.

thinkyhead, 2015-08-08:
> we should (MUST) have a standard

@Wackerbarth Yes, and it would be good to figure out what's useful vs. what's pointless and build on that. For instance, there's only one "protocol" (GCode, I presume, and we haven't defined it) so there's no use currently for `PROTOCOL_VERSION`. Likewise the `MACHINE_TYPE` is just showing the `MACHINE_NAME` (which can have spaces) and that's user-defined. The only use for `M115` is to display the output to the user — there's nothing in it for hosts to use.

Spaces are possible within values, so they aren't the ideal delimiters. The current "delimiter" can be reduced to the regex: `/([A-Z_]+):(.*)([A-Z_]+:|$)/` where `$1` gives the key and `$2` gives the value. Without a regex engine, it gets messy. With a semicolon delimiter the regex becomes the simpler `/([A-Z_]+):([^;]*)(;|$)/`. And, without a regex engine, the string can simply be split – on semicolon, then colon.

More useful information to give the host would be the machine kinematics, the movement limits, the size of the build area, the center coordinates, the positions of endstops, perhaps a list of enabled features, and stuff like that which would allow software to get a concise picture of the machine being queried.

Wackerbarth, 2015-08-08:
M115 --SHOULD be something that is parseable by the Host software and provide useful information. But it also needs to be "readable" by humans.

There is not only one "protocol".  As the set of acceptable codes, etc. change over time, PROTOCOL_VERSION: would be a way to describe the parser to use, etc. Providing versioning information is an essential part of creating something that is reasonably useful to the Host software.

Wackerbarth, 2015-08-08:
It isn't MANDATORY to have a regex parseable string of the form that you describe in order to split out certain key/value pairs.  Think of the keys as "reserved words" just as you are using ";" as a delimiter.
First split on those keywords and then, within each fragment, remove the keyword and strip leading/trailing whitespace. The remainder is the "value".

"More useful information to give the host ..." -- I totally agree. But, perhaps, we should make that into some "query" M-code where the Host asks for specific information (as is done in SNMP, and many other schemes).

M115, and specifically the PROTOCOL_VERSION therein, would tell the Host program HOW to get that additional information.

closed 2016-05-04 by jbrazio