MarlinFirmware/MarlinDev #48: G29 E broken since switch to MarlinDev

Natealus, 2015-08-05:
I've gone over a plethora of variables to try and see what the root cause is. Even tried the new bed leveling code that hasn't been merged yet and same result. It has something to do with the servo system I believe because it just moves to the first spot and engages the servo up and down about 5-10 times, might move a slight bit then goes to the next line as it climbs upwards to a crash with the Y tower. Offsets, number of points, servo tweaks don't seem to be the affecting variables. I've pull fresh copies and rewritten the configuration.h files 2 times to make sure I got all of the changes over. 

Note: I pulled the delta/generic example configs so if there is something not changed there to match the new enable/disable system that could be an issue.

Natealus, 2015-08-05:
It would be really helpful if we could use the suggestion I had in issue #12 about a configuration option to enable debugging output. Especially in regards to ABL.

Natealus, 2015-08-05:
Maybe we should bring some back story into it. Back when I installed the hardware a couple of months ago when delta bed leveling was being implemented...G29 E is the only way I could get the probing array to probe all the way through successfully. If I use M401 to pre-extend the z-probe and just use G29 V4 then it goes down and does 0 to 2 probes successfully then stops on the next probe press and makes a weird sound...its not a stepper skip but it sounds something rough for a second then it stops and gives me a sequence like this.

->G29 V4

<-G29 Auto Bed Leveling
<-Bed X: 46.000 Y: -74.000 Z: 11.775
<-Bed X: 22.000 Y: -74.000 Z: 16.762
<-Bed X: -2.000 Y: -74.000 Z: 19.500
<-Bed X: -26.000 Y: -74.000 Z: 24.487
<-Bed X: -50.000 Y: -74.000 Z: 29.475
<-Bed X: -74.000 Y: -50.000 Z: 34.462
<-Bed X: -50.000 Y: -50.000 Z: 39.450
<-Bed X: -26.000 Y: -50.000 Z: 44.437
<-Bed X: -2.000 Y: -50.000 Z: 49.425
<-Bed X: 22.000 Y: -50.000 Z: 54.412
<-Bed X: 46.000 Y: -50.000 Z: 59.400
<-Bed X: 70.000 Y: -50.000 Z: 64.388
<-Bed X: 94.000 Y: -26.000 Z: 69.375
<-Bed X: 70.000 Y: -26.000 Z: 74.363
<-Bed X: 46.000 Y: -26.000 Z: 79.350
<-Bed X: 22.000 Y: -26.000 Z: 84.338
<-Bed X: -2.000 Y: -26.000 Z: 89.325
<-Bed X: -26.000 Y: -26.000 Z: 94.313
<-Bed X: -50.000 Y: -26.000 Z: 99.213
<-Bed X: -74.000 Y: -26.000 Z: 104.200
<-Bed X: -74.000 Y: -2.000 Z: 109.188
<-Bed X: -50.000 Y: -2.000 Z: 114.175
<-Bed X: -26.000 Y: -2.000 Z: 119.163
<-Bed X: -2.000 Y: -2.000 Z: 124.150
<-Bed X: 22.000 Y: -2.000 Z: 129.138
<-Bed X: 46.000 Y: -2.000 Z: 134.125
<-Bed X: 70.000 Y: -2.000 Z: 139.113
<-Bed X: 94.000 Y: -2.000 Z: 144.100
<-Bed X: 94.000 Y: 22.000 Z: 149.088
<-Bed X: 70.000 Y: 22.000 Z: 154.075
<-Bed X: 46.000 Y: 22.000 Z: 159.063
<-Bed X: 22.000 Y: 22.000 Z: 164.050
<-Bed X: -2.000 Y: 22.000 Z: 169.038
<-Bed X: -26.000 Y: 22.000 Z: 174.025
<-Bed X: -50.000 Y: 22.000 Z: 179.013
<-Bed X: -74.000 Y: 22.000 Z: 184.000
<-Bed X: -74.000 Y: 46.000 Z: 188.988
<-Bed X: -50.000 Y: 46.000 Z: 193.975
<-Bed X: -26.000 Y: 46.000 Z: 198.963
<-Bed X: -2.000 Y: 46.000 Z: 203.950
<-Bed X: 22.000 Y: 46.000 Z: 208.938
<-Bed X: 46.000 Y: 46.000 Z: 213.925
<-Bed X: 70.000 Y: 46.000 Z: 218.913
<-Bed X: 46.000 Y: 70.000 Z: 223.900
<-Bed X: 22.000 Y: 70.000 Z: 228.888
<-Bed X: -2.000 Y: 70.000 Z: 233.875
<-Bed X: -26.000 Y: 70.000 Z: 238.863
<-Bed X: -50.000 Y: 70.000 Z: 243.850
<-Bed X: -26.000 Y: 94.000 Z: 248.838
<-Bed X: -2.000 Y: 94.000 Z: 253.825
<-Bed X: 22.000 Y: 94.000 Z: 258.813
<--83.53 1.26 6.25 -8.71 -23.67 -34.14 -44.60 -44.11 -3.73 
<--43.54 -28.66 16.22 11.24 6.25 3.51 -1.48 21.21 36.17 
<-80.98 21.21 26.20 31.19 36.17 41.16 46.15 51.14 1.35 
<-95.94 90.95 85.96 81.06 76.08 71.09 66.10 61.11 56.13 
<-90.95 95.94 100.93 105.91 110.90 115.89 120.88 125.86 130.85 
<-175.74 170.75 165.76 160.78 155.79 150.80 145.81 140.83 135.84 
<-240.58 175.74 180.73 185.71 190.70 195.69 200.68 205.66 160.78 
<-185.71 200.68 230.60 225.61 220.63 215.64 210.65 250.55 265.51 
<-220.63 275.49 260.53 235.59 240.58 245.56 240.58 235.59 300.43 
<-ok

->G28 X0 Y0 Z0

<-ok
->M114

<-X:0.00 Y:0.00 Z:368.05 E:0.00 Count X: 603.96 Y:603.96 Z:603.96
<-ok

->G29 V4

<-G29 Auto Bed Leveling
<-Bed X: 46.000 Y: -74.000 Z: 11.663
<-Bed X: 22.000 Y: -74.000 Z: 16.650
<-Bed X: -2.000 Y: -74.000 Z: 21.637
<-Bed X: -26.000 Y: -74.000 Z: 26.625
<-Bed X: -50.000 Y: -74.000 Z: 31.612
<-Bed X: -74.000 Y: -50.000 Z: 36.600
<-Bed X: -50.000 Y: -50.000 Z: 41.587
<-Bed X: -26.000 Y: -50.000 Z: 46.575
<-Bed X: -2.000 Y: -50.000 Z: 51.562
<-Bed X: 22.000 Y: -50.000 Z: 56.550
<-Bed X: 46.000 Y: -50.000 Z: 61.537
<-Bed X: 70.000 Y: -50.000 Z: 66.525
<-Bed X: 94.000 Y: -26.000 Z: 71.513
<-Bed X: 70.000 Y: -26.000 Z: 76.500
<-Bed X: 46.000 Y: -26.000 Z: 81.488
<-Bed X: 22.000 Y: -26.000 Z: 86.475
<-Bed X: -2.000 Y: -26.000 Z: 91.463
<-Bed X: -26.000 Y: -26.000 Z: 96.450
<-Bed X: -50.000 Y: -26.000 Z: 101.438
<-Bed X: -74.000 Y: -26.000 Z: 106.425
<-Bed X: -74.000 Y: -2.000 Z: 111.413
<-Bed X: -50.000 Y: -2.000 Z: 116.400
<-Bed X: -26.000 Y: -2.000 Z: 121.388
<-Bed X: -2.000 Y: -2.000 Z: 126.375
<-Bed X: 22.000 Y: -2.000 Z: 131.363
<-Bed X: 46.000 Y: -2.000 Z: 136.350
<-Bed X: 70.000 Y: -2.000 Z: 141.338
<-Bed X: 94.000 Y: -2.000 Z: 146.325
<-Bed X: 94.000 Y: 22.000 Z: 151.313
<-Bed X: 70.000 Y: 22.000 Z: 156.300
<-Bed X: 46.000 Y: 22.000 Z: 161.288
<-Bed X: 22.000 Y: 22.000 Z: 166.275
<-Bed X: -2.000 Y: 22.000 Z: 171.263
<-Bed X: -26.000 Y: 22.000 Z: 176.250
<-Bed X: -50.000 Y: 22.000 Z: 181.238
<-Bed X: -74.000 Y: 22.000 Z: 186.225
<-Bed X: -74.000 Y: 46.000 Z: 191.213
<-Bed X: -50.000 Y: 46.000 Z: 196.200
<-Bed X: -26.000 Y: 46.000 Z: 201.188
<-Bed X: -2.000 Y: 46.000 Z: 206.175
<-Bed X: 22.000 Y: 46.000 Z: 211.163
<-Bed X: 46.000 Y: 46.000 Z: 216.150
<-Bed X: 70.000 Y: 46.000 Z: 221.138
<-Bed X: 46.000 Y: 70.000 Z: 226.125
<-Bed X: 22.000 Y: 70.000 Z: 231.113
<-Bed X: -2.000 Y: 70.000 Z: 236.100
<-Bed X: -26.000 Y: 70.000 Z: 241.088
<-Bed X: -50.000 Y: 70.000 Z: 246.075
<-Bed X: -26.000 Y: 94.000 Z: 251.063
<-Bed X: -2.000 Y: 94.000 Z: 256.050
<-Bed X: 22.000 Y: 94.000 Z: 261.038
<--81.39 3.40 8.39 -6.57 -21.54 -36.50 -51.46 -46.47 -1.59 
<--41.49 -26.53 18.36 13.37 8.39 3.40 -1.59 23.35 38.31 
<-83.20 23.35 28.34 33.32 38.31 43.30 48.29 53.27 3.40 
<-98.16 93.18 88.19 83.20 78.21 73.23 68.24 63.25 58.26 
<-93.18 98.16 103.15 108.14 113.13 118.11 123.10 128.09 133.08 
<-177.96 172.98 167.99 163.00 158.01 153.03 148.04 143.05 138.06 
<-242.80 177.96 182.95 187.94 192.93 197.91 202.90 207.89 163.00 
<-187.94 202.90 232.83 227.84 222.85 217.86 212.88 252.78 267.74 
<-222.85 277.71 262.75 237.81 242.80 247.79 242.80 237.81 302.65 
<-ok

Wackerbarth, 2015-08-05:
@Natealus -- "It would be really helpful ... debugging output"  
Nate, I fully agree.

I just rewrote it as a fully featured debug option.
Try the debug_G29 branch on https://github.com/Wackerbarth/MarlinDev

Natealus, 2015-08-07:
Here's another one with just G29 instead of G29 E. Doing M401 because G29 doesn't extend the Z-probe servo. After I command G29 the effector moves down and goes to the first point as normal then goes down and stops. At this point the probe hadn't moved once from first probe position and just made an odd motor sound from the steppers as the terminal was flashing code all down the screen as if the sequence was proceeding at turbo speed and finished after 5 seconds giving this array. I hit Z+ 0.1 in my control software and it starts climbing almost to top...I didnt want it to crash up there so I reset.

http://pastebin.com/0hJnV52z

Natealus, 2015-08-07:
Here's one using G29 E where it went and probed a few points correctly...would messup on a point then continues in a diagonal Z fashion. Eventually screws z down till the servo gets too low and starts skipping. Manually turned it off.

http://pastebin.com/ycAFSxzG

Natealus, 2015-08-07:
This G29 E leveling session started well for a few points but then started probing ever 1 mm or so and staying more to the 100, -100 (X, Y) general area by the Y tower and the effector moving in odd up and down patterns like it thinks Z is significantly different every probe.

http://pastebin.com/bwXju0D5

thinkyhead, 2015-08-08:
A lot to catch up on here. Will have a look in the clear light of day.

closed 2016-05-04 by jbrazio