MarlinFirmware/MarlinDev #49: Z-height differs depending on whether topography_map is on or off

AnHardt, 2015-08-05:
Topography_map decides about the sequence of the probing points. Ether we have a sawtooth pattern (!zig) or we have a square wave pattern (zig). If AUTO_BED_LEVELING_GRID_POINTS is even the coordinate of the last probing point differs. Result is a different z-height.

Suggestions:
a) Drop the sawtooth pattern. Will result in increasing the code for the output of the topography_map.
b) Drop the square wave pattern. The sum of the ways to travel between the points is shorter here.
c) Add an additional probing point at [Z_SAFE_HOMING_X_POINT, Z_SAFE_HOMING_Y_POINT]. That would improve the likelihood to get the same z-height as with G28 (without parameters and Z_SAFE_HOMING on). Here i have to see how complicated this will be.

Wackerbarth, 2015-08-05:
Why don't we just FIX the tracking of the print head?
There are a number of coordinate systems
For example,
1) The primary extruder nozzle,
2) The probe,
3) Additional Extruders,
4) The machine frame.

If I instruct the machine to move to (0,0,0) those refer to the CURRENT coordinate system.
If I switch Extruders, nothing should happen immediately. But if I again tell the printer to move to (0,0,0), the print head will move to bring the new extruder to the proper location.

At the end of any of these bed adjustment procedures, we don't really care WHERE the print head is located. However, we do need to KNOW where it is.

If we are changing the leveling matrix, then that shifts to relation between the Extruder coordinate system and the frame coordinate system.

Since we are not moving the print head at the moment that we change the matrix, we need to make sure that we are in that invariant coordinate system -- the FRAME.

Be can accomplish that in either of two ways,
(1) ALWAYS track the location in the FRAME reference, or
(2) switch from the current EXTRUDER reference to FRAME, change the matrix, and switch back to EXTRUDER.

Trying to decide WHERE the print head is located, based on the execution point in the code is the cause of much of the difficulty.  Just figure out where you are (during G28), and TRACK each movement thereafter.  The tracker should not care WHY the movement was made.

AnHardt, 2015-08-05:
Nice idea, but i doubt the ___just___ .

Roxy-3DPrintBoard, 2015-08-05:
> Suggestions:
> a) Drop the sawtooth pattern. Will result in increasing the code for the output of the topography_map.

I can confirm that is why the code was written the way it was.   I suspect it would be possible to write nearly as tight of code to take the zig-zag points and output them in a meaningful way, but my guess is that code would be very cryptic.

Wackerbarth, 2015-08-05:
What do you mean? The only time that we KNOW where we are is when we are in contact with some endpoint sensor (limit switch or probe, etc.) 

As far as I can tell, there are only two use cases.
(1) The system has 3 limit switches (or other sensors) which define HOME.
(2) The system has only 2 such switches and uses a probe to align the Z-axis.

In the first case, when we hit the HOME position, we ARE at the FRAME_ORIGIN. Applying the inverse kinematic transform, we have a position in some EXTRUDER system.

When we perform bed leveling, the G29 computation will compute a new matrix which makes the appropriate corrections.

In the second case, we can assume that we are at ANY arbitrary Z-position and, again the results of probing will correct any error in that assumption. (A single point probe is just a special case of the more general computation performed in grid leveling. However, it is allowed to adjust only the Z-axxis translation)

AnHardt, 2015-08-05:
@Wackerbarth 
Mathematically i'm with you, but i have no clue how this will look in actual code.

Wackerbarth, 2015-08-05:
Well, there are only a few places affected.

There are many calls to ```
do_blocking_move_to_xy(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER); // this also updates current_position
``
or similar. So they already handle tracking. So, we need to identify any other calls that move the head WITHOUT updating current_position. Those should also do the update.

Then, as I identified, when we want to change the transform matrix, we need to wrap it with code to preserve the position in FRAME coordinates. That need be done only in the matrix change subroutine.

And, finally, as we hit limit switches, we need to set position in the FRAME coordinates and inverse transform that back to the Extruder current position.

On power reset, we set some appropriate matrix and assume a current position (but it is marked as unknown).

There are three things to consider, physically moving the head, changing the current_position, and changing the matrix.   Just make sure that ALL of those operations go through a VERY limited set of chokepoints.

AnHardt, 2015-08-05:
@Roxy-3DPrintBoard 
A little additional array may help.
 int8_t index_into_AB[xCount][yCount] = probePointCounter++;
Initialized with with -1 even output for deltas (with missing points) is possible.

Roxy-3DPrintBoard, 2015-08-05:
Ah!  I think I see what you are saying...   You are suggesting to remember the coordinates (or X & Y indexes) of each probed point so later it is easy to access any point needed.   Good thinking!   With that put in place, it would make sense to always have the zig zag pattern active.

closed 2015-08-14 by AnHardt