MarlinFirmware/MarlinDev #50: Probe deploy-stove conditions for deltas only work when first and last points are inside DELTA_PROBABLE_RADIUS

AnHardt, 2015-08-05:
```
          #if ENABLED(DELTA)
            // Avoid probing the corners (outside the round or hexagon print surface) on a delta printer.
            float distance_from_center = sqrt(xProbe*xProbe + yProbe*yProbe);
            if (distance_from_center > DELTA_PROBABLE_RADIUS) continue;
          #endif //DELTA

          ProbeAction act;
          if (deploy_probe_for_each_reading) // G29 E - Stow between probes
            act = ProbeDeployAndStow;
          else if (yCount == 0 && xCount == xStart)
            act = ProbeDeploy;
          else if (yCount == auto_bed_leveling_grid_points - 1 && xCount == xStop - xInc)
            act = ProbeStow;
          else
            act = ProbeStay;

          measured_z = probe_pt(xProbe, yProbe, z_before, act, verbose_level);
```
Maybe a decision based on probePointCounter is better. 
ProbeDeploy if probePointCounter == 0
ProbeStow if probePointCounter == auto_bed_leveling_grid_points*auto_bed_leveling_grid_points - skipped_points (+-1). 
With:
```
 if (distance_from_center > DELTA_PROBABLE_RADIUS) { skipped_points++; continue;  }
```

AnHardt, 2015-08-05:
No.
ProbeDeploy works, but ProbeStow is skipped (continue) again.

thinkyhead, 2015-08-08:
The solution is not to deploy on the first point, but on the first _allowed_ point. And the same with the last point. Meaning, it should stow at the end if it didn't already stow. The only "neat" way to do this is to add a new argument to `probe_pt` that indicates whether or not to _really_ probe. If the value is `true` then proceed normally, otherwise skip the `run_probe` part, but do the deploy/stow actions. If the action is `ProbeDeployAndStow` then don't _really_ deploy-and-stow unless also _really_ probing.

```cpp
bool really_probe = true;

#if ENABLED(DELTA)
  // Avoid probing the corners (outside the round or hexagon print surface) on a delta printer.
  float distance_from_center = sqrt(xProbe*xProbe + yProbe*yProbe);
  really_probe = (distance_from_center <= DELTA_PROBABLE_RADIUS);
#endif //DELTA

ProbeAction act;
if (deploy_probe_for_each_reading) // G29 E - Stow between probes
  act = ProbeDeployAndStow;
else if (yCount == 0 && xCount == xStart)
  act = ProbeDeploy;
else if (yCount == auto_bed_leveling_grid_points - 1 && xCount == xStop - xInc)
  act = ProbeStow;
else
  act = ProbeStay;

measured_z = probe_pt(xProbe, yProbe, z_before, act, verbose_level, really_probe);
```

AnHardt, 2015-08-08:
An other solution is to split the 'probepoint-generating-clip-deploy-probe-retract-loop' in two.
The first, complicated (zigzaging) then generates the sequence of the probing points and their coordinates and clips. The second, linear one deploys, probes the points and finally retracts.

thinkyhead, 2015-08-08:
@AnHardt Not a bad idea. Then the second (actual probing) loop can be a single loop instead of a nested one.

thinkyhead, 2015-08-11:
I like that last idea more and more. You could have a `struct` that can describe each individual movement before it even proceeds, then just put them into a list. Raise to Z, move to XY, probe here, raise to Z, stow now, etc... It would make it better for deltas and other round-bed machines too.

closed 2016-05-04 by jbrazio