MarlinFirmware/MarlinDev #51: G30 should use Z_RAISE_BEFORE_PROBING and Z_RAISE_AFTER_PROBING

AnHardt, 2015-08-05:
Maybe we have the wrong wording.
 Z_RAISE_BEFORE_SERVO_DEPLOY and Z_RAISE_BEFORE_SERVO_STOVE
Or more universal
RETRACT_BEFORE_SERVO_DEPLOY and RETRACT_BEFORE_SERVO_STOVE per axis.

Or as the space we need to deploy and to stove is the same
RETRACT_BEFORE_SERVO_ACTION per axis.

thinkyhead, 2015-08-08:
In fact, all we _really_ need is `Z_PROBE_SERVO_CLEARANCE` which gives the amount of lift needed to allow the servo to move without hitting the bed.

And another thing that would really help is to store the probe's current state (deployed or stowed) so that anytime a probe action is done it can automatically see if it needs to zlift-then-deploy, without having to do it anywhere else. Then you just need a `probe_is_complete()` function to do the final lift-and-stow (or sled drop-off, or allen key trick).

AnHardt, 2015-08-08:
While i am in that area. Z_RAISE_BEFORE_HOMING and Z_RAISE_BETWEEN_PROBINGS have the same purpose - to be above the obstacles on the bed (clamps, screws).

@thinkyhead 
Do you have a good name?

Wackerbarth, 2015-08-08:
"Then you just need a probe_is_complete() function to do the final lift-and-stow (or sled drop-off, or allen key trick)." -- Alternately, you could use a `stow_probe()` function that becomes a no-op if the probe is not currently stowed. Then, it would not hurt to call it an extra time at the end of the bed leveling sequence.

AnHardt, 2015-08-08:
Maybe https://github.com/AnHardt/Marlin_O/issues/18 (Probe - Servo - Endstop) gives some enlightenment.

thinkyhead, 2015-08-11:
#72 just applies `Z_RAISE_BEFORE_PROBING` for the time-being. This will be superseded pretty soon by clearance-based defines pretty soon, so in my PR I've just put the code in place for `G30` so that it can be included in these updates shortly.

closed 2016-05-04 by jbrazio