MarlinFirmware/MarlinDev #52: Tree

MoonshineSG, 2015-08-05:
It's clear which one is the Stable Release, but what are the other 3 ? Which on is the most updated, latest crazy codes,  'on the edge' ...

Marlin release
Marlin Development

MarlinDev master
MarlinDev dev

Why is there a need for a separate repo is github provides branches.... ?

Wackerbarth, 2015-08-05:
There are two versions of the repository in an effort to create a separation between the two kinds of development --
The Marlin repository is for those using tagged versions, either Released or Pre-Released.
The MarlinDev (view of the same) repository will be for Very Experimental Development.

As much as anything, having two repositories is to separate the issues related to the "public" version from the very messy "internal".

Within MarlinDev, there are two branches because the GitHub workflow creates a very difficult to follow commit history. The 'master' branch serves as an INBOX allowing Pull Requests to be accepted on GitHub.
The 'dev' branch is just a restatement of those same commits, refiled in to a single orderly stream.
The 'master' branch will be flushed periodically to avoid merging requests that are not based on a recent version of 'dev.'

The 'Development' branch in the Marlin repository has been DEPRECATED and will soon disappear as we flush out the few remaining PR submissions over there.

Once we release a new candidate, we will be accepting PATCHES-ONLY in Pull Requests within the Marlin repository. ALL Feature Requests, etc. will go to MarlinDev.

MoonshineSG, 2015-08-05:
So for someone who used Development from Marlin (didn't mind putting up with bugs here and there as a way to help with testing) which branch under the new repo should pull now ? Dev or Master ?

Wackerbarth, 2015-08-05:
You can use either. Except for the short interval while I process things, if you check, you will find that there are no differences.  But PLEASE make ALL Pull Requests against 'master'.

closed 2016-05-04 by jbrazio