MarlinFirmware/MarlinDev #53: Z_SAFE_HOMING - Z_RAISE_BEFORE_HOMING

AnHardt, 2015-08-05:
What is Z_RAISE_BEFORE_HOMING doing now?
If !Z_SAFE_HOMING we raise Z by this amount before any other move, when ABL is defined (else Z_RAISE_BEFORE_HOMING is not defined).
If Z_SAFE_HOMING homes X an Y. Then drives to [Z_SAFE_HOMING_X_POINT, Z_SAFE_HOMING_Y_POINT, Z_RAISE_BEFORE_HOMING] when Z_HOME_DIR > 0 (not a z-max-endstop). 

What is Z_RAISE_BEFORE_HOMING for?
I'd like to see it as a rise of the nozzle to jump over the obstructs on the bed (clamps, screws). In this case it should be independent from ABL and go to the homing section. Together with Z_SAFE_HOMING it's used in the wrong order. Z should be lifted first, then homing X and Y, then moving to the mid and finally probing Z. Together with Z_SAFE_HOMING (what makes most sense with a z-probe - not endstop) it's used instead of a RETRACT_BEFORE_SERVO_ACTION[Z] (aka Z_RAISE_BEFORE_PROBING). As i do not expect obstructs in the middle of the bed a lift to max(Z_RAISE_BEFORE_HOMING, RETRACT_BEFORE_SERVO_ACTION[Z]) should be sufficient. (remember, we do not now our z-height until now. Moving upwards is dangerous!)

What is Z_SAFE_HOMING doing now?
See above.

What is Z_SAFE_HOMING for?
If we have a z-probe (and neither a z-max-endstop nor a z-min-endstop), we can not home at every X,Y-coordinate. If we have a z-max-endstop or a z-min-endstop we can home everywhere. The probe has to hit the bed. For the middle of the bed it's very unlikely the probe will not hit the bed - therefor this is our default; but we could probe anywhere in the area what is described by ( LEFT_PROBE_BED_POSITION, RIGHT_PROBE_BED_POSITION, FRONT_PROBE_BED_POSITION, BACK_PROBE_BED_POSITION ).

What is needed for Z_SAFE_HOMING?
X and Y must be homed before Z. The probe has to hit the bed. If we have a probe, we have to deploy and stove it.

Has this anything to do with ABL?
No. It has to do with the ability to probe, where in the simplest case the endstops are the probe. In the second simplest case we have a fixed (inductive, captive, FSR, ...) probe, we no not have to deploy. In the most complicated case we have to deploy the probe (servo, sled, allenkey, ...)
Yes, because the ability for probing was invented for ABL. So ABL depends on probing, but not the other direction.

What to do?
Move the definitions for Z_SAFE_HOMING and Z_RAISE_BEFORE_HOMING to the homing section.
Recognize Z_RAISE_BEFORE_HOMING even the when ABL is not defined.
Repair the move to the mid for Z_SAFE_HOMING to not do a lift. But make a separate lift to deploy the probe is needed for that type of probe. (additional lift = ( (Z_RAISE_BEFORE_PROBING - Z_RAISE_BEFORE_HOMING) > 0 ?  Z_RAISE_BEFORE_PROBING - Z_RAISE_BEFORE_HOMING : 0)
Cleanly deploy and stove the z-probe when needed.

thinkyhead, 2015-08-08:
Let's systematically fix this stuff. I'm off to bed now, need to go and get a tetanus shot in the morning…. So some time over this weekend we should make these all make sense.

AnHardt, 2015-08-13:
Thinking a bit further there is only one lift you need. You need that much lift you can operate all possible maneuvers safely. This lift will differ, depending on your type of probe, but for every combination of circumstances there is ___one save minimal lift to make all operations___. I can't find any combination of circumstances where the minimal needed lift could be zero or less.
The remaining special z-heights are:
Zero - nozzle tip touches the bed.
Z_PROBE_OFFSET_FROM_EXTRUDER - difference from Zero to the trigger point of the probe.
A save action height - to operate G28 and G29 without crash (with your kind of probe).

For servo probes i made a little scad-movie (https://www.youtube.com/watch?v=w9RO8eRG2fU).

Wackerbarth, 2015-08-13:
I see a need for two lift numbers.  One for deploy/retract and the other to control the probing motion.
To deploy/stow, we want our minimum height (In extruder coordinates) to be Z_PROBE_OFFSET_FROM_EXTRUDER + Z_PROBE_DEPLOYMENT_CLEARANCE. This is the same as  Z_PROBE_DEPLOYMENT_CLEARANCE in probe coordinates.
When we actually probe, we want to raise to Z_PROBING_HEIGHT in probe coordinates (aka Z_PROBE_OFFSET_FROM_EXTRUDER + Z_PROBING_HEIGHT in extruder coordinates.)

AnHardt, 2015-08-13:
Is there any good reason for not just taking the bigger one? Yes, in one of the cases you can save a few mm of movement - is that worth the trouble.
We could save a lot of code if we'd do all probe activating/deactivating like the Z_PROBE_END_SCRIPT. Of cause we then need a Z_PROBE_START_SCRIPT to. A 3 part sequence (start - mid - end) with a mid repeatable if the probe is still triggerd/untrigerd, would also do the job for allen-key-probes. The aim i see here is to a) unify the probe handling for less complexity and code, b) save some space in the PROGMEM, c) therefore move the complexity to the CPP maybe with a separate `Configuration_Probe.h`.
The complexity for every single type of probe is low, mostly only driving to some coordinates, switching a magnet or servo, testing if the probe is still triggered or not. Everyone could easily make it's  own sequence for activating/deactivating his/her special probe.

Wackerbarth, 2015-08-13:
I fully agree with the idea of treating ALL probes the same with customized micro-coded deploy, probe, and stow sequencing scripts. As for distinguishing between the difference in "PROBING_HEIGHT" and "ADDITIONAL_DEPLOYMENT_CLEARANCE", I have no idea how much they differ.

Particularly if we start splitting up the Configuration into multiple sections, we can easily have comments documenting how to choose the appropriate value in that file. However, we first have to convince Scott to abandon his "everybody has everything in one file" configuration. (IMHO, the sooner the better)

thinkyhead, 2015-08-14:
> we first have to convince Scott to abandon his "everybody has everything in one file" configuration

All I'm concerned with is avoiding disruption in migration from older Marlin for the time being. I have suggested in the past to break up the configuration into several parts by theme, but this has not been a popular idea. I say, go ahead and implement it, then see how much it is loved or hated. Try making the overridable configuration concept at the same time (where only altered options need to be included) and see how that goes over. Yes, my personal preference is still to keep the two self-documenting configuration files — not to split them up by theme, and not to add a `Configuration_overrides.h` (for example) file. My mind will be changed by seeing an actual implementation that people like, but someone with more enthusiasm than I will have to build it.

Wackerbarth, 2015-08-15:
I can equally say that expecting every configuration to include EVERY option is also "not popular".

In fact, you have already "broken" that concept by treating DELTA and SCARA differently.

Utilizing a single unified configuration file (for those who wish to do so) is not incompatible with a structured scheme. Just think of it as a special "custom" case where EVERYTHING is overridden.

The problem is that your insistence that doing it that be "the rule" rather than "an exception" is forcing a burden on the rest of us.

thinkyhead, 2015-08-16:
> you have already "broken" that concept by treating DELTA and SCARA differently

I just kept it as it was, but I would have preferred to keep those sections in all the other configurations too.

thinkyhead, 2015-08-16:
> your insistence that doing it that way

You keep attributing feelings and intentions to me that have nothing to do with me. I have merely kept things as they have been historically to avoid breaking everyone's existing configurations, and only made gradual changes, adding warnings to `SanityCheck.h` for things that have changed. I have not presumed to rewrite the entire configuration scheme, a process which comes with many pitfalls.

Wackerbarth, 2015-08-16:
Yes, I DO attribute intentions to you because you insist that, "for historical reasons", making any change to the scheme which PERMITS a structured scheme is OFF LIMITS and cannot be considered.

What is being proposed will not change anyone's configuration, nor REQUIRE that they change to a revised scheme. 
Anyone who has made ANY modifications to their personal configuration is, de facto, NOT using the "default" anyway. They can still compare theirs to your default and go from there.
The only thing that I really want to change that might affect them is that they would need to look in a slightly different place to find this "default" (de jour) configuration with all of the options expressly presented.

thinkyhead, 2015-08-31:
> you insist that, "for historical reasons", making any change to the scheme which PERMITS a structured scheme is OFF LIMITS and cannot be considered.

I don't recognize myself in this mirror.

My feeling is that for "historical reasons" it's not a good idea to replace the configuration scheme for this release. But of course I am open to an experimental branch where this can be worked out. And once it's mature deploy it. But the work needs to be done first, and since you are so adamant, I think you should build it.

Wackerbarth, 2015-08-31:
Look again, "it's not a good idea to replace the configuration scheme for this release"  I have already presented the concept, I have even attempted to update my presentation a few times to chase after all the changes that you continue to make to the configuration files. But you were unwilling to accept the concept then and I see no point in attempting to keep it "current" until you are ready to consider and discuss it.

thinkyhead, 2015-09-07:
@Wackerbarth Hold on to your concept until such time as is more convenient. Meanwhile, please link in this thread to the branch in your fork where your new configuration scheme is implemented. As time allows someone will try to bring it up to date.

Wackerbarth, 2015-09-07:
@thinkyhead -- Equally, also hold onto all of the suggestions of everyone else, and yourself, so that we can get RC1 out the door and thus make it "more convenient" for this and other changes to the file layout.  

We are wasting far too much time attempting to work around a scheme that is antiquated and that we have outgrown instead of reorganizing things to make it easier to accomplish the functional changes.

closed 2016-04-18 by thinkyhead