MarlinFirmware/MarlinDev #58: Pause Resume print feature request

paulusjacobus, 2015-08-06:
Just post this feature request to see if there is any interest in a Pause Resume feature in order to allow a swap or refill/run out of filament.

Some experimentation has been done in the link below although not ideal:

>http://forums.reprap.org/read.php?2,534549,538584#msg-538584

This post refers to the use of the kill switch which is not intended for that (the user wanted to use the existing LCD control panel which has only one switch so he swapped the functionality. Not something I would do but rather add a auxiliary switch/ through the pin.h config)

Roxy-3DPrintBoard, 2015-08-06:
The M600 command is what is typically used for filament changes.   The big rub is this code depends on having an LCD display to use it.   Probably, this was done as an incentive to get people to buy LCD panels.   The code is very nicely done.

I don't have an LCD panel.  But I made it work by adding a switch and writing a couple of small routines to check if the switch was pressed (to indicate it was safe to proceed).   

I do plan on making this generic code available in the main Marlin branch when the bug fix phase is over.  But if you want to use it sooner, you can pick it up at:  

http://3dprintboard.com/showthread.php?3023-Filament-Change-Command-for-Non-Ultipanel-Users

Probably, you don't have a PrintRBoard.  If that is true, you will have to find a pin to wire up to a switch.  If you have trouble finding a pin to use, I have some code that will help you do that too.

DaveZA, 2015-08-06:
@Roxy-3DPrintBoard 

Hi Roxy. I am the 'author' of the referred code.

Thank You for your comments about my code.

While I agree with your comment about M600, it is pretty much useless if the Pause is needed at a random ( not pre-planned ) point in the print, and the printer is running only from the SD card ( not connected to a PC ).

The code and post was not done to encourage anyone to buy LCD panels - I do not sell hardware, and wrote the code purely for personal use. The only reason for posting the code was to share and give back for all the things I have learnt from the internet that others have shared. Essentially, I had a personal need for this feature and that was the motivation behind the code.

'The code depends on having an LCD' is, imho, a bit of a lame comment. That's like saying that all the code in Marlin related to any form of LCD should not be in Marlin because it 'depends on having an LCD'.

I am glad for you that you wrote your own code to use a seperate switch / button, and also glad that you decided to share that code. Now anyone browsing for a solution of this type has multiple options available.

Roxy-3DPrintBoard, 2015-08-06:
I missed the fact the person wanted to pause the printer at a random time.   I just looked at the code in that post.   That code is very nicely done too!    This is a feature a lot of people can benefit from having.   It would have saved a fair number of my prints if I could have just gotten the printer to pause so I could take care of an issue.

It would work for me the way it is, but I would be tempted to add a #define to control how much the Z axis gets lifted.  Or maybe...  Since I'm pressing a switch, if the switch is still being held when the enqued command gets processed, just keep raising the Z-axis until the switch is released.  (And debounce the switch so it doesn't cause an immediate continue.)  That would let me control how much I need the extruder to retract from the print so I can fix what needs to be fixed. 

I think I would also want a #define to control if the X, Y, and Z axis motors get turned off.   The reason is I won't be able to replace a spool of filament without moving the nozzle by accident.  For me, I would want the X, Y and Z motors holding position while I messed with the extruder.

I like what was done here a lot!   

> 'The code depends on having an LCD' is, imho, a bit of a lame comment. That's like saying that all the code in Marlin related to any form of LCD should not be in Marlin because it 'depends on having an LCD'.

No...  Not lame at all.  It is a fact.   And not everybody has an LCD panel.  So that nice piece of code can't be used by them.   I wonder if we should configure a way for non-LCD panel users to have some input into the firmware's execution.   One approach would be to have some default routines to look for switch presses if an LCD version of the lcd_clicked() routine is not available.   When the bug fix phase is over, it would be good to make progress on this question!

DaveZA, 2015-08-06:
Hi Roxy
The XYZ motors are not disabled during the ParkPause process for the exact reason you mentioned - so they can't ( easily ) be moved and the code is able to return to the original position once resumed.

Like I mentioned before, this was really only coded for my own use, so adding variables for the XYZ movement were not a concern - I knew exactly what position I wanted to go to for the Park. I agree that if this was to become a standard addition to Marlin, that it would need those values set in the config file.

It would be great if someone ( with the technical ability and understanding of the code that I do not have ), could take this further and add the functionality to a separate push button that any Marlin user could benefit from.

Roxy-3DPrintBoard, 2015-08-06:
> Hi Roxy
>The XYZ motors are not disabled during the ParkPause process for the exact reason you mentioned - so they can't ( easily ) be moved and the code is able to return to the original position once resumed.

Oooops!   You are correct!   I was hopping back and forth and missed that I was looking at the wrong version of the kill routine.   My mistake!   

> Like I mentioned before, this was really only coded for my own use, so adding variables for the XYZ movement were not a concern - I knew exactly what position I wanted to go to for the Park. I agree [...]

Nobody is being critical!   You obviously write enough code  to know no matter how nice you do a piece of code, when you show it to the intended audience, they always say "That is real nice, but I need this or that feature."    That seems to be one of the rules of the universe.    I like what you did a lot!  (But I would change it for me!)

> It would be great if someone ( with the technical ability and understanding of the code that I do not have ), could take this further and add the functionality to a separate push button that any Marlin user could benefit from.

Yes.  Agreed.   I already had plans to add a generic lcd_clicked() function so that everybody (with any kind switch) could use the M600 command.   (Obviously, the name would have to be something else because the user doesn't have an LCD panel)    In fact, I had started toying with the idea of letting the user specify one of the limit switches.   For example, if a M600 command happened, maybe there is no reason to wire up another switch.  Maybe the user could just press one of the limit switches to say it is ok to start moving again.

But now with this code, a better thought out answer to that question is probably appropriate.   I think a limit switch 'confirmation' for M600 makes sense (as an option).  But that doesn't make sense to trigger this code when the printer is running.   It would be good to hear some discussion on this topic!

paulusjacobus, 2015-08-07:
@Roxy-3DPrintBoard thanks for all your information, it is extremely useful and it seems that there is a lot of code being developed outside of the Marlin Github site. May be we can have a parking lot (label Release 1.0.4?) with developed items so people don't raise identical features multiple times. I thought these features might be a bit over the top but having seen now that they have already been build makes clear that there is a genuine need for them. I realise that there are just too many web forums and developments to follow to get a good understanding what is out there. (reprap forum, 3dhubs, 3dprinterboard etc) Also the official Marlin wiki seems less informative compared to what others have done (i.e. @Wurstnase love the information there).

DaveZA, 2015-08-08:
@Roxy-3DPrintBoard if you're looking for new features to add to the Marlin code, I have also just added some DS18B20 digital OneWire temperature sensors to my print case, with the temperatures appearing on my Graphical LCD screen ( yes, sorry, that LCD screen again is required :) ).

I have documented the code changes and there is a pic of my display on the reprap post :
http://forums.reprap.org/read.php?1,543447,543665#msg-543665

I am sure that it won't be long before I decide to use the temperature readings to initiate an emergency power-down of the extruder and bed in the event of certain temperate levels being reached.

Also thinking about adding a case heater to control case temperature while printing ABS, and maybe even for controlling the post-print cool-down process to avoid any warping, etc.

Regards
Dave

thinkyhead, 2015-08-12:
I haven't looked to closely at this feature. But, do other implementations move XY away from the print before lifting Z? And does it do a retraction at the same time? The reason I ask is because (a) raising Z first is bad, as it often leaves a wide strand poking out of the top of the print, which can get knocked-into and dislodge the object, whereas a fast move in X / Y will usually only leave a thin hair; and (b) a good amount of retraction (~5mm) to prevent any oozing would be really helpful here too.

DaveZA, 2015-08-12:
Hi thinkyhead

there is a section in the code where you can specify the moves and modify it to your own preferred method : 

target[Z_AXIS]+= 5;        //lift Z by 5mm
        plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], feedrate/60, active_extruder);
        st_synchronize();

        target[X_AXIS]= 5;
        target[Y_AXIS]= 225;
        plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], feedrate/60, active_extruder);
        st_synchronize();

You could swap those around ( so XY before Z ) and add code for the E axis if you wanted.

adam6666, 2016-02-28:
Hi All,

My name is Adam and I am new to all this coding in marlin firmware, but I have been using open source 3d printers for quite some time now. I just have simple thing to ask - How does one implement the Pause function using full graphic lcd by pressing the Pause Menu on LCD. Right Now when I press the LCD Pause menu the printer just stops at the point where it is printing. I want the extruder head to move to a different position and when I press the Resume on LCD it should start printing from the same point where it got paused. I am using RAMPS 1.4 with Arduino mega 2560 and full graphic LCD.

Please help

Thanks

DaveZA, 2016-02-28:
Hi Adam
I did this in July for my i3 with Marlin, Mega 2560 and graphics LCD

I documented the changes I made on this page :+1: http://forums.reprap.org/read.php?2,534549,538584#msg-538584

Please ensure you backup or have your firmware copied before playing with code changes.

Regards
Dave

adam6666, 2016-02-28:
Hi Dave,

Thanks a lot for your quick response

Yes I have gone through your work, I must say nice work. But I do not want
to change the functionality of the "kill" button on LCD to  "PausePark".
What I am looking for is to use the PAUSE menu in LCD which comes up by
clicking the rotary encoder. So that I am randomly able to pause and resume
the prints by using the rotary encoder without changing the functionality
of the "Kill" button.

Is it possible?

Regards
Adam

DaveZA, 2016-02-28:
Hi Adam

I am sure it must be possible, but it has been a while since I last looked in to the Marlin code, so unfortunately I wouldn't be the person to tell you how / where to do it.

From a personal experience, I would just say that when things go wrong, and I need to park/pause, I need to do it quickly. I don't want to waste time entering the LCD menu and scrolling down to Pause.

So as a solution that allows you to keep the LCD Kill button, you may want to consider one of the other suggestions of adding the park/pause option linked to a seperate button connected to an input pin on the Mega.

Regards
Dave

adam6666, 2016-02-28:
Hey Mate,

I really appreciate your efforts, I'll surely try out your method and will
also keep looking for the one that I have in mind. In case I come across
any solution will let you know.

Regards
Adam

stratop80, 2016-04-05:
Hi devs, and Dave. 
I was looking for a way to stop a long print and resume it say, next morning, turning the printer and computers completely off, and came across your approach, full of good ideas.  It would be nice if it was  progressing here.

I had sketched a flowchart to clarify myself (no coding, out of my capabilities) this is it:

- On calling Pause, (LCD option, button,...) finish command queue, and store current z height (zstop)
- Move nozzle out of the print and store parked x y z position (xpark, ypark, zpark)
- Edit a copy of the g code that was being printed.  
    - Find the layer where the print paused (zstop).
    - Delete all text in the file before the row  “LAYER:”(zstop)
    - Add heating instructions, priming, M92 E0,… as needed at the beggining, and write file (resume.gcode).
- On restart, if a file named resume.gcode is found in the SD card, ask for resume (y/n), consider Z is homed and set in zpark, (can home X Y if desired), and print.

Just my 2cents. Best regards!

jbrazio, 2016-04-07:
I see two issues:
 - When we remove power from the printer the home position is lost and we cannot home again because we have a bunch of plastic laying out on the building plate.
 - As we cannot home we could assume the "home position" before power loss, but we can't trust this either, as we have removed power from the motors they could have slipped due to external forces (i.e. weight).

stratop80, 2016-04-12:
oops, I understand, jbrazio.
Just guessing: what if , on restart, ( with enough z clearance ), a xy homing could be performed, and a "manual z" resume was proposed from the LCD?, 
anyway, its up on you, devs, it was just a rough idea, it would be nice to be able to "cold resume" a print,  thanks anyway for your answer and congrats for the great work!

stratop80, 2016-04-13:
One more point, if the printer had a z-max endstop and therefore it could be feasible to do a z homing in the positive direction, then the problem could find a solution, isn't it?.
just for the wishlist :-)

closed 2016-05-04 by jbrazio

DaveZA, 2016-05-04:
Hi jbrazio
I see you have closed this issue.
Does that mean that it is not going to be considered for inclusion in the Marlin code ?
Regards

jbrazio, 2016-05-04:
It was closed due to inactivity. Thus we kept track of it using the "inactive" label.

It's something discussed already multiple times, we have no intend to integrate it on the current Marlin but due it's high request we will for sure do it somewhere in the future.

paulusjacobus, 2016-05-05:
@Jbrazio This was just the request. The actual implementation requires
tickets. Just closing the feature request tickets won't give you a
direction what to develop next. I suggest you or someone else creates a
list of requested features where members can cast their vote in the main
page or child page of the wiki. The ones the most votes combined with their
complexity and contributing value  get selected as a candidate for the
development. The dev team takes on the development and release it as a
Feature request release 1.1.x. Testers test it and feedback the bugs. The
dev fixes the bugs and a new formal 1.x++ release comes out

thinkyhead, 2016-05-05:
This one is high on the list. It will not be forgot.

adam6666, 2016-07-16:
Hi Dave

I was trying to do what you have done.

But during compiling the code I am getting this error. I guess it is
probably because of the Version of Marlin which I am using. I am using
Version Of Marlin i.e MARLIN RC6 1.1.0.

Please find attached screenshot of the error.

Your help is greatly appreciated.

Thanks