MarlinFirmware/MarlinDev #59: Mommy, Mommy, Are We There Yet?

Wackerbarth, 2015-08-06:
It appears that everything is, for the most part, now working. (E.g. Bed leveling for most, but perhaps not all, cases)

So, the question is "Should we generate a Release Candidate and post it on the main Marlin repository?"

In any case, we need to get ALL of the edits to the accompanying DOCUMENTATION in order.

Sniffle, 2015-08-06:
the arc code is extremely slow but other than that it is definitely worthy
for cartesian printers

fmalpartida, 2015-08-06:
I would say that we are I a good position to go for a release candidate. Been testing over the weekend and during this week. All good on my end.

clefranc, 2015-08-06:
Never been able to use ABL on my delta, this needs documentation/tutorial.

thinkyhead, 2015-08-07:
@clefranc You're the tester we've been looking for. I'd really prefer to get bed leveling totally solid before RC.

thinkyhead, 2015-08-07:
> arc code is extremely slow

How does it compare if you change the segment size? The default 1mm is pretty small. I can imagine even raising it to 1.5 would improve speed a little.

alexborro, 2015-08-07:
Check these videos about Arc Test.

It is made on a "old"  version of Marlin (check video date) but I suppose
this code has not changed.

CoreXY Arc Test:

http://www.youtube.com/playlist?list=PLuhzZFbGQkipfO_yOf_AehmgFCS7PqFu3

Cheers.

Alex.

thinkyhead, 2015-08-07:
@alexborro Very nice, thanks!

alexborro, 2015-08-07:
I think we should set 2mm as default.
I tried to convince the Simplify3D team to adopt arc moves but they didn't
listen to me. It helps keeping the buffer full and renders better circles.

Wackerbarth, 2015-08-07:
@clefranc -- By that, I assume that you mean that Bed Leveling does not compensate for the "dish" effect that you get because of mechanical deviations from the dimensional parameters that you compiled into your firmware.

Please understand that this procedure is not a calibration of the actuators, but a compensation for the tilt that results from a (theoretically) flat bed which is not seated in the frame absolutely perpendicular to the Z axis.

The current "delta" code does not have enough variables/controls to compensate for mechanical assemblies that are not "spot on".

clefranc, 2015-08-07:
@thinkyhead Will try ABL again on my delta this morning. BTW, I'm sure you can make an ABL tutorial with nice hand drawings (don't worry, if it's ugly I'll tell you).

@Wackerbarth You're right, I use my "customized" firmware that made prints "accurate". I load it when I need it. For testing ABL, I use the "plain" dev branch. I've tried to help debug ABL since my probe was installed, without success. It's only a ~~stupid~~ configuration issue, because it works with Repetier.

Wackerbarth, 2015-08-07:
@clefranc -- So, are you saying that you are unable to get the printer to go through the motions of bed leveling and output a matrix of probe depths? Or are you saying that you are unable to get decent prints, even after running G29?  If the latter, will you please set AUTO_BED_LEVELING_GRID_POINTS to a reasonably large value (like from 7 to 11 depending on your available memory) and publish the resulting probe matrix here.

Natealus, 2015-08-07:
@thinkyhead I've also been testing ABL code on a delta but with a servo for the Z-probe arm. With issues but it seems I'm a special case. 

I ordered an inductive proxy sensor to test and see if my heat bed below some borosilicate glass would trigger the probe. If that were the case I could take the servo out of the equation. 

But it does make me think that it's something in the servo code if its working fine for the majority of other people. I'm available to test it out all weekend I have some days off work. :+1:

Wackerbarth, 2015-08-07:
@thinkyhead @boelle @fmalpartida @AnHardt  and anyone else willing to help.
I've started a temporary branch "RCStaging" in this repository to prepare everything before moving it over to the Marlin repository as "RC". Please make pull requests against that branch for any changes that you would like to see.

The actual code will be cleanly merged from the "dev" branch at the last moment, so if the changes are to that part, please make them in our normal manner on "master".

fmalpartida, 2015-08-07:
I will give it a go over the weekend. Thanks for sharing.

Wackerbarth, 2015-08-27:
@thinkyhead @boelle @fmalpartida @AnHardt @Roxy-3DPrintBoard 

PING -- If we are EVER going to "release" a new version of Marlin, we need to be doing SOMETHING in a coordinated manner.

fmalpartida, 2015-08-27:
Fully agree, just got back from vacations and getting my bearings in the right direction. Back to work too, but...

Roxy-3DPrintBoard, 2015-08-31:
Hi Richard!

Can you tell me what the decision was about doing a real release?    The
thing that concerns me the most about doing a formal release is it seems
the Delta Auto Bed Leveling is not ready.   Do you have that concern also?

Wackerbarth, 2015-08-31:
NOTHING was decided. For the second time in a row, I was the only committer who was in attendance.
Scott seems to have returned on Sunday and made a large number of postings. I haven’t had time to dig through them yet. But I think that he mentioned something about RC1.

As far as I can tell, Delta bed leveling does exactly what it should, with perhaps some exception concerning the way that it terminates if you do not have an allen key or sled to stow.

Unfortunately, it is JUST BED LEVELING, and does not meet the expectations of many users (to CALIBRATE the mechanism).

There exists code to do this calibration, but it has some problems (in addition to being based on VERY old Marlin code).

I have some improved code, but do not want to introduce it into the code base until we straighten out some underlying issues such as converting the file structure into a library.  And those kinds of changes have been “on hold” for a number of months because we were supposedly going to get this release out “real soon” and make said changes in Marlin 1.2.

BuzzBumbleBee has proposed a partial solution. But I am opposed to including his code because it is both incomplete and essentially “yet another wart” rather than being incorporated as a part of any improved organization. Including it at this time will only serve to create an additional burden in two aspects.  First, it only partially addresses the issue. This will cause support issues because even with the code, it will still fail to meet user expectations. Secondly, it adds a mechanism that should be abandoned immediately. Including it now just adds legacy issues.

Therefore, my position is that we should release the code FOR NON-DELTA ONLY and then proceed to do a more comprehensive change which will include delta, corexy, corexz, etc. in a unified structure.

Richard

BuzzBumbleBee, 2015-09-01:
@Wackerbarth let me know where my solution falls short and ill add whats needed, i have some time to play with the delta now my FSR probe implementation works better (side note ... I hope to have this on github soon)

At the moment it allows the users to "calibrate / trim" the rod lengths and delta radius for each tower. The only other things that could be modified are :- 

1) the angular positions of the towers around the bed (so not 120 deg)
2) if the towers are not all parallel to each other

Id think both of those are thing that need to be assumed mechanically correct on the printers.

Roxy-3DPrintBoard, 2015-09-01:
@Wackerbarth  You have mail at:  rkw@dataplex.net

Wackerbarth, 2015-09-02:
@BuzzBumbleBee --
I support the concept of adding additional trim items so that we can properly calibrate our delta printers.
The issue that I have concerning the inclusion of your implementation is that it generates additional conflicts with a more unified approach that has been on hold for months because that code will also affect  cartesian systems and there is a desire to avoid such impact on the present release.

If we include your implementation now, it will just create additional "legacy" issues that will have to be addressed (and the resulting clamor to "support the old way, too").

So, please wait a bit longer, as I have been doing, and let's fix this in Marlin 1.2.

A comment on the two issues that you mention above:
1) To allow variance of the angular position introduces additional degrees of freedom. You DO NOT want to DO THIS because there would no longer be a unique representation of the configuration. In the XY plane, the relative positions of the towers have only three degrees of freedom. There are various ways that you can describe them (for example, you could describe the lengths of the three sides of the triangle that they form, or, since three points describe a circle, you could use the radius of that circle and the two relative angular positions, or, the method that I prefer, you can provide the three corresponding sides of triangles which form a central angle of 120ª. But, the important thing is that you have three, and only three, variables.
2) I don't think that we have the ability to measure "tower skew" unless we have some reference object that allows us to take accurate measurements at multiple elevations.  Therefore, I think that we will have to assume that the towers are parallel.

@Roxy-3DPrintBoard -- I hope that this is a sufficient explanation.

BuzzBumbleBee, 2015-09-02:
@Wackerbarth 

Can we have more information on this "unified" system thats been on hold ? (commits ect). I guess its hard for the rest of us to see whats blocking these changes.

closed 2015-09-23 by Wackerbarth