MarlinFirmware/MarlinDev #61: ABL delta for noobs

clefranc, 2015-08-07:
I need help to configure Marlin for ABL.
I'm using a delta printer.
Wish this will be used to write a tutorial.

**Step 1**: Download and unzip the latest code from https://github.com/MarlinFirmware/MarlinDev/tree/RCStaging.

**Step 2**: Copy my delta configuration from ...\MarlinDev-RCStaging\Marlin\example_configurations\delta  (in my case from biv2.5 folder) to D:\MarlinDev-RCStaging\Marlin.

**Step 3**: Check that the Z_PROBE_PIN pin for the probe is defined in my pin board file (pins_RUMBA.h.)

**Step 4**: Adjust all endstops height (from Repetier ABL tutorial).

**Step 5**: ?

At this point, scrolling from Configuration.h, I stump first to the `Manual Bed Leveling` section.

What is manual bed levelling vs auto?
What does a mesh bed levelling do?

@thinkyhead  @Wackerbarth  What do you propose for this step. I think I need to test the probe with M119.

I'm going to bed (the real one after a night shift), I'll look at your comments when I wake up...

I'll update the OP with your help.

Thank you!

Wackerbarth, 2015-08-07:
Thanks. This is a good start to defining the kind of detail that we need to provide.
Obviously, the actual URL components will change as we move to different branches and repositories.

FYI, at the present time you should get CODE from the "dev" branch. (replace RCStaging with dev)
When we get to the public release candidate, you will use "Marlin" (rather than "MarlinDev") and "RC".

AnHardt, 2015-08-07:
Step 2:
Copying a Chnfiguration.h over an other is in almost all cases an error. Make a .diff from your actual Configuration.h against the original Configuration.h of the branch/version you used. Then try to apply it to the new config. Only that way you will realize if there are relevant changes.
Just copying results in a lot of unwanted headache.

Step 3: (Cartesians) If you do not have a z-m??-endstop, just connect the probe to the z-min-endstop - and you are fine.

Natealus, 2015-08-07:
I think using M119 to check the Z-probe for proper functionality is an important as well when first setting your hardware up. I had an intermittent wiring issue a few weeks ago that was crushing the probe down and I found out the wiring wasn't working 100% of the time when testing with M119. The servo movement was causing it to short.

clefranc, 2015-08-07:
@Wackerbarth Noted. @AnHardt Say I never used Marlin, or I'm replacing a very old release, I need to copy over the files for delta, since the base of Marlin is for cartesian. For a firmware refresh, using my config files can be just fine using a diff tool. @Natealus Checking the probe hardware/electronics with M119 (and other command for servo) before attempting ABL is mandatory.

So, what I've to do to test the probe with M119? What's the minimum `#define` I must use? Right now, there is no ABL configuration set in my config.

I'm not even talking probe offset, servo, speed, or anything, just testing the probe manually.

Any advice?

clefranc, 2015-08-08:
Ok, trying some defines...

`#define MANUAL_BED_LEVELING`

I get this:

```
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp: In function 'void _lcd_level_bed()':
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2020:44: error: 'MESH_NUM_X_POINTS' was not declared in this scope
         int ix = _lcd_level_bed_position % MESH_NUM_X_POINTS,
                                            ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2022:13: error: 'iy' was not declared in this scope
         if (iy & 1) ix = (MESH_NUM_X_POINTS - 1) - ix; // Zig zag
             ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2023:9: error: 'mbl' was not declared in this scope
         mbl.set_z(ix, iy, current_position[Z_AXIS]);
         ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2023:23: error: 'iy' was not declared in this scope
         mbl.set_z(ix, iy, current_position[Z_AXIS]);
                       ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2025:58: error: 'MESH_NUM_Y_POINTS' was not declared in this scope
         if (_lcd_level_bed_position == MESH_NUM_X_POINTS*MESH_NUM_Y_POINTS) {
                                                          ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2026:38: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
           current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                      ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2033:38: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
           current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                      ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp: In function 'void _lcd_level_bed_homing()':
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2056:34: error: 'MESH_HOME_SEARCH_Z' was not declared in this scope
       current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
                                  ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2058:34: error: 'MESH_MIN_X' was not declared in this scope
       current_position[X_AXIS] = MESH_MIN_X;
                                  ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2059:34: error: 'MESH_MIN_Y' was not declared in this scope
       current_position[Y_AXIS] = MESH_MIN_Y;
                                  ^
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp: In function 'void lcd_level_bed()':
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\ultralcd.cpp:2072:5: error: 'mbl' was not declared in this scope
     mbl.reset();
     ^
Error compiling.
```

Looks like I need to define `MESH_BED_LEVELING`

Now I get this:

```
In file included from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Configuration_adv.h:607:0,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Configuration.h:958,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Marlin.h:22,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\blinkm.cpp:5:
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\SanityCheck.h:117:8: error: #error MESH_BED_LEVELING does not yet support DELTA printers.
       #error MESH_BED_LEVELING does not yet support DELTA printers.
        ^
Error compiling.
```

Seem mesh bed leveling is not supported for delta printers.

Dead end...

clefranc, 2015-08-08:
Going for `#define ENABLE_AUTO_BED_LEVELING`

```
In file included from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Configuration_adv.h:607:0,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Configuration.h:958,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\Marlin.h:22,
                 from C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\blinkm.cpp:5:
C:\Users\CHRIST~1\AppData\Local\Temp\build2385303078048021960.tmp\sketch\SanityCheck.h:140:12: error: #error ENABLE_AUTO_BED_LEVELING requires a Z_MIN or Z_PROBE endstop. Z_MIN_PIN or Z_PROBE_PIN must point to a valid hardware pin.
           #error ENABLE_AUTO_BED_LEVELING requires a Z_MIN or Z_PROBE endstop. Z_MIN_PIN or Z_PROBE_PIN must point to a valid hardware pin.
            ^
Error compiling.
```

My `Z_PROBE_PIN` is not valid?

clefranc, 2015-08-08:
Uncommented `#define Z_PROBE_ENDSTOP`

Now it compiles. Uploading..

Trying to home, oups, something wrong, the towers are homing in sequence, X, Y, then Z is not going up, it lower a few millimeters and stop.

Another missing or wrong #define?

clefranc, 2015-08-08:
At least I can now test the probe using M119:

```
11:04:40.547 : Reporting endstop status
11:04:40.548 : x_max: TRIGGERED
11:04:40.549 : y_max: TRIGGERED
11:04:40.549 : z_max: open
11:04:40.550 : z_probe: open
```
```
11:04:37.134 : Reporting endstop status
11:04:37.135 : x_max: TRIGGERED
11:04:37.136 : y_max: TRIGGERED
11:04:37.136 : z_max: open
11:04:37.137 : z_probe: TRIGGERED
```

clefranc, 2015-08-08:
Can someone explain why I can't home properly? A delta is homing all axis at the same time, not in sequence.

clefranc, 2015-08-08:
Enough for this morning, see you later after some sleep.

Wackerbarth, 2015-08-08:
The delta performs homing in two steps. First it moves all of the actuators to their upper endstop position and then it individually aligns each of them.

This is necessary because moving any one actuator without knowing the position of the others can cause it to move to a position that is outside the allowable envelope.  However, we do know that if we move all three in unison, that the head will move only vertically and not make the horizontal position any worse.

Then, as some actuator reaches its endstop, we know that it is "above" the others and we stop moving it until all of the actuators have reached the (almost) same level.

Once we are in that configuration, we know that the horizontal position is near the origin and that we can safely move individual actuators a short distance to properly align them.

clefranc, 2015-08-08:
@Wackerbarth Thank you for the explanation. I should have stated that I've been using a delta for nearly a year, a cartesian for 3 years, so I know the proper behavior for delta/cartesian homing, and this isn't explaining why when activating ABL on my delta, everything is wrong.

clefranc, 2015-08-08:
Breakthrough!
To explain the strange homing behavior using ABL on my delta, lets take a look at my probe:

**PHOTO 1**. This the probe, at rest, `z_probe: TRIGGERED`:
[image]

**PHOTO 2**. This is how I deploy the probe, using a magnet, `z_probe: TRIGGERED`:
[image]

**PHOTO 3**. This is the probe deployed, `z_probe: open`:
[image]

**PHOTO 4**. This is the probe ready for ABL, `z_probe: open`:
[image]

**PHOTO 5**. This is how I retract the probe, using a finger, `z_probe: open`:
[image]

Doing a G28, when the probe is retracted (`z_probe: TRIGGERED`), the homing is wrong; the X tower is homed first, followed by the Y, finally the Z one is not going up but lowered a few millimeters.

Doing a G28, when the probe is deployed (`z_probe: open`), the homing is correct.

Is this normal? What I've to do to correctly home when my probe is retracted?

Wackerbarth, 2015-08-09:
The movement that you describe seems to indicate that when the probe is triggered, it is skipping the first stage of the procedure that I described and you are seeing the motion related to the second phase.

clefranc, 2015-08-09:
@Wackerbarth If it is like you said "second phase", then why the Z tower is not homing? I think the logic of Marlin is broken when the probe is triggered before issuing G28. But I suppose you can't test it with your delta?

Wackerbarth, 2015-08-09:
I have an "Allen Probe".  Mine homes properly whether its switch is open or closed when I start.

I guess the first thing I need to check is to compare you EXACT configuration to mine. Can you commit the exact configuration and post it on your Github account?

Wackerbarth, 2015-08-09:
"I think the logic of Marlin is broken ..." -- I tend to agree that there appears to be a problem.
But without knowing your exact configuration, it is difficult to identify the path through the code.
As I requested earlier, please post a branch that includes your exact configuration so that I might be able to reproduce it.

clefranc, 2015-08-09:
@Wackerbarth https://github.com/clefranc/MarlinDev/tree/ABL_config

clefranc, 2015-08-09:
@Wackerbarth 
> I have an "Allen Probe". Mine homes properly whether its switch is open or closed when I start.

Just by curiosity, when you did the homing test with the probe switch, was the effector at bed level, or at the top of the delta? If your test was done when the effector was near the top, you can have missed the behavior I see on my delta.

AnHardt, 2015-08-09:
As far i remember in the 'allen key code' is a prevention for triggered_when_stoved sensors.

Boelle also had a problem with a triggered_when_stoved sensor on a modified Mendel90 with a servo

Maybe we need a general ignore_probe_sensor_state_when_stoved option.

clefranc, 2015-08-09:
@Wackerbarth Or perhaps a more "general rule"; ignore Z probe state when homing.

Wackerbarth, 2015-08-09:
Yes, I really want to rewrite ALL of this homing/probing code.  We should be able to reduce it to one "checklist" that everyone follows. The differences would only be in HOW you execute the step (e.g.. deploy_probe() ). Some printers will do a "song and dance" to get their probe in position, others may have nothing (or set an enable flag) to do.

But that all needs to wait for the next release cycle. For now, we need to add another wart to patch things into something that does whatever is necessary.

BTW, I noticed that your probe is farther from the nozzle than the reduction in bed radius.  Just to try to eliminate some differences from my configuration, you might try reducing the probe-able radius by 50 and set the number of extruders to 1.  Once we identify the actual cause of the failure, you can return them to their larger values.

clefranc, 2015-08-09:
@Wackerbarth OK, I've reduced the number of extruder to 1.

Figured out the `DELTA_PROBABLE_RADIUS` after getting the G29 message: ?Probe (F)ront position out of range.

You didn't answer my question above:
> Just by curiosity, when you did the homing test with the probe switch, was the effector at bed level, or at the top of the delta? If your test was done when the effector was near the top, you can have missed the behavior I see on my delta.

Wackerbarth, 2015-08-09:
I've done it from many positions, both inside and outside the print volume and everywhere from touching the bed to having one arm pushed up into its top limit switch.  For me, they all work.

clefranc, 2015-08-09:
@Wackerbarth Thank you.
@boelle Do you have any info on this:
> Boelle also had a problem with a triggered_when_stoved sensor on a modified Mendel90 with a servo

clefranc, 2015-08-10:
@Wackerbarth @AnHardt @thinkyhead Finally found a solution in stepper.cpp for homing and probing with my unusual delta probe:

https://github.com/clefranc/MarlinDev/blob/ABL_config/Marlin/stepper.cpp#L452

Can you review it and tell me if it worth a PR?

AnHardt, 2015-08-10:
@clefranc 
I'm quite sure you don't have a homing problem with this patch, but i'm also quite sure G29 will not work anymore as the z-probe will never trigger again (for deltas).

clefranc, 2015-08-10:
@AnHardt My delta is working fine, homing and probing correctly now:

```
20:32:54.539 : G29 Auto Bed Leveling
20:33:51.614 : -0.76 -0.55 -0.56 -0.59 -0.42 -0.68 -0.60 -0.36 0.18
20:33:51.616 : -0.72 -0.73 -0.57 -0.59 -0.65 -0.61 -0.45 -0.08 0.05
20:33:51.619 : -0.41 -0.68 -0.70 -0.59 -0.62 -0.54 -0.34 -0.28 -0.37
20:33:51.621 : -0.46 -0.55 -0.64 -0.67 -0.66 -0.67 -0.61 -0.48 -0.35
20:33:51.624 : -0.66 -0.72 -0.69 -0.70 -0.75 -0.70 -0.53 -0.42 -0.33
20:33:51.626 : -0.58 -0.64 -0.70 -0.77 -0.79 -0.73 -0.69 -0.72 -0.75
20:33:51.629 : -0.59 -0.78 -0.83 -0.85 -0.81 -0.79 -0.73 -0.67 -0.91
20:33:51.631 : -0.86 -0.89 -0.82 -0.87 -0.92 -0.89 -0.92 -0.73 -0.65
20:33:51.634 : -0.95 -0.85 -0.87 -0.89 -0.91 -0.99 -1.07 -1.05 -0.73
```
```
21:21:11.072 : G29 Auto Bed Leveling
21:22:07.898 : -0.80 -0.55 -0.44 -0.43 -0.39 -0.72 -0.64 -0.34 0.26
21:22:07.900 : -0.73 -0.75 -0.57 -0.51 -0.67 -0.63 -0.44 -0.04 0.07
21:22:07.903 : -0.36 -0.68 -0.70 -0.59 -0.62 -0.54 -0.34 -0.26 -0.34
21:22:07.905 : -0.47 -0.55 -0.63 -0.65 -0.66 -0.65 -0.59 -0.48 -0.37
21:22:07.908 : -0.65 -0.68 -0.74 -0.75 -0.70 -0.71 -0.54 -0.44 -0.40
21:22:07.910 : -0.62 -0.65 -0.68 -0.83 -0.76 -0.78 -0.66 -0.69 -0.72
21:22:07.913 : -0.59 -0.79 -0.82 -0.84 -0.80 -0.77 -0.78 -0.64 -0.84
21:22:07.915 : -0.81 -0.81 -0.81 -0.92 -0.89 -0.86 -0.89 -0.78 -0.67
21:22:07.918 : -0.80 -0.83 -1.03 -1.00 -0.97 -0.95 -0.93 -0.92 -0.78
```
@Wackerbarth Can you test my mod on your delta please?

Wackerbarth, 2015-08-10:
"the z-probe will never trigger again (for deltas)." --- True, it will never trigger as a MAX Z endstop.

But he left it as a MIN Z endstop.   In my configuration for an Allen Key, I don't define Z_PROBE_ENDSTOP at all (It just uses Z_MIN_ENDSTOP)

So, the change doesn't affect me one way or the other.
And it looks appropriate to me.

I don't see why ANY probe would ever serve as a MAX endstop.

All of this code really needs a major rewrite. It is too convoluted with variables being overloaded, and redefined on the fly.

AnHardt, 2015-08-10:
Sometimes i'm glad to have been wrong.

clefranc, 2015-08-11:
The saga continue. Now that I can home properly, probe properly, then what?

What's supposed to happen after the last probed point? Is it suppose to go up with `Z_RAISE_AFTER_PROBING`, because in my case, the nozzle sit at the last triggered position. I can move up or down, but I'm pretty sure the `Z_PROBE_OFFSET_FROM_EXTRUDER` is ignored.

Once again seeking for advice. Thank you all!

thinkyhead, 2015-08-11:
@clefranc Looking at the `gcode_G29` code now… It appears there is no final raise after probing on Delta, except in the case of an allen key probe, where it will do the moves to put it away. The final raise for the case of `DELTA` && `ENABLE_AUTO_BED_LEVELING` would go before the lines that read…
```cpp
#if ENABLED(Z_PROBE_SLED)
  dock_sled(true); // dock the probe
#elif ENABLED(Z_PROBE_ALLEN_KEY) //|| SERVO_LEVELING
  stow_z_probe();
#endif
```
The line to add:
```cpp
do_blocking_move_to_z(current_position[Z_AXIS] + Z_RAISE_AFTER_PROBING);
```

Wackerbarth, 2015-08-11:
@thinkyhead -- Please don't do it that way! Instead make it an alternative to the dock/stow. There is no reason for the extra move if we are going to be putting the probe away.

thinkyhead, 2015-08-11:
See my PR #71 which moves towards a better approach – but does note that #63 might already address this.

clefranc, 2015-08-11:
@thinkyhead OK, now there is a move after the last probe. Good!

But now the new 0 is at this `current_position[Z_AXIS] + Z_RAISE_AFTER_PROBING`, I mean I can go up but not down...

thinkyhead, 2015-08-11:
@clefranc So, is the current position before the raise `0 - Z_RAISE_AFTER_PROBING` ?

thinkyhead, 2015-08-11:
@clefranc Although #63 isn't really ready, you should give that a shot and see if it fixes the final raise issue too… unless you already did.

clefranc, 2015-08-11:
@Wackerbarth What's the problem to raise the effector `Z_RAISE_AFTER_PROBING`, then do a probe retract? It'll certainly not damage the probe. If you don't want to raise the probe after probing then just set it to 0 (zero)?

thinkyhead, 2015-08-11:
> What's the problem to raise the effector `Z_RAISE_AFTER_PROBING`, then do a probe retract?

The problem is, it's already supposed to be doing that. So my patch here might only be a bad hack, and a proper fix would be better.

clefranc, 2015-08-11:
@thinkyhead OK, will try #63 later, thanks.

thinkyhead, 2015-08-11:
@clefranc With any luck #63 will be merged soon. I think it's good, just not sure if it deals with the specific issue of the last raise / stow being dropped. If I have it straight… on your machine, it does do the stow now, but it forgets to do the raise beforehand. Is that correct?

clefranc, 2015-08-11:
@thinkyhead I don't have a servo for deploy/retract the probe, it's done manually.

thinkyhead, 2015-08-11:
@clefranc I see. The current setup is probably still making `Z_RAISE_AFTER_PROBING` dependent on the presence of a servo. I think you're correct, that this should be used after any type of probe, regardless of the presence of a servo or not – as long as it is set > 0.

clefranc, 2015-08-20:
Back to nine days ago! The probe stop at the last trigger position.

The LCD indicates 3.24mm in Z. If I try to **lower** the nozzle, **nothing** (probe retracted or not).

If I try to **raise** the nozzle, the effector is going **down**, ramming in the glass. **Not good at all**. Even if I only send up 1mm, the effector go down as crazy.

So, no ABL today for me...

Wackerbarth, 2015-08-20:
@clefranc -- Christian, Scott indicated that he thought it likely that there was an error involving the probe positioning at the end of the leveling sequence. By your comment, are you trying to tell us that things had gotten better but have now reverted?

Also, it would be very useful if you could create a personal branch in your github repository, install your actual Configuration.h, etc. and push the actual files built.  Then I would be able to know exactly what configuration and revisiting of the firmware you are attempting to use.

clefranc, 2015-08-20:
@Wackerbarth Forget my last comment, it was a stupid config error, no more nozzle crash now.

I've uploaded my printer config: https://github.com/clefranc/MarlinDev/tree/my_printer_config, I'm using the latest dev.

I think the suggested fix by @thinkyhead was never merged, at least not in @AnHardt #63.

And yes, I'm back to nine days ago! The probe stop at the last trigger position.

The fact is I only need to probe to be raised at the end of G29 so I can put my finger under the probe to retract it.

thinkyhead, 2015-08-31:
Hi @clefranc I'm back from my camping trip and starting to sort through issues again. How's this looking? Do we seem to need an extra lift at the end of probing? We should talk about it over on #74 (which I'll be revisiting shortly). That PR adds an extra lift just for DELTA. I'm fuzzy on the leveling code after my time away, so we should have a closer look at it and see just where that final lift needs to be, and under what conditions.

clefranc, 2015-09-03:
@thinkyhead Sorry for the late answer, I was here but... still trying to grasp ABL concept, usage, methodology, etc...

Still can't make sense of Z_RAISE_AFTER_PROBING. @Wackerbarth don't want it to interfere with the dock/stow, told him to set it to 0 (zero). What's the purpose of Z_RAISE_AFTER_PROBING again?

Anyway, I just put some start G-code in my slicer:
```
G28 ; home all axes
M0 ; wait for manual probe deploy
G29 ; bed leveling
G1 F6000 X0 Y0 Z200; take up the effector
M0 ; wait for manual probe retract
```

Are you going to finish assemble your @$%?/* delta?

Printing/testing 6776d665b2 right now...

thinkyhead, 2015-09-05:
@clefranc I hope to get my delta together one day, but right now my living situation is simply too unstable and I don't have the space/time to work on it. It will be a while. I'm happy to borrow any delta however!

The purpose of `Z_RAISE_AFTER_PROBING` is to allow an extra raise move after the last probe, but I believe it has only been in effect (so far) for printers that use servo probes, to make room for the servo to rotate before rotating it and stowing the probe. The idea is to make such an option always-on with bed leveling, regardless of the probe type (excepting perhaps the docking sled and allen key probes which do special movements to deploy and stow). My PR #75 implements an idea that was discussed, to get rid of these specific moves and have more general options specifying how much clearance is needed and then allow the code to decide which clearance values to use. Same general idea, and again not limited to servo-based probes. In my view all probes (except inductive or Hall effect) should have some raise between and at the end.

closed 2016-05-27 by jbrazio