MarlinFirmware/MarlinDev #69: Does the MarlinDev branch only compile clean with Arduino v1.6.x  ?

Roxy-3DPrintBoard, 2015-08-10:
I can't seem to compile the MarlinDev branch with anything but Arduion v1.6.x.     The latest release I can actually use is v1.0.5 because I have a PrintRBoard.   The processor in my PrintRBoard isn't supported directly by Arduino, and certainly not by v1.6.x.    

Can somebody give me a few suggestions?

Thanks!

Wackerbarth, 2015-08-10:
Well, other than the possibility of some minor pin differences, your PrintRBoard should be just like my Brainwave or the SAVMkI. (AT90USB1286 processor). I'm doing it all in Arduino IDE 1.6., and have been throughout the life of their 1.6 branch (I was previously on 1.5).

Other than the possible issues with the conflict between the Marlin ISR and the USB Serial causing communications problems, it works just fine.

Are you using the extensions that I have supplied in the ArduinoAddons?

Can you drop in on our [Hangout channel](https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia)? Trying to hash out these details by email has far too much latency.

Roxy-3DPrintBoard, 2015-08-11:
> Well, other than the possibility of some minor pin differences, your PrintRBoard should be just like my Brainwave or the SAVMkI. (AT90USB1286 processor). I'm doing it all in Arduino IDE 1.6., and have been throughout the life of their 1.6 branch (I was previously on 1.5).

In the past, I got the PrintRboard stuff from Lincomatic at : http://blog.lincomatic.com/?p=502    But his stuff doesn't work with the v1.6 of Arduino.     I did find your extensions and I dropped them into my avr directory.   There was a problem doing that.   The platform.txt file made it so Arduino could not find the gcc files.    But reinstalling Arduino and just putting the boards.txt file there gave me Tools / Boards entries for Brainwave and SAVMki.    

I selected each of those and continued to get the error message:

#error Oops! Make sure you have 'Arduino Mega' selected from the 'Tools -> Boards' menu.

So...   I don't know if this is allowed, but I set my board type to Arduino Mega / Mega 256 and the processor type to ATMega1280.    That seems to compile and link OK.  I have no idea if it generating good code.

>Other than the possible issues with the conflict between the Marlin ISR and the USB Serial causing communications problems, it works just fine.

Ok!

> Are you using the extensions that I have supplied in the ArduinoAddons?

Now I am.  In the past, I was using Lincomatic's stuff to get the PrintRboard going.

> Can you drop in on our Hangout channel? Trying to hash out these details by email has far too much latency.

Yes.   But it might be a day or two.   I'm trying to get my machine setup so I can join the Marlin Hangout on Saturday.  The problem is my computer is a ThinkPad T61p that came with Windows XP and I'm running Windows 7 on it.  I have to hunt down a driver for the camera and microphone for it.

Wackerbarth, 2015-08-11:
I also have IRC chat, etc. Anything with faster turn-around is better than just posting messages here.

There are some real problems getting the configurations "right" for various versions of the Arduino IDE (at 1.6.6, they changed the tree structure yet again -- I really wish that they would call the current layout 1.7).

Let's start with the following:
A) Obtain a VERY RECENT Arduino IDE (from their nightly builds, no older than August 1)
B) Use the BleedingEdge branch from my github MarlinDev repository
C) Install the ArduinoAddons -- Be sure to quit and restart the IDE after you make ANY changes to the add ons.
D) Select the "Kossel Pro (HID Bootloader)" board and the "gcc 4.8.1" compiler. Verify that you can compile the same code that I have run on my printer
E) Modify the Configuration to match your hardware.  -- We may need to do some work on the Boards.txt to create an appropriate entry for your class of machines.

NOTE: The version of things in BleedingEdge has a few differences that are not yet incorporated in "master". So carefully examine, and justify to yourself, each change that you make relative to my KosselPro.
F) Depending on the boot loader that you have installed, there may be some issues uploading. But you should be able to take the .hex file that this builds and upload it using the scheme that you were previously using. (Perhaps an old version of avrdude, etc.)

Let me know how it goes and post pull requests over in my repository.

Roxy-3DPrintBoard, 2015-08-14:
@Wackerbarth  Thanks so much for the detailed instructions!  I've been swamped with a million other things and still have not had the chance to make use of your help.  But for sure, I will work through the steps you outlined as soon as I get time to take a breath!

Thanks again...   I'll let you know as soon as I can what happened.

closed 2016-05-04 by jbrazio