MarlinFirmware/MarlinDev #77: Bug in acceleration calculation?

CCS86, 2015-08-12:
Hi guys, I hope this message isn't out of place on here. I'm curious if this pull request had ever been confirmed or denied in your work on Marlin?

https://github.com/Ultimaker/Marlin/pull/7

Thanks!

thinkyhead, 2015-08-12:
The code we currently have in place now has been updated, so that particular PR would have to be reworked if it isn't already applied in a different way. Here's the same section of code in MarlinDev:

```cpp
  // Compute and limit the acceleration rate for the trapezoid generator.  
  float steps_per_mm = block->step_event_count / block->millimeters;
  long bsx = block->steps[X_AXIS], bsy = block->steps[Y_AXIS], bsz = block->steps[Z_AXIS], bse = block->steps[E_AXIS];
  if (bsx == 0 && bsy == 0 && bsz == 0) {
    block->acceleration_st = ceil(retract_acceleration * steps_per_mm); // convert to: acceleration steps/sec^2
  }
  else if (bse == 0) {
    block->acceleration_st = ceil(travel_acceleration * steps_per_mm); // convert to: acceleration steps/sec^2
  }
  else {
    block->acceleration_st = ceil(acceleration * steps_per_mm); // convert to: acceleration steps/sec^2
  }
  // Limit acceleration per axis
  unsigned long acc_st = block->acceleration_st,
                xsteps = axis_steps_per_sqr_second[X_AXIS],
                ysteps = axis_steps_per_sqr_second[Y_AXIS],
                zsteps = axis_steps_per_sqr_second[Z_AXIS],
                esteps = axis_steps_per_sqr_second[E_AXIS];
  if ((float)acc_st * bsx / block->step_event_count > xsteps) acc_st = xsteps;
  if ((float)acc_st * bsy / block->step_event_count > ysteps) acc_st = ysteps;
  if ((float)acc_st * bsz / block->step_event_count > zsteps) acc_st = zsteps;
  if ((float)acc_st * bse / block->step_event_count > esteps) acc_st = esteps;
 
  block->acceleration_st = acc_st;
  block->acceleration = acc_st / steps_per_mm;
  block->acceleration_rate = (long)(acc_st * 16777216.0 / (F_CPU / 8.0));
```

CCS86, 2015-08-12:
Very nice. I look forward to testing it out! It looks like you guys are getting close.

Thank you,
Lars

CCS86, 2015-08-12:
Is the "advance" function in working order? 

Thanks!

thinkyhead, 2015-08-13:
@CCS86 I don't have any confidence in `ADVANCE` at this time. I think we need to re-do it.

CCS86, 2015-08-13:
@thinkyhead 

Thanks for the update Scott.

I feel like extruder pressure management is crucial functionality to achieve precise printing, across a wide range of machines; with different mechanics, motion tuning, and gcode. 

Printing at slower speeds, with aggressive acceleration/jerk mitigates the negative effects when lacking pressure control, but results in slow prints with surface defects. Using conservative acceleration / jerk with higher speed accentuates the need for pressure management, but allows for faster printing, with better surface quality.

Another big potential benefit I see, is the ability to run much smaller retract moves. You no longer have to depressurize the nozzle with the retract move. Instead, you have bled off extruder pressure beforehand, so only a small retract move would be needed (in theory). Smaller retract = less nozzle dwell time = better print quality (especially if retracts occur on an external perimeter).

I'm not much of a coder, but as an ME, have some ideas on modeling the important factors and keeping the algorithm as light as possible. Happy to help where I can on the next attempt.

Lars

alexborro, 2015-08-13:
I have read a lot about Advance implementation and the one in SailFish
works pretty good.. We just need to port it to Marlin.

Cheers.

Alex.

Nandox7, 2015-08-13:
I also started looking to it sometime ago, but  free time isn't much recently.
But I'll have it again soon. :) (i hope)

Hopefully we're getting enough people attention to have more looking into this and fix/implement it properly.

In a different issue # i shared some of my findings and testing results so far.
For example I see no change whatsoever when changing the values despite the debug output calculations seem ok.
The current implementation doesn't seem to be taking and using the calculated values.

I compared similar behaviour with some slicers that already do this in the produced gcode and it's definitely a much welcome feature as it greatly improves specific parts of the printing process.

Wackerbarth, 2015-08-13:
@Nandox7 - "For example I see no change whatsoever when changing the values"  To which values do you refer? Are you referring to "advance" or something else? I do presume that you are referring to a recent version of master (or dev) from this repository.

Nandox7, 2015-08-13:
@Wackerbarth, to the value of K in the ADVANCE feature settings.
I'm using a old version of DEV but don't think it matters as there was no change to that feature so far.

More details discussed in this old issue #130134380

Wackerbarth, 2015-08-13:
As Scott (@thinkyhead) indicated, the implementation of this feature is, at best, questionable.
Hopefully we will be able to address it AFTER our pending release.

For now, I would consider the feature as "under development", "not for production release".

thinkyhead, 2016-02-25:
I see this issue is still alive!

The `ADVANCE` feature is still not well-trusted and needs a lot of testing and refinement to make sure it's following the right idea. I just downloaded Matthew Roberts' custom firmware in which he implements his `ADVANCE` feature and I'm using that for reference. I'd like to gather together all developers and users who are interested and form a little working group to attack this feature and get it working properly.

I have also peeked at the "JKN Advance" algorithm in Repetier, Sailfish, and SmoothieWare, all of which follow similar but divergent approaches. What I'd like to do first is to flowchart all these implementations so we have a best understanding, compare their similarities and differences, and settle on how we want to patch Marlin to capture the best elements of all these implementations.

Note that Slic3r now includes its own handling of Advance. Slic3r may be able to do Advance more smartly than Marlin because the slicer allows you to specify the filament diameter. In contrast, Marlin has only a default filament diameter (`DEFAULT_NOMINAL_FILAMENT_DIA`) and an EEPROM setting that can be changed and saved for "Volumetric Extrusion" (`M200` E units in mm^3), but it has no adjustable setting for filament diameter that would apply to `ADVANCE`. I believe that for `ADVANCE` to work well, it should use an adjustable filament diameter value and not a fixed one, as it now does. I believe this is one way that firmware-based `ADVANCE` needs to be improved if it is to work as well as slicer-based Advance.

Related issues:
- MarlinFirmware/Marlin#1600
- https://github.com/Ultimaker/Marlin/pull/7
-

CCS86, 2016-02-25:
I am super psyched to hear that! 

I really think this feature is crucial to taking print quality to the next level, especially on bowden style machines.

I'm not a great programmer, but am definitely looking forward to brainstorming the best implementation.

thinkyhead, 2016-03-03:
See #3071 for completion of the changes proposed in https://github.com/Ultimaker/Marlin/pull/7

thinkyhead, 2016-03-03:
> definitely looking forward to brainstorming the best implementation

I may decide to focus on that this weekend, but I will keep you posted.

Sebastianv650, 2016-04-02:
Hope it's OK I'm posting here regarding the JKN feature..
I think a working JKN (or similar) implementation could improve "high speed" print quality dramaticaly. "" because high speed is everything with print speed > jerk speed, and with usual jerk values thats all low speed ;)
That's why I'm thinking about how to get it into Marlin. During my resarch I found this issue, glad to see an active person like thinkyhead has this on the list also :)
I had a look at the sailfish implementation here: https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/Motherboard/StepperAccelPlanner.cc#L525
Together with this description: https://github.com/Smoothieware/Smoothieware/issues/645#issuecomment-135102063

The basic math seems to be realy simple (given the fact someone invented it already, of course!). I compared the code to the Marlin one, it looks even quite compatible in some areas. At last compatible enough to give a good basis for someone already in the Stepper control files, which is not true for me at the moment.
Long text for a short question: Is there a reason why a "copy and modify where neccesary" won't work? At the moment, it seems to be to easy compared to the fact nobody tried it in all the years..

thinkyhead, 2016-04-12:
@Sebastianv650 The hook is in place with an added interrupt when `ADVANCE` is enabled. As far as I know, "JKN" uses the same technique, just slightly different maths.

Sebastianv650, 2016-04-13:
I got the advance to work after a little bit of debugging and I also implemented somithing with the calculation method used by JKN. I think I will create a pull request for the implemented advance version because it is always better to have a working feature instead of a broken one.
But I got two conclusions:
.) The result with a calibrated k value is the same for JKN like method and "advance". OK not much a surprise as they both add or subtract esteps / loop. But
.) I'm not happy with the results. I thought I should get a nearly perfect square looking on top of a test cube without bleeding edges, but that's not the case. It's better then without advance, but it could be much better.. I think the method of distributing the advance e-steps equaly through the acceleration and deceleration phase is not good enough. For a deceleration phase for example, I think it would be better to start releasing the pressure slowly at first, but retract much faster during the last decel steps..
I have to do more tests during the next days if I can find some time.

CCS86, 2016-04-13:
Great to hear that you got it up and running!

I wonder whether we could create a simple gcode test to simplify and quantify how well each phase is working. Maybe a single line commanded at a very high print speed, and repeated for some number of layers. Then with calipers you could measure the width near the start and ends to gauge the extrusion vs print head speed mismatch.

Does marlin support logic in gcode? (ie. IF/THEN/GOTO)

alexborro, 2016-04-13:
Hi Sebastian,

I'm temped to work in the "advance" algorithm as well, actually, trying a
port of JKN from sailfish to marlin. Everybody says JNK is wonderful.

Did you understand how JKN works? If yes, I appreciate any simple
explanation. We can work together on this.

Cheers.

Alex.

CONSULitAS, 2016-04-14:
@Sebastianv650 
Great!

>I think I will create a pull request for the implemented advance version because it is always better to have a working feature instead of a broken one.

We would like to see a PR for your fix. 👍 A lot of people are waiting for this, i think.

>I'm not happy with the results. 

Even if it is not the perfect solution, every improvement counts. Could you perhaps post a photo that illustrates the improvement so far?

> For a deceleration phase for example, I think it would be better to start releasing the pressure slowly at first, but retract much faster during the last decel steps..

IMO it is clear, that a linear solution for non linear behavior of the pressure in the nozzle is not perfect. There seems to be a nonlinear factor missing in the maths...

jbrazio, 2016-04-14:
I'm much more keen for this development initiative than for the "JKN" that appeared a couple of days ago.

Thumbs up.

CCS86, 2016-04-14:
@jbrazio I think that "JKN" is just the initials of those who created the advance function, in the sailfish firmware.

It seems well regarded, but I think that if we work on it, we can improve the concept further.

I'm the one who originally derailed my own thread here, but maybe we should create a new issue for it now, since we are gaining some traction.

jbrazio, 2016-04-14:
@CCS86 I do know, I was talking about #3424.

closed 2016-05-04 by jbrazio