MarlinFirmware/MarlinDev #80: Auto parallel X with bed

boelle, 2015-08-15:
Hi all...

I'm still thinking about adding a 2nd Z driver so that marlin can adjust each Z rod so that the X axis is parallel with the bed

to do it manual takes time and have the chance of not being the same each time

how i do it is using the 2 back corners of the bed that are not adjustable anyway (i only adjust the 2 at the front). i then take a 0.05mm steel blade and make the nozzle just grab it at one corner, then i go to the other corner and adjust the Z rod there.... going back and forth until i get the same grip... i then have some pointers on the rod that points to the smooth rod just next to it so i can see when they start to get out of wack

for an automated system i would use the touch probe and just go back and forth until the height is the same.... maybe use the midway as a ref point?

for parameters i think i would just have to define the 2 spots where adjustments are made and maybe the ref point

Wackerbarth, 2015-08-15:
Mechanically, I think that it is a good idea. In the current scheme, you could use the "baby stepping" type of adjustment to establish your desired operating point.

In the next major iteration (1.2 or 2.0), I suspect that we will be adding mechanisms which will make it much easier to describe the adjustment space and calibrate configurations such as this.

boelle, 2015-08-15:
babystepping is while printing right?

i meant this to be a kind of G29 but only with 2 set points and driving the 2 Z stepper drivers up and down until each corner is at the same height

Wackerbarth, 2015-08-15:
Perhaps I don't understand "baby stepping".  I thought that it was intended to be an "off the books" manual adjustment of an actuator (stepper). But that is somewhat an aside.

What you are suggesting is that you modify the hardware so that instead of having a 3-axis (+extruders) configuration, you have a 4-axis system. It is constrained in the sense that you cannot move one Z-axis very far without also moving the other one. This is commonly accomplished by using a common electrical control signal to the two steppers. You propose to separate these drivers and utilize the firmware to move the two in concert similar to the way that COREXY control is performed. A normal Z step would move both steppers at the same time. But as a part of the mechanical registration, you would move one a short distance without moving the other.

boelle, 2015-08-15:
no what i suggest is that i add another stepper driver module that in all cases are driven like the first one

But when i want to make my X axis parallel with the bed the firmware can step them individual so i dont have to spend time on turning the rods by hands

i do have a small board ready to hold that stepper driver... i just have to connect it

and if i remember correct the pins file allready have sections for 2 Z steppers.... 

but i would not know how to code the new function....

Wackerbarth, 2015-08-15:
It's not "in ALL cases ... like the first". During normal printer movements, they do move exactly that same. However, as a part of the mechanical registration, there will be some independent motion. If each has its own registration sensor (endstop), then you will move them a (slightly) different amount from that point to reach the "registration" position. Thereafter, they will continue to move in unison.
If you are doing a manual registration, you will utilize two different commands (or variants of some command).  When adjusting one side, you will use the command that moves both together. But on the other, you will move just that one side.

We code the stepper control in a manner similar to that used for the CORE movements. Namely, each Z-movement produces the same movement on both Z axis steppers. However, we also provide a "trim" which moves only one of them without moving the other or changing the presumed Z location.

Part of the difficulty in describing the action is that the G28 command has been functionally overloaded.
There are two aspects to "homing". One is to perform "mechanical registration" and the other is to "park" the nozzle in some convenient location.

But, as I stated earlier, I believe that it will even easier to implement this as a part of the next iteration.

boelle, 2015-08-15:
"If you are doing a manual registration, you will utilize two different commands (or variants of some command). When adjusting one side, you will use the command that moves both together. But on the other, you will move just that one side."

for manual work i dont use any commands at all

". If each has its own registration sensor (endstop),"

they dont... they should use the touch probe

AnHardt, 2015-08-15:
@boelle 
Please have a look at Z_DUAL_STEPPER_DRIVERS and especially Z_DUAL_ENDSTOPS in Configuration_adv.h

boelle, 2015-08-15:
@AnHardt 

yes but i dont want to use dual endstops

boelle, 2015-08-15:
what i want is to go to 1 point and messure distance... then go to another point and messure distance 

with the touch probe

and then move each Z stepper motor individual so the distance is the same at each point

then redo the check to make sure it really is the same distance

AnHardt, 2015-08-15:
A kind of two-point_auto-z-leveling.
One step further we could correct the x-inclination we get by a G29 to pre-adjust the both z-steppers. With that we could minimize the z-movement for x-moves. That would speed up x-movement - especially for a slow z-axis.

boelle, 2015-08-15:
YES

its the X inclination i talk about.... its that i wanted to do automatic instead of like now where i use a steel blade to adjust the grib the nozzle has.... if the Z's are very much out of sync it can take 5 times back and forth to get it there... and then it might not be precise each time

my whole point with it is that e machine should be as square as level mechanical it can be... auto level is not a tool you should use to correct something that can be done otherwise

also a wrong x inclination can make the Z smooth rod bearings go tighter than they need..

Wackerbarth, 2015-08-15:
You are now assuming that there IS a touch probe. I said "manual registration". By that I meant the situation where there is no probe. In that case, you use commands (either through the host interface or an LCD interface) to reposition the steppers until they are in the position corresponding to the XYZ location of the nozzle in object space.  If you have the electrical means to drive the two Z steppers either in unison or independently, you will use those commands to replace the mechanical turning of an alignment screw.

When you also have a probe, this can be automated. Think of end stops as probes that cannot be moved.

@AnHardt -- Yes, exactly what we need to do.  In the next iteration, I hope that we change the way we handle the transformation from object space (XYZ) to actuator space (ABCD), I think it will greatly simplify the code.  Then all of the adjustments reduce to a vector of transformation parameters which we can store in the configuration rom and can adjust through automated "bed leveling" operations.

boelle, 2015-08-15:
yes i assume that.... to make it user independent

but oh well, i will look once in a while and see if the idea gets implemented

Wackerbarth, 2015-08-15:
@boelle -- As I have said, it will be much easier to do when we organize the code based on a transformation from object space to actuator space instead of all of this special case #ifdef overlaying of the code for different kinematic designs.

boelle, 2015-08-15:
yep, and i will just be patient and wait...

boelle, 2015-08-15:
its not that big of a rush for me.... just a nice to have feature

Wackerbarth, 2015-08-15:
What I was trying to suggest is that if the only electrical difference is the addition of some driver board component, you could go on and wire it in now. Then you replace the mechanical turning of an adjustment screw with a "baby step" command.
When we get better leveling code, you will be ready to use it.

nitewing76, 2015-08-24:
Also, what if instead of Z_DUAL_STEPPER_DRIVERS, there were Z1/Z2-AXIS, Z_DUAL_ENDSTOPS and Z_OPTO_ENCODERS (like the strip on commercial 2D printers) to measure, verify and adjust each axis independently as needed?

closed 2016-05-04 by jbrazio