MarlinFirmware/MarlinDev #87: Z_MIN_PROBE for delta printers

Wackerbarth, 2015-08-16:
#define ENABLE_AUTO_BED_LEVELING
#define Z_MIN_PROBE_ENDSTOP
#define DISABLE_MIN_ENDSTOPS  // At the present time, DELTA printers use Z_MIN_ENDSTOP for probing

Fails on Sanity check.  

Rather than using a Z_MIN_ENDSTOP as the replacement for the probe, this configuration SHOULD work.

Natealus, 2015-08-16:
@Wackerbarth yeah even with those defined it seems to still fail SanityCheck. I'm wondering if the check perhaps is at fault? its looking for 

#if ENABLED(Z_MIN_PROBE_ENDSTOP)
  #if !PIN_EXISTS(Z_PROBE)

In SanityCheck.h, shouldn't that be Z_MIN_PROBE? But when I change that it throws a different error I don't understand.

Wackerbarth, 2015-08-16:
@thinkyhead  is the one who is doing all of this convoluted sanity checking and defining / undefining of options based on pins, etc.   We should allow him to tell us how it SHOULD work in an understandable manner.

Natealus, 2015-08-17:
Well its a good idea to check things out at compile to make sure its not going to break printers. I just think we might have found a false positive.

clefranc, 2015-08-17:
@Wackerbarth @Natealus Mine compile with these options if DISABLE_Z_MIN_PROBE_ENDSTOP is undefined. It took me time to find a configuration that works with my delta, and I don't even scratch the surface of the ABL.

clefranc, 2015-08-17:
@Wackerbarth @Natealus Just downloaded the latest DEV, something wrong with ABL config.

Wackerbarth, 2015-08-17:
@clefranc -  Although you are reporting a problem, I have NO IDEA what you are reporting. Where have you posted the configuration that you are using? What's "wrong"?

"It hurts, make me well."

Natealus, 2015-08-17:
Well now it doesn't move the steppers at all. G29 E just deployed and retracted the servo as many times as it would have probed if it was actually moving the effector at all.

http://pastebin.com/YbtmvN5X

Another test, just homing with G28 and initiating G29 made a sound as if it were activating movement in the steppers but they didn't budge and it only took about a second or two for it to output this lot. Probe never touched anything but we did comment out Z_MIN_PROBE_ENDSTOP and enabled min endstops to compile. But it doesn't probe anything at the moment.

http://pastebin.com/VdukR7Zx

Natealus, 2015-08-17:
Scratch that, I thought simpler and tracked down that the Z-Min was inverted. It now goes down and probes with G29 E... but gives me an odd array as it seems its traveling the effector upward in a diagonal pattern...making each progressive probe in that line farther from the surface. And some of the probes were only a millimeter or less from the previous one and probed faster in those spots, slower in other spots that were moving further up in Z taking longer for it to trigger the switch.

http://pastebin.com/divnvgVP

clefranc, 2015-08-17:
@Wackerbarth @Natealus I've redone PR #65, mainly renaming the Z_PROBE to Z_MIN_PROBE with the current dev. It work now!

Couple of renames omission and macro expansion errors. I'm looking to do a PR, later, if no one has not already done it.

Wackerbarth, 2015-08-17:
@clefranc -- I'm confused. PR #65 was merged about a week ago. Just what are you doing? That isn't in the master as of yesterday?

clefranc, 2015-08-17:
@Wackerbarth No confusion here. Did someone tested PR #65 a week ago? Not me. You?

I'm just going to retire in 4 years, I wish I had the time to check all PR.

Natealus, 2015-08-17:
@Wackerbarth @thinkyhead @AnHardt  To be honest the ABL for Delta hasn't worked correctly since we switched over to MarlinDev. I'm not saying that was the catalyst but something we did just before we switched over might have caused the initial issues with incorrect probing movements. Because a week before the changeover, everything with ABL was working consistently and I was just trying to fine tune it. 

Recently we've changed more things like how the Z-probe is named and such so it might have added to the confusion of the original problem. I think AnHardt made some improvements to the "zig-zag" pattern that got it to move further in the process without skipping servo gears or steps. But still something is off in the process and I think we should glance back about 2 weeks ago when we started changing everything over for overridable options. There were a few changes to servo and leveling around then too that might be of curiosity. Starting at like PR#2542 18 days ago.

@Wackerbarth I think I saw in another post that you've updated the debug branch so I'll give it a roll and see if I can get some more specifics. :)

I have a few errands to do outside of the office today gents but I will try to be here to help and test. :)

nitewing76, 2015-08-19:
Even in non-Delta printer, I cannot pass the SanityCheck in the Dev version I obtained 12Aug, nor the one I downloaded a few hours ago (18Aug 22:45 CST). But it works fine in the Dev version I downloaded 3Aug at 23:30 CST.

SanityCheck failure:
     #if ENABLED(Z_MIN_PROBE_ENDSTOP)
           #if !PIN_EXISTS(Z_PROBE)
             #error You must have a Z_MIN_PROBE_PIN defined in your pins_XXXX.h file if you enable
                 Z_MIN_PROBE_ENDSTOP.
           #endif

Configuration.h
     #define MOTHERBOARD BOARD_RAMPS_13_EFB

     //#define DISABLE_Z_MIN_PROBE_ENDSTOP
     #define Z_MIN_PROBE_ENDSTOP

pins_RAMPS_13.h
     #define Z_MAX_PIN          -1
     //#define Z_MIN_PROBE_PIN    -1

     #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        #define Z_MIN_PROBE_PIN  19
     #endif

  -nor-
     #define Z_MIN_PROBE_PIN    19

     //#if ENABLED(Z_MIN_PROBE_ENDSTOP)
       // #define Z_MIN_PROBE_PIN  19
     //#endif

  -nor-
     #define Z_MIN_PROBE_PIN    19

     #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        #define Z_MIN_PROBE_PIN  19
     #endif

Wackerbarth, 2015-08-19:
There is a SEVERE issue with the renaming of Z_PROBE to Z_MIN_PROBE and it also involves things like PIN_EXISTS(Z_PROBE).

MY question is WHY was this change made in the first place?  In the hardware, aren't all Z-probes, min-probes?  And, actually, aren't all of the probes Z-probes?

If so, rather than calling them Z_MIN_PROBE, couldn't we just say "PROBE" and be done with it?

clefranc, 2015-08-19:
#91

thinkyhead, 2015-08-31:
> @thinkyhead is the one who is doing all of this convoluted sanity checking and defining / undefining of options based on pins, etc.

My commits have consistently moved away from the pre-existing convolution and pre-existing defining/undefining of pins, but I haven't gotten to everything yet. Let's focus on solutions and stop all this weird personality stuff, or I _will_ take my ball and go home. For those of you who haven't yet read it and taken it to heart, here's [http://opensoul.org/99ways/#1](99 Ways to Ruin an Open Source Project).

Wackerbarth, 2015-08-31:
> On Aug 30, 2015, at 11:59 PM, Scott Lahteine <notifications@github.com> wrote:
> @thinkyhead <https://github.com/thinkyhead> is the one who is doing all of this convoluted sanity checking and defining / undefining of options based on pins, etc.
> 
> My commits have consistently moved away from the pre-existing convolution and pre-existing defining/undefining of pins, but I haven't gotten to everything yet.
> 
> 

Scott (@thinkyhead),

I’m not BLAMING you for the convolution. However, I think that you will acknowledge that it exists and you seem to be the only one who has any grasp of that aspect of the code.

Frankly, I have no desire to dig into it. Therefore, you are the appropriate one to do the evaluation of this aspect of the issue.

Richard

closed 2016-05-04 by jbrazio