MarlinFirmware/MarlinDev #89: motors disabled

ruggb, 2015-08-17:
I think this is an old issue but i couldn't find the thread.
Anyway, I thought I didn't like the idea of the motors disabling immediately after a move BUT.
my printer so far has held the position (I haven't printed anything heavy enough and xy movement is very neutral) and I find it very useful to manually micro adjust the Z0 after the print starts.
I am hoping u find a cure for babystepping on a corexy - at least for the Z axis as that is what my prime use would be. Then u can bring back the 60 second delay after commands r finished.

I am using dev 8/0/15 and the quickhome appears to be working flawlessly.
thx

thinkyhead, 2015-08-17:
Will definitely look into polishing these areas and making everything work in a sane manner....

AnHardt, 2015-10-06:
https://github.com/MarlinFirmware/Marlin/issues/2661

closed 2015-10-06 by AnHardt