MarlinFirmware/MarlinDev #90: Firmware Retract?

CCS86, 2015-08-17:
Hi guys,

I recently enabled firmware retraction on my machine (not running this dev branch of Marlin).

It works in the sense that G10/G11 triggers retractions and recoveries, and the FW retraction settings appeared in my EEPROM menu . But despite having a much slower "recovery" feed rate set, both motions appear to be the same speed (fast).

Any thoughts as to why this might happen?

Thanks!

thinkyhead, 2015-08-17:
Could be a bug. I'll look into this over the next couple of days. Probably some simple oversight.

CCS86, 2015-08-17:
Thanks @thinkyhead !

CCS86, 2015-10-20:
@thinkyhead , Did you happen to find the disconnect?

Roxy-3DPrintBoard, 2015-10-21:
ThinkyHead may be slow to answer...  I think he is moving.

thinkyhead, 2016-02-19:
> may be slow to answer

That's possible…

@CCS86 The code for `G10` / `G11` does appear now to set `feedrate` explicitly to the retract or recover rate. Are you still seeing this any more, or have you switched back to slicer retraction?

CCS86, 2016-02-19:
I have been using slicer based retraction for now. I'm looking forward to
trying the new Marlin release, but might have to wait until someone makes
an Ultimaker branch of it.

thinkyhead, 2016-02-21:
> might have to wait until someone makes an Ultimaker branch

@CCS86 Ultimaker is well supported by Marlin since two of its primary authors work for Ultimaker. In fact, Ultimaker used to be the default configuration. What special settings do you need for Ultimaker that are not provided here?

CCS86, 2016-02-21:
Ah, good to know! I couldn't find a list of changes between the master branch and the Ultimaker branch, and haven't had time to try to compare myself.

I've been making Marlin customizations and building / flashing through Arduino for a while now. But the build instructions for the last RC 1.1 I looked at were a little intense.

I'm psyched to try it out though!

thinkyhead, 2016-04-22:
Closing this old issue, since we'll be migrating the `RC` code here after the 1.1.0 release. If firmware retraction is working in `RCBugFix` then we're also good here. If it's not working there, I'm sure someone will tell us!

closed 2016-05-27 by jbrazio