MarlinFirmware/MarlinDev #92: ABL Z Probe and Dual Z Endstops Question/Issue

NarimaanV, 2015-08-18:
Hi all,
     I'm currently using a Makerfarm 10" Prusa i3v running a RAMPS board and I'm trying to get dual-z endstops as well as ABL setup so that both the aligning of the two z motors as well as "leveling" the bed can be automated and save me time before each printing session. What's become apparent to me is that you apparently can't have z-MIN endstops concurrently with a servo-mounted z-probe (somebody please correct me if I'm wrong), so I'd have to mount the z-endstops as MAX endstops. My issue lies in that my z-motors move very slowly and the travel from 0 - 260mm takes a few minutes by itself, so I can't have "home" be at the z-MAX position because it would have to go all the way up to the Z-MAX to then come back down when I want to actually probe. It also seems that Marlin doesn't have the G-Code for "homing to maximum" so there would be no way to actually "home" the z-axis motors to the top endstops if home was at z-MIN. So what's the best way to configure dual-z endstops along with a z-probe in a setup like mine?

RicardoGA, 2015-08-18:
I think you can have dual z endstop homing at min and zprobe at same time just check the pins that you have to conect the endstops. For Z_DUAL_ENDSTOPS the default pin is  (#define Z2_MAX_PIN 36) and for zprobre in ramps the default pin is (#define Z_MIN_PROBE_PIN  32) and also you have to disable Z_SAFE_HOMING.

And in other hand when you want to  home to Max you dont need a special gcode just  have to change in configuration #define Z_HOME_DIR -1 to #define Z_HOME_DIR 1

NarimaanV, 2015-08-18:
@RicardoGA The thing that worries me about setting both dual z-min endstops and a probe is not the wiring, but the physical conflict between trying to use both. By this I mean, if I mount the MIN endstops higher than normal to allow both motors to align before the hotnend could possibly hit the bed, then the issue becomes that when it tries to probe, the x-carriage would have to run through the z-MIN endstops to be able to get the hotness close enough to the bed again. Conversely, if I mount the MIN endstops lower than normal, then the probe could probe the bed without obstacle, but then the hotend would crash into the bed when attempting to trigger the dual z-MIN endstops.

And with homing to MAX, I'm aware I could just reverse the direction, but the annoyance with the actual home being at the MAX is that my slow z motors would take a while to home all the way up then come back down to probe. What I was referring to was to be able to have something like 2 "homes", where one is the actual home and origin (I know they don't have to be the same but in my case they are), and the other home be an option that could only be triggered with a special G-Code (according to the RepRap Wiki, Teacup has a special G-Code G162 which allows you to home to the max). So if that were possible, I could have the traditional G28 home to all the MIN endstops, where for the z it would use the probe, then if I entered the special G-Code, it could home the z to the max purely to align the z-motors (I know this isn't actually possible now but hypothetically speaking). Worst comes to worst I could try and deal with the wait to home but I'm trying to see if there's a better option I could do.

RicardoGA, 2015-08-18:
Ok now i understand what you want to do, i think you could do exactly what you want in easier way by changing the z endstop for a hall sensor switch o r changing the probe for a capacitive sensor, both gives you a few millimeter to avoid mechanical contact.

NarimaanV, 2015-08-19:
@RicardoGA Oh wow I've never thought of hall effect sensors as endstops (totally still a noob at all this). Are they precise and reliable as that function? If so I'd probably go with this one:
https://www.adafruit.com/products/158

And it's accompanying super strong magnet:
https://www.adafruit.com/products/9

Only thing would be figuring out a way to make two holders for the sensors that are almost identical in size, but that shouldn't be too hard.

In response to the idea of using a capacitive probe, do you mean an inductive probe like the one PrintrBot uses on all their bots that have aluminum beds? Because if so, my bed is unfortunately a PCB heater with a sheet of borosilicate glass on top (been trying to find a way to get an aluminum bed for just that purpose but the company that made my printer hasn't come up with one yet and I don't have the money to get a custom piece done).

RicardoGA, 2015-08-19:
@NarimaanV Yes that sensor should work fine, and by using a capacitive sensor as a probe it's similar to the inductive sensor that use PrintrBot but it´s a different type of sensor.

The inductive Sensor detects metal (needs a metal bed) but the capacitive sensor detects different types of materials like plastic, water, glass, etc.

I use a capacitive sensor as a probe with pcb heatbed and 3mm glass,  this is the model of the capacitive sensor i use (LJC18A3-B-Z/AX)  

This sensor is a NPN normally close, the reason why i use a NPN sensor is because I can connect it without using resistor or voltage regulator in the sensor output, You can use a PNP sensor but you have to put a resistor or a voltage regulator in the output

NarimaanV, 2015-08-19:
@RicardoGA Woah! I didn't even know a sensor like that existed! That's awesome! I'll definitely add that to the top of my list of things to get for my 3D printer. Why doesn't everyone use a sensor like that instead of a servo? Have you had good experience with that sensor? Does it provide reliable/repeatable results? Also you said you don't need to add a voltage regulator to it, but according to these two sources, it operates on 6-36V and most boards run on 5V (at least RAMPS in my case does):
http://www.amazon.com/gp/product/B00MCEK8PE/ref=ox_sc_act_title_1?ie=UTF8&psc=1&smid=A35C93BBDX68G6
http://www.newegg.com/Product/Product.aspx?Item=9SIA3XT1DH4786

RicardoGA, 2015-08-19:
@NarimaanV yes the sensor works between 6-36v. the sensor have 3 wires (
1.-blue (Ground)
2.- brown (voltage )
3.- black (Output)
i use ramps and the way i've connected the sensor is:

NPN Connection:

Voltage directly to 12V, 
Ground to ground zmin (or other ground point in ramps)
Output to signal zmin. (or probe pin)

what i mean you don't need a voltage regulator or a resistor is because the sensor is npn in case o a pnp sensor the output need a resistor or a voltage regulator:

PNP connection:

Voltage directly to 12V
Ground to gound zmin (or other ground point)
Output to resistor or voltage regulator to drop 12v to 5v and from the voltage regulator to signal in zmin or probe pin

you notice the connection is different if the sensor is NPN or PNP i prefer NPN because it's simpler.

I,ve been using this sensor since january and its really accurate and has a great repeatability you can see one of my autolevel reports in this link

https://github.com/MarlinFirmware/MarlinDev/pull/42

NarimaanV, 2015-08-19:
@RicardoGA Lol, silly me forgot RAMPS runs on 12V (tired brain was thinking of how Mega and servos run on 5V). That's awesome then, I'll definitely pick up that sensor and probably some hall effect sensors as well since they seem to be easy to work with and there are no mechanical parts that could break or die after a certain number of uses (I'm sure the sensor itself would still die after sometime but probably later than a mechanical switch). I do however look forward to https://github.com/MarlinFirmware/MarlinDev/issues/80 becoming a feature after the next release so I don't even need the dual z-endstops anymore (i.e. the probe would align the motors for me). You sir, have made my mission of completely automated calibration a whole lot easier, and I thank you for that.

On a slight tangent, but kind of relating back to my original post, I'm interested in hearing the opinion from any of the devs like @thinkyhead or @Wackerbarth (not limited to just them, I'd appreciably any dev's input/opinion) regarding the idea of implementing a G-Code that lets you home to MAX or MIN (according to the RepRap wiki, Teacup and MakerBot have a G161 command for homing to MIN and G162 for homing to MAX). It could potentially help anyone else wanted to something similar to me with either the same printer or a similar design who doesn't want to change out their existing probe/endstops or make their z-MAX the home position.

nitewing76, 2015-08-19:
Why do your Z-motors move so slowly? 

I assume that you calibrated your Step-per-mm? Tweaked feed-rate and acceleration settings? Heatsinks and fans on your stepper drivers and possibly your motors also? Edited the feed-rate settings in your host program (Pronterface, Repetier-Host, etc)?

NarimaanV, 2015-08-19:
@nitewing76 Yes I have gotten the steps/mm calculated, it's more an issue of feedrate and finding the sweet spot between so slow that it feels like I'm watching paint dry, to so fast that everything is either getting too hot or homing approaches the endstop too fast. I do have heatsinks on my A4988 drivers and a pretty powerful small PC case fan constantly blowing air across them and the whole RAMPS board to dissipate the heat, and have been considering getting fans for the z-motors as well to be able to turn up the speed without worrying my printer's going to ignite itself. I think one issue lies in that my printer uses M5 threaded rods, as opposed to something nicer like 8mm-pitch trapazoidal lead screws (which I've looked into but the added speed wil probably reduce the torque to such the extent that I'd need to upgrade my z-motors to either stronger ones or geared ones, which opens a whole new can of worms). Another thing I might consider now that I know about using Hall effect sensors as endstops, is to set my z-endstops as Hall effect sensors and turn up the speed since they should detect the incoming x-carriage with some advance notice so even if it goes a few mm further than the signal to stop was sent, it won't actually hit anything. Many things to try, many ideas floating in my head, and I agree that I can provably tune my settings better to get faster z-movement, but the idea of the special G-codes was something I thought might prove useful as well.

closed 2016-05-04 by jbrazio