MarlinFirmware/MarlinDev #101: auto-levelling again (still?)

MoonshineSG, 2015-08-22:
Is auto levelling working for anyone ? 

After a G29, I move the head to X30 Y10 Z0.2 and then send it to Y190 (The points are 2 of the probing points during G29). I can see the Z motors working to compensate, but.... then it reaches Y190 it's not at Z0.2 but somewhere below 0 (although M114 reports Z0.2) 

So obviously the compensation doesn't work well.

Does this work differently for anyone ? 

I'm on the latest version before the move to MarlinDev

Wackerbarth, 2015-08-22:
There have been some corrections made since the move the MarlinDev. Please try working from the latest version of "master" on this site and let us know if the issue persists.

MoonshineSG, 2015-08-22:
OK. I'll do that and report back. Thanks.

MoonshineSG, 2015-08-22:
No, no difference. I'm on Marlin 1.1.0 Development From MarlinDev (a7822e3ff7c204d081a6b7e7276564262ae71779) 

The difference on the front and back measuring point for G29 is around 0.5mm.
My probe is accuracy is quite high, so it's not that.... 

Anyone does auto-levelling with success (as in accurate) ?

Sniffle, 2015-08-22:
I am using the entire bed to print on an 8" printer all the time with the
current marlin dev master.  Using auto bed leveling, bottoms come out
looking consistent.

MoonshineSG, 2015-08-22:
how accurate is the levelling ? I mean the manual levelling. Maybe there little need for compensation and then the "auto" works well... ? 

My next step will be to painfully manual level the bed.. :(

Sniffle, 2015-08-22:
If all the prints stick well, and look consistent on the bottom then the
auto leveling is working sufficiently to print properly and correcting
properly.  I havent had any issues with my print quality.  And most of the
time when i have its been because of the bed warping from heat not tye
machine moving improperly from the correction.

MoonshineSG, 2015-08-22:
Can you tell me which version you're on ? 

I'm testing with cold bed.

Roxy-3DPrintBoard, 2015-08-22:
> how accurate is the levelling ? I mean the manual levelling. Maybe there little need for compensation and then the "auto" works well... ?

> My next step will be to painfully manual level the bed.. :(

The auto leveling works regardless of how tilted the bed is.  But it works MUCH better if the bed is as level as you can get it.

MoonshineSG, 2015-08-22:
Well, isn't the whole point of auto levelling so you don't have to do it manually ?

If the measurement points are wrongly compensated, I don't know what else should I expect. 

I just did a small experiment. Printed (with auto levelling) a relatively large test piece (4 relatively large hexagons that cover middle and corners)
On the first layer I could tell that one corner was too high so I paused the print, adjusted the screw, resumed, pause, adjust... until it looked ok.
After 2 layers (which looked ok) I stopped the print. Clean the bed (as gently as I could) and without any other adjustments, I started the print again (it would re run the auto-levelling)
To my surprise, the first layer started exactly as before I adjusted the problem corner. G29 compensated again for the diff. There's only so much a screw can be tightened.

Wackerbarth, 2015-08-22:
Which version of leaving are you using? What numbers do you get for the grid probing?
Try raising the number of grid points to 7, 9 , or 11 (only for the purpose of giving us additional output, not permanently).

There is a debug version (ddbe733) which adds some debugging output (available on Wackerbarth/MarlinDev or as PR #54 here). It is based on today's master. If I get some confirmation that it works (generates the additional debug output), I will make it part of master so that everyone will have access to optionally include the extra debug information)

fmalpartida, 2015-08-22:
I currently have the latest marlindev and is working very well, it compensates the entire bed surface. I use an inductive sensor. 
The thing is that my print surface is flat, no bumps or lumps and works as expected given the fact that we have a pure linear correction, i.e. will only compensate for tilts. 

Is your surface completely flat? Is it a glass or aluminum sheet? What sensor are you using. 
In the repetitive test, the last time I tried I had a 0.01 resolution with a 0.001 std deviation, if I remember correctly. Have you tried the repetitive test? What results are you getting?

Hope it helps

MoonshineSG, 2015-08-23:
I'm using MK3 aluminium bed with induction sensor. The repetitive test looks good to me

Send: M48 V3
Recv: M48 Z-Probe Repeatability test
Recv: Positioning the probe...
Recv: 1 of 10   z: 0.844828 mean: 0.844828   sigma: 0.000000
Recv: 2 of 10   z: 0.844828 mean: 0.844828   sigma: 0.000000
Recv: 3 of 10   z: 0.844828 mean: 0.844828   sigma: 0.000000
Recv: 4 of 10   z: 0.844828 mean: 0.844828   sigma: 0.000000
Recv: 5 of 10   z: 0.849754 mean: 0.845813   sigma: 0.001970
Recv: 6 of 10   z: 0.847291 mean: 0.846059   sigma: 0.001881
Recv: 7 of 10   z: 0.847291 mean: 0.846235   sigma: 0.001794
Recv: 8 of 10   z: 0.854680 mean: 0.847291   sigma: 0.003258
Recv: 9 of 10   z: 0.852217 mean: 0.847838   sigma: 0.003440
Recv: 10 of 10   z: 0.847291 mean: 0.847783   sigma: 0.003268
Recv: Mean: 0.847783
Recv: Standard Deviation: 0.003268

Roxy-3DPrintBoard, 2015-08-23:
> Recv: Standard Deviation: 0.003268

Very respectable!

RicardoGA, 2015-08-23:
@MoonshineSG You should try the @AnHardt branch (inverse-matrix-z), pull request #42, i've been using this branch for a couple of weeks with a capacitive sensor and works great.

MoonshineSG, 2015-08-23:
I will try that. 

Meanwhile I did some measurements, and it does seams like my aluminium bed is in a bad shape...  :(

[image]

fmalpartida, 2015-08-23:
Yeap that is what I thought. It is wrapped a bit and hence the difference in height you are experiencing.

MoonshineSG, 2015-08-23:
But what is puzzling is the fact that even the measuring points end up at wrong level...

Wackerbarth, 2015-08-23:
@RicardoGA , @MoonshineSG : I think that the changes of PR#42 already have been incorporated in the current  "master".  @AnHardt . Would you please confirm this.

MoonshineSG, 2015-08-23:
It doesn't look like #42 is in. And there are some conflicts...

Wackerbarth, 2015-08-23:
I'm not claiming that it was merged. Rather, someone incorporated many of the changes (I'm not sure if it is all) into another change.

MoonshineSG, 2015-08-28:
I tried to merge the codes but i find it too difficult without fully understanding the codes... Back to manual levelling for a while...

Wackerbarth, 2015-08-28:
I would like to help.  Can you post a repository commit of your latest attempt so that I might make suggestions?

MoonshineSG, 2015-08-28:
Thanks, but I tried to do that on my local machine using BeyondCompare...

Wackerbarth, 2015-08-28:
Well, "Beyond Compare" seems to be a Windows product. Our development is really based on the GitHub infrastructure. Establish a personal Github account, "fork" MerlinDev, and push your build configurations to a personal branch (don't use "master", but create a branch unique to your purpose (you may end up with multiple branches, each dressing particular issues)
As the first part of your build, push the changed files to GitHub and you can ten provide the commit SHA so that we can retrieve the EXACT configuration that you are attempting to use.

MoonshineSG, 2015-08-28:
Was a Windows only but now they have Mac and Linux versions. Best file compare utility, in my opinion. I compare local files with Github files and even 3 way compare/merge... 

I was trying to merge the changes on PR#42 (which are on a older version or MarlinDev - pre renaming of the z_probe to z_min_probe) into current MarlinDev

But the best tool is only as good as it's user.. So I am in over my head :D 

I will try to follow your instructions but I'm off on a 2 week holiday tomorrow, so it will have to wait.. (or maybe slowly work on it during the holiday :D )

Wackerbarth, 2015-08-28:
Well, (1) enjoy your holiday and (2) try to isolate your changes from some particular version of the source and then apply those changes to the current version.

MoonshineSG, 2015-08-28:
I give up. I almost made a rubbish pull request on the MarlinDev...

thinkyhead, 2015-09-01:
@MoonshineSG When you get back I can help you with Github voodoo. Meanwhile check out http://pcottle.github.io/learnGitBranching/ – an interactive lesson on basic branching, merging, rebasing, etc. By the time you finish it you will have much more confidence!

MoonshineSG, 2015-09-19:
I managed to merge the codes from PR#42 but the results are not better. The output of G29 

```
Send: G29 P3 V4 T
Recv: G29 Auto Bed Leveling
Recv: Bed X: 32.000 Y: 5.000 Z: 3.660
Recv: Bed X: 111.000 Y: 5.000 Z: 4.259
Recv: Bed X: 190.000 Y: 5.000 Z: 4.734
Recv: Bed X: 190.000 Y: 95.000 Z: 5.200
Recv: Bed X: 111.000 Y: 95.000 Z: 4.852
Recv: Bed X: 32.000 Y: 95.000 Z: 4.333
Recv: Bed X: 32.000 Y: 185.000 Z: 4.704
Recv: Bed X: 111.000 Y: 185.000 Z: 5.039
Recv: Bed X: 190.000 Y: 185.000 Z: 5.328
Recv: Eqn coefficients: a: 0.00540936 b: 0.00447911 d: 3.65275335
Recv: planeNormal x: -0.005409 y: -0.004479 z: 1.000000
Recv: 
Recv: Bed Height Topography:
Recv: +-----------+
Recv: |...Back....|
Recv: |Left..Right|
Recv: |...Front...|
Recv: +-----------+
Recv: Measured Bed Topography:
Recv:  +4.70443 +5.03941 +5.32759
Recv:  +5.19951 +4.85222 +4.33251
Recv:  +3.66010 +4.25862 +4.73399
Recv: 
Recv: Corrected Bed Topography:
Recv:  +1.04433 +1.37931 +1.66749
Recv:  +1.53941 +1.19212 +0.67241
Recv:  +0.00000 +0.59852 +1.07389
Recv: 
Recv: Corrected Bed Topography in new coordinates:
Recv:  +0.23809 +0.14573 +0.00657
Recv:  +0.28160 +0.36165 +0.26929
Recv:  +0.00000 +0.17118 +0.21921
Recv: 
Recv: Height from Bed to Nozzle
Recv: (+) above, or (-) below surface :
Recv:  +0.12356 +0.21592 +0.35508
Recv:  +0.08005 +0.00000 +0.09236
Recv:  +0.36165 +0.19047 +0.14244
Recv: 
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +0.999985 +0.000000 +0.005409
Recv: -0.000024 +0.999990 +0.004479
Recv: -0.005409 -0.004479 +0.999975
Recv: echo:enqueueing "G1 X0 Y0 Z15 F5000"
Recv: ok
Recv: ok
```

thinkyhead, 2015-09-29:
@MoonshineSG Unfortunately, while I very much want to help you with this, I'm boycotting the Marlin project in protest until serious problems with the code and the development team are remedied. From now on you'll find me focusing on Repetier and Smoothie.

Hopefully some coder will be available to help you. At the moment, unfortunately, most "contributors" here do not seem to be experienced coders. But I encourage you to reach out to people who are actually posting bug fixes and/or making features. Those are the people who are most likely be able to figure this out.

MoonshineSG, 2015-09-29:
@thinkyhead I did notice the "scandal".... It's a shame... I downloaded Repetier yesterday, but the thought of understanding, reconfiguring and testing a new firmware is painful.... I guess I'll have to live with the current (pre-scandal) version... 

Again, it's a shame.

MoonshineSG, 2015-09-29:
@thinkyhead  thanks for taking the time to reply.

thinkyhead, 2016-02-19:
@MoonshineSG I've been poking around the issues queue for a couple of weeks to see where I can help patch things up. I'm curious to see if you've had any success with bed leveling in the interim, and where you are at.

MoonshineSG, 2016-02-19:
@thinkyhead I took the easy way out. I replaced my bed and now it seams that things are better. But that has nothing to do with the codes.

thinkyhead, 2016-04-22:
Closing this since we've done so much to `RCBugFix` around leveling, and that code will be migrating over here.

closed 2016-04-22 by thinkyhead