MarlinFirmware/MarlinDev #102: Some additional subdivision of the configuration file.

Wackerbarth, 2015-08-24:
There seems to be a real problem maintaining the %*&%( Configuration file since, when someone adds something to the configuration, it is expected that that item will be added to ALL configuration files.

We have already "broken" this model in two respects.

1) The delta and scara configurations are not maintained the cartesian configuration file, and 
2) there is already a split into the primary file and an "advanced" file which contains some critical values which must be customized on some machines.

I propose that, in addition, we immediately split out the display configuration from the rest of the primary file.

Having a display is an option and, at least to a large extent, the display configuration parameters are independent of the kinematic design and processor.  Thus, one "generic" configuration file could be shared among multiple "motherboards".

The primary advantage in doing this is that a developer who adds some display feature would add it in only one place rather than either adding to ALL of the example files himself, or making the existing example file "out-of-date" because they do not include this new feature.

Roxy-3DPrintBoard, 2015-08-25:
I definitely agree.  But I know a lot of people feel that all configuration should be in one place.  The problem with that view point is it adds so much clutter.    I would prefer having the main configuration done in Configuration.h and then a separate file to define things that get turned on such as LCD Panels and Auto Bed Leveling.

Roxy-3DPrintBoard, 2015-09-02:
How about a little discussion on this topic?   

It would be good to hear where everybody's head is on this.   In particular, I would like to make sure we are getting the benefit of the whole team's best thinking.  

It would be good to hear thinking that says "This isn't as good as it sounds." and other thinking that is worried "This is way too hard and we shouldn't even be thinking about this."

Because right now...   If we could wave a magic wand and make this happen, I think the code base would be a lot cleaner and easier to maintain.

thinkyhead, 2015-09-03:
I'd like to see some PRs exemplifying these ideas to test their viability.

closed 2016-05-04 by jbrazio