MarlinFirmware/MarlinDev #103: Duplication not working in dual_x_carriage

darkjavi, 2015-08-26:
I've been playing for a month with two 'dual x carriage' machines, machines that have the same frame and mechanics but different electronics(rumba/megatronics).

Dual X Carriage have 3 modes of operation:
 When it's configured in mode 0/1 (full control/autopark) it works as expected as a dual extruder machine, both x axes stepper works as well as both extruder steppers and heaters.

Replication mode(2) has a bug though, when is active both x axes steppers are working, and also the heaters, but only the first extruder stepper works.

I'm bug hunting the issue, and I narrowed down to the 'enable pin' of the second extruder, looks to me like is timing out and disabling the stepper.

While exposing the issue I've tried to manually put the enable pin of the second extruder stepper to gnd(with a wire) and the second extruder starts working, so **is receiving steps but the enable pin not enabled**

Now I'm looking through stepper/planner/stepper_indirection files looking for the point in which the steppers are enabled, anyone can give a clue about it?

darkjavi, 2015-08-26:
I found the problem, it has to do with the active_extruder mechanism which is forgetting to keep the e1 stepper enabled while in duplication mode. I made a little fix for that I can craft into a pr if next days testing goes well.
https://goo.gl/photos/dLAUbJSKcsstmRsT7

thinkyhead, 2015-08-28:
@darkjavi Good news, and so cool to watch in action!

barus93, 2015-08-30:
I had the same problem months ago , I solved by importing the variable "extruder_duplication_enabled" in planner.cpp and adding this code in line 633

if (extruder_duplication_enabled) 
{ enable_e1();
} else
 if (g_uc_extruder_last_move[1] == 0) 
disable_e1();

darkjavi, 2015-08-30:
Nice, that is pretty much what I did, with an additional '#ifdef DUAL_X_CARRIAGE' wrapping it.
So far is working properly, though I want to improve the 'unpark' secuence of the duplication so it lifts the gantry before moving the second extruder to the 'duplication offset' so it doesnt drag the glass holding clips in that movement. I'll be working on a pr for it next week.

thinkyhead, 2015-09-01:
@darkjavi @barus93 Have a look at #120 and let me know if I've missed anything. I get the point of `if (g_uc_extruder_last_move[1] == 0) disable_e1();` since a dual-x setup can operate as two distinct extruders with their own movements. I'm wondering, however, if I should also add the following:
```cpp
#if ENABLED(DUAL_X_CARRIAGE)
  if (extruder_duplication_enabled) g_uc_extruder_last_move[1] = BLOCK_BUFFER_SIZE * 2;
#endif
```
The reason for that would be similar to the reason for the `disable_e1` checking. It's because the movement of E0 is also applying to E1.

closed 2015-09-13 by darkjavi