MarlinFirmware/MarlinDev #105: Issue with auto bed leveling

ImDroided, 2015-08-27:
Hey guys and gals I am not sure what is going on here I cant find any info on this on google so I came here. 

Ok so trying to set up my bed leveling and for some reason my printer is saying its moving my X axis to 170mm for probing but it is only moving like maybe 50mm here is my ABL config to where my sensor will still reach the bed.

    #define LEFT_PROBE_BED_POSITION 45
    #define RIGHT_PROBE_BED_POSITION 170
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION 265

And here is a video of what is happening. Can someone check this out and let me know whats up. I was on the release marlin in this video and had x set to 225. I am currently on the dev version and getting the same thing happening. 

<iframe width="560" height="315" src="https://www.youtube.com/embed/k9_YOhhJsIA" frameborder="0" allowfullscreen></iframe>

ImDroided, 2015-08-27:
I should add all my movements and printing are just fine manually. 50mm is 50 mm

Wackerbarth, 2015-08-27:
You have quoted only part of the settings which affect bed leveling -- Perhaps you should post your entire configuration. The easiest way is to create your own github fork of MarlinDev and then, within that repository, create a unique branch for your personal build.  Commit all of your changes to that branch and push it to your repository.

That way, someone, such as myself, should be able to reproduce EXACTLY the firmware that you are attempting to use.

PSimonovich, 2015-08-28:
The probe_pt function in Marlin_main.cpp subtracts X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER from the probe coordinates before lowering the Z probe.
Therefore, your actual probe points become: [(45-41,20-0) (170-41,20-0) (45-41,265-0) (170-41,265-0)] or [(5,20) (129,20) (5,265) (129,20)]. Try increasing your LEFT_PROBE_BED_POSITION and RIGHT_PROBE_BED_POSITION values by +41.

closed 2016-05-04 by jbrazio