MarlinFirmware/MarlinDev #107: Hangout -- Saturday August 29 UTC 2000

Wackerbarth, 2015-08-27:
Name  | Summer Time Zone
------------ | -------------
@thinkyhead    |  UTC-7
@Wackerbarth  |   UTC-5
@Jnesselr    |    UTC-5
@clefranc     |   UTC-4
@alexborro    |   UTC-3
@fmalpartida   |  UTC+1
@AnHardt     |    UTC+1
@Wurstnase    |   UTC+1
@CONSULitAS   |   UTC+1
@paclema     |    UTC+1
@boelle    |      UTC+1
@emartinez167   |  UTC+8
@ntoff    |       UTC+10
@Roxy-3DPrintBoard  |   UTC-5

[The "Official" UTC Clock](http://www.timeanddate.com/time/aboututc.html)
[Link to the Hangout](https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia)

It's been two weeks. Time for another session.
I hope that we might get some better turnout.

It's time that we put out a Release Candidate.

Please provide input about what we need to discuss.
[Link to Google Document listing discussion topics (Week 35)](https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing)

fmalpartida, 2015-08-27:
Cool, tomorrow evening! Forget the previous comment on the other issue.

fmalpartida, 2015-08-27:
Well Saturday.

Jnesselr, 2015-08-27:
I'm UTC-6 now btw.

Wackerbarth, 2015-08-27:
@Jnesselr -- Colorado? I thought you were in Arkansas (Central Daylight Time, like myself)

Jnesselr, 2015-08-27:
I moved to Colorado a couple of months ago.

Wackerbarth, 2015-08-27:
OK, fixed

Jnesselr, 2015-08-27:
Thank you very much

thinkyhead, 2015-08-28:
Definitely time to get out RC1 and start getting feedback (especially on `G29`) from the wider community. And we might take time to look at some of the pending PRs and decide where they stand.

Wackerbarth, 2015-08-28:
I hope that any "almost ready to be included" items can be upgraded and included before Saturday.

closed 2016-05-04 by jbrazio

scalez, 2016-05-09:
@thinkyhead @jbrazio have the hangout meetings stopped? I would still be interested in attending them.

jbrazio, 2016-05-09:
@scalez they have been stopped for a while. It may be something we restart once the GA is out if enough interest is shown by the community.

But to be honest, I like the openness of Github, everything we discuss is widely open to be read and commented by the community, I do prefer this rather than a "back alley talk between a few".