MarlinFirmware/MarlinDev #110: Feature request: direct g-code drive/mixing with multiple extruders

Virtox, 2015-08-28:
I was directed here after finding https://github.com/MarlinFirmware/Marlin/issues/923
With the arrival of the diamond hot-end 3-in-1-out and similar type extruders, the need for mix-truder support is growing. And I'd be happy to work up with some-one to see if we can implement something useful.
But reading the mentioned thread above, with regard to the load on the mcu:
If the g-code is implemented to specify mixing with something like E<length0>:<length1>:<length2> or E<length>:<ratio0>,<ratio1>,<ratio2>, I expect the extra loads should be minimal?
This would be my personal preference: a direct ratio/extrusion control of separate extruder motors via g-code.
To keep things simple it could require the g-code generator to make sure the combined extrusion from the one nozzle = 100%.
As I think the way of virtual tools/extruders as is done in repetier and reprapfirmware creates too much overhead for smooth color transitions and limits the possibilities.
Any advice is welcome before I spend too much time on a lost cause ;-)

thinkyhead, 2015-09-01:
I'm definitely thinking in the direction of making tools in Marlin into more abstract entities, but my hope is that we can do that without adding any more overhead compared to the current (less OOP) approach. A mixing extruder basically has 2 or more extruder motors that move together, very much like the XYZ axes move together to do diagonal movement. Obviously there are differences, but the principle of setting up and doing stepper movement is the same. Building off the current Marlin code-base, I guess we would first need to deal with E axis movement with colon separators, and then we would need to take the current single-stepper-per-E-axis and supplant it with a 2-or-more-stepper-per-E-axis code blob. I think that can be done using the current conditional compilation technique we currently use, and without first creating an abstract tool object.

Virtox, 2015-09-03:
Cool! Pretty much sums up exactly what I was hoping!

thinkyhead, 2015-09-04:
@Virtox I hope you will find the code-base readable and manageable. I've written up some guidance about [Contributing](http://www.marlinfirmware.org/index.php/Contributing) and [Coding Standards](http://www.marlinfirmware.org/index.php/Coding_Standards) at the Marlin wiki.

As for the starting-point for a mixing extruder, start with the `stepper.cpp` and `planner.cpp` files to see how stepping and planning are being set up. You will note right away that there is only a single E axis currently supported throughout the codebase. So the first thing to do is turn every static variable that refers to the E axis and make it into an array. The compiler will tell you a lot of what to change after that!

When creating a feature like this, you will of course create a branch for it, based off of `MarlinDev/master`, and modify that. Start by adding lines like these to `Configuration.h`:
```cpp
// Enable this option for a mixing (multi-stepper) extruder
//#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define EXTRUDER_AXES 3 // Number of steppers in the extruder
#endif
```
When modifying code use the `ENABLED` / `DISABLED` macros to conditionalize. For example…
```cpp
  plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS],
    current_position[E_AXIS],
    #if ENABLED(MIXING_EXTRUDER) && EXTRUDER_AXES > 1
      current_position[E_AXIS + 1],
      #if EXTRUDER_AXES > 2
        current_position[E_AXIS + 2],
        #if EXTRUDER_AXES > 3
          current_position[E_AXIS + 3],
        #endif // AXES > 3
      #endif // AXES > 2
    #endif // MIXING_EXTRUDER
    feedrate/60, active_extruder);
```
Actually, for something like that, you will more likely want to make a helpful macro:
```cpp
#if ENABLED(MIXING_EXTRUDER) && EXTRUDER_AXES > 1
  #if EXTRUDER_AXES == 4
    #define EPOS(var) var[E_AXIS],var[E_AXIS+1],var[E_AXIS+2],var[E_AXIS+3]
  #elif EXTRUDER_AXES == 3
    #define EPOS(var) var[E_AXIS],var[E_AXIS+1],var[E_AXIS+2]
  #else
    #define EPOS(var) var[E_AXIS],var[E_AXIS+1]
  #endif
#else
  #define EPOS(var) var[E_AXIS]
#endif
```
…and then use it all over the place, as for example…
```cpp
plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], EPOS(current_position), feedrate/60, active_extruder);
```
This is a very complex feature and won't make the Marlin code look prettier, but it should be very efficient AVR code when fully implemented. Going from the usual 4 axes to 6 or 7

thinkyhead, 2015-09-04:
Another thing that will change, affecting code all over the place (so watch out for anywhere it is being misused), is the value of `NUM_AXIS` – currently fixed at `4`. The mixing extruder will change it to:
```cpp
#if ENABLED(MIXING_EXTRUDER)
  #define NUM_AXIS (3 + EXTRUDER_AXES)
#else
  #define NUM_AXIS 4
#endif
```
`NUM_AXIS` is mostly used in loops and in defining the lengths of arrays. In fact it affects all arrays (like `float current_position[NUM_AXIS];`) that are meant to represent XYZE so you don't have to worry about those.

jhsandell, 2015-09-09:
@thinkyhead  and @Virtox 

You might also consider adding lines in the SanityCheck.h to rule-out those machine configurations where this will not work. For example the DUAL_X_EXTRUDER thingy.
Tool-change will also be a challenge. 

PS. Tool-change (extruder/filament/color) on the diamond head will be super-easy. Just make a mix ratio of 0,1,0 og similar.

thinkyhead, 2016-02-19:
Sorry if I missed your birthdays. I got you this: https://github.com/MarlinFirmware/MarlinDev/pull/361

Virtox, 2016-02-19:
Now worries, birthday was only couple of days ago ;-)
Many thanks! This is awesome!!
I had just switched to Repetier to get at least some mixing going on but it is limited, even with tricks and firmware hacks.
So I am looking forward to try this out, but love the motion smoothness of repetier on my delta though.
but I will definitely try this out on the next delta I am building. 
I have some rudimentary code snippets and scripts to add color to gcode, but still need to code the part to get the color straight from vertex color models.

thinkyhead, 2016-02-21:
> love the motion smoothness of repetier on my delta

I hope we can steal some of their best ideas. A delta with graphical display pushes the limits for Marlin.

Virtox, 2016-02-21:
Cool :-) Have similar issues with Repetier: delta with full graphics + bed leveling gave me random crashes due to memory shortage. So I switched to 2004 LCD, which at least works, but could be smoother still. Going to try due+radds+raps128 for main printer, likely with Repetier, on account of nor Marlin support. But will try Marlin mixing with 2004LCD+rumba on second machine in couple of weeks.

yaniekarlina, 2016-03-01:
hello!
may i join this conversation? and i have a question. do you need stepper motor for mixing color? or you  use nozzle only?
thank you :)

Blue-Marlin, 2016-03-01:
This approach is about one nozzle and multiple filaments. Every filament is feed by its own stepper motor.
The complex off mixing the filament in the nozzle, to get a homogeneous colour, is completely omitted here.
But stirring the colours in the nozzle seems to be not soo complicated (for the software).
At the moment we have things like the http://reprap.org/wiki/Diamond_Hotend in mind.

Sniffle, 2016-03-01:
You mean you need more than setting extruders to 3 and having all of ths
offsets be 0?

Blue-Marlin, 2016-03-01:
If you want the pure colours of the filament - no.
If you want blended/mixed colours - yes.

thinkyhead, 2016-03-02:
> Going to try due+radds+raps128 for main printer, likely with Repetier, on account of nor Marlin

@Virtox You might also give https://github.com/MagoKimbra/MarlinKimbra4due a try.

thinkyhead, 2016-03-02:
> You mean you need more than setting extruders to 3 and having all of their offsets be 0?

@Sniffle The way the feature works (#361) is that you must `#define EXTRUDERS 1`, enable the `MIXING_EXTRUDER_FEATURE`, and then set `MIXING_STEPPERS` to the number of steppers that your extruder has. The mixing itself is done following the Repetier-firmware approach, where you can set the mix directly, one component at a time, and then you can save the mixing factors as "tools." This allows you to use tool-selection commands —`T0`, `T1`, etc.— to switch between the different mixes you have saved. This will also work as a multi-color (non-mixing) extruder, but you need a wipe tower to clear the nozzle between colors.

yaniekarlina, 2016-03-02:
i found this https://paulwanamaker.wordpress.com/six-color-color-mixing-hot-end-for-reprap/ and http://www.particlesciences.com/news/technical-briefs/2011/hot-melt-extrusion.html so i think the nozzle need a stepper motor for mixing color. can you give me a picture or video about your 3d printer mixing color?

Blue-Marlin, 2016-03-02:
https://www.youtube.com/watch?v=qCF_Nzo2ET4

closed 2016-05-27 by jbrazio