MarlinFirmware/MarlinDev #111: Forward porting delta auto calibration code from RichCattell/Marlin?

mairas, 2015-08-28:
I read a bit the previous issues and seems like the delta auto-leveling code is more or less missing in the current dev head version. Deltas not only benefit from bed auto-leveling, they can also calibrate all of their geometry using the auto-leveling probe. There's one implementation of such code at:

https://github.com/RichCattell/Marlin

Unfortunately, that code seems to originate from a really old fork. I wonder what the reactions would be if I made an attempt to forward-port the delta auto calibration code back to the curren MarlinDev master? Would it be likely that such a pull request would be accepted (assuming normal code quality/hygiene standards)?

Wackerbarth, 2015-08-28:
You are correct that Rich's code is based on a VERY old version of Marlin. I started to port it over but ran into some problems. 

I also found that it fails to converge reliably. (I think that is likely to be a flaw in the underlying mathematical model whereby, the locations probed are unable to uniquely determine a solution vector)

I would not spend the effort to port this code at this time because, in the next iteration, we will be doing some significant rewrites to unify the various geometric designs. As a result, any delta-specific code will likely need to be rewritten.

We should end up with a system with one common calibration system that determines the coordinate transformation coefficients for any of our geometries.

If you are interested in helping with that effort, your assistance would be very welcome.

mairas, 2015-08-28:
Thanks for the fast reply. I'll contain my excitement for the time being, then. I might be interested in contributing - I seem to get nerd-sniped by this topic very easily, anyway. :-) See:

https://github.com/mairas/delta_calibration

What's your primary communication channel?

closed 2016-05-04 by jbrazio