MarlinFirmware/MarlinDev #113: G30 / G28 issue when z_probe is still triggered

o-lampe, 2015-08-30:
I have tried the following sequence
G28
G0 F8000 Z0
G30
M119
G28
several times on my delta.
Whenever m119 showed the z_min_probe switch is triggered after G30, the G28 command failed in a way, that only one axis moves to home position. The other axis follow, when the first axis hits the max. endstop.
My understanding of G30 would be to move down until the probe gets triggered, then slowly move up, until the switch opens again. That is the new z_home position. ( Automatically send  m428?)

I´m using marlin 1.03dev from 14.08.2015. The sequence is sent by pronterface.
Cheers
o_lampe

thinkyhead, 2015-09-01:
Thank you for this report. Are you using the `ENDSTOPS_ONLY_FOR_HOMING` setting? I'm not sure what else to suggest. The homing/leveling/probe code is currently in flux and there might be some lingering problems with any of the related commands.

I wish I had a delta to help test this, but I can only suggest things you might try and cross my fingers. With the latest code you can enable `DEBUG_LEVELING_FEATURE` and then use `M115 S16` to enable extra logging. The log output might reveal what's going on, or we might be able to add more logging to get more information.

o-lampe, 2015-09-01:
[code] ENDSTOPS_ONLY_FOR_HOMING  [/code]
is disabled.
I only use z_probe, not z_min. The nozzle is my probe, so I disabled z_safe_homing.
Probe offsets are all zero, except z_offset is set to -0.01mm

thinkyhead, 2015-09-03:
Did changing the Z homing rate help at all? It should cause probing to happen at a faster rate, and that might be what this needed.

thinkyhead, 2015-09-03:
To clarify,`G30` does do a Z probe, but it doesn't set the Z home position at all. It just reports the XYZ position where the probe was triggered.

o-lampe, 2015-09-03:
I see, it´s not possible to send a M428 after each G30, because G29 is just an array of G30´s

My idea of a G30 is to move down until it triggers, than slowly move up until the probe-pin is open again. ( requires additional temporary flag variable )
That way you eliminate friction and backlash of the move. ( at least in my imagination )
Then report the XYZ coords.
I just received new parts for my delta, so it will take a while to test your suggestion @thinkyhead 

Another sidenote: The problem exists, because all endstops are handled the same ( z-probe should be different from max. endstops ) "ERR: Endstop hit ZP" should be nonexistant

Wackerbarth, 2015-09-03:
"That way you eliminate friction and backlash of the move. ( at least in my imagination )"

Perhaps it is your imagination. Why do you think that friction and backlash do not apply to the reversal of direction which occurs while the switch is closed?

o-lampe, 2015-09-03:
Of Cours they apply @Wackerbarth , but they have a negative sign on them, so they ( more or less ) eliminate each other.
( again just my wild fantasy ) :)

Wackerbarth, 2015-09-03:
I don't understand that last explanation at all.

clefranc, 2015-09-03:
@Wackerbarth Hysteresis?

Wackerbarth, 2015-09-03:
Yes, hysteresis applies in both directions. If anything, it raises the uncertainty as the the actual position when the switch changes state.

closed 2016-05-04 by jbrazio