MarlinFirmware/MarlinDev #114: LCD #Move Axis way to slow for delta z-axis

o-lampe, 2015-08-30:
When I use the smart LCDs "move axis" menue, for z-moves it is way slower than  for x and y axis.
For a delta this is a PITA. Since deltas use different configuration.h and configuration_adv.h, it would be beneficial to define a speed setting there.

Cheers
o_lampe

thinkyhead, 2015-09-01:
Good idea. I'm putting this on my to-do list. We've had some other requests related to the speed of delta Z movement, homing, and probing.

o-lampe, 2015-09-01:
I´m just chewing through the newest dev files and found the right line in config_adv.h
 #if ENABLED(ULTIPANEL)
  #define MANUAL_FEEDRATE {50*60, 50*60, 4*60, 60} // Feedrates for manual moves along X, Y, Z, E from panel

This was copied from the "example configuration/ generic delta subfolder. Should be set to 50*60 for all three axis´.

thinkyhead, 2015-09-03:
Good catch, @o-lampe thanks!

closed 2015-09-03 by thinkyhead