MarlinFirmware/MarlinDev #118: Sanity check ignores wrong delta settings

o-lampe, 2015-08-31:
I made a typo in the delta settings and set the delta printable radius to 1400. Compiler doesn`t do any feasibility check?
I then reduced diagonal rod length to 39, again compiled without error message.
-Olaf

thinkyhead, 2015-09-02:
What limits would you suggest?

o-lampe, 2015-09-02:
Sanity check could do the 20° test:
DELTA_DIAGONAL_ROD*COS(20) / 2  > DELTA_PRINTABLE_RADIUS

Wackerbarth, 2015-09-02:
Shouldn't that be
```
DELTA_DIAGONAL_ROD*COS(20)  > (DELTA_PRINTABLE_RADIUS + DELTA_RADIUS)
```

o-lampe, 2015-09-03:
@Wackerbarth 
I was under the impression, that 20° was a fixed minimum value by design? 
With your formula I would only have a 240mm dia printbed instead of 300.
(300 was what I got when I calculated with 3rd party formulae.)
https://docs.google.com/spreadsheets/d/1eFuVzpm1HHwWnixWByc1Y2XjecCbTer4-jzD-YRUceY/edit?pli=1#gid=2118974717

Wackerbarth, 2015-09-03:
The 20ª figure is a practical design value. However, there is nothing inherent in the mechanics that would REQUIRE that it be the limit. I might choose to assure that my device stay above 30ª or I might push the limits and permit 18ª. By making the angle a defined value, with default of 20ª, the user can customize the threshold however he desires. 

My argument for DELTA_PRINTABLE_RADIUS + DELTA_RADIUS is that that is the actual horizontal distance involved. The tower is effectively DELTA_RADIUS from the center and not located right on the edge of the printable area.

o-lampe, 2015-09-03:
I agree with you @Wackerbarth 
but IMHO sanity check is not meant as a delta design guide, just to evaluate missing decimal points or wrong numbers in general. To reflect this, we could calculate the cos(angle) and when it´s in the range of 18-30° we are good. Including the Delta_Radius means, we have all variables taken into account.

Wackerbarth, 2015-09-03:
My reason for making the angle a parameter is so that the user can change the limit when he knows that he is pushing the normally accepted limits.

thinkyhead, 2015-09-07:
> Sanity check could do the 20° test: `DELTA_DIAGONAL_ROD*COS(20) / 2 > DELTA_PRINTABLE_RADIUS`

Actually the pre-compiler can't do anything near that complex. It can't even process floats. Some compiled code could do such a test at startup and produce a boot error, as an alternative.

o-lampe, 2015-09-07:
@thinkyhead  Then we could add a comment to the `DELTA_PRINTABLE_RADIUS` line, like:
**don´t use values greater than  DELTA_DIAGONAL_ROD*COS(20) / 2 **  ?

closed 2016-05-04 by jbrazio