MarlinFirmware/MarlinDev #119: Testers; how do we identify which version we tested

shampine1, 2015-08-31:
I've just downloaded the master to test on printrboard, but its not obvious what I should call it when I report back.  It says commit 962120d1d2, or the total number of commits of 3,768.  Any preferences?

Wackerbarth, 2015-08-31:
The commit number (962120d) is the better description.

Which OS and toolset are you using to compile?
I ask because if you are building from the git repository, we have an automatic versioning mechanism that is partially implemented (works fine on MacOSX and Linux if you are using the Arduino IDE 1.6.3+)

It provides the version designation in the M115 output.

Help getting this working on Windows would be appreciated.

thinkyhead, 2015-09-01:
I personally don't favor going back in history to confirm issues. In my view, it's always best to test with "the latest code," and then if you find a bug make sure to test once again with "the very latest code" (if it differs) just before posting an issue. Changes to the code-base are usually incremental and slow, and we usually have a pretty good window to deal with bugs in any specific portion of the code. This is of course different from how we deal with bugs in Marlin/Release, where we might only document that a bug exists rather than try to patch it.

closed 2016-05-04 by jbrazio