MarlinFirmware/MarlinDev #121: Confusing SanityCheck.h FRONT_PROBE_BED_POSITION > BACK_PROBE_BED_POSITION

hovissimo, 2015-09-01:
I just cloned MarlinDev and configured my Printrbot 1403.

Here's the relevant sections of Configuration.h

    #define X_MAX_POS 150
    #define X_MIN_POS 0
    #define Y_MAX_POS 150
    #define Y_MIN_POS 0
    //  ... omitted ...
    #define LEFT_PROBE_BED_POSITION X_MIN_POS + 15
    #define RIGHT_PROBE_BED_POSITION X_MAX_POS - 15
    #define BACK_PROBE_BED_POSITION Y_MIN_POS + 15
    #define FRONT_PROBE_BED_POSITION Y_MAX_POS - 15

When compiling (with Arduino IDE) I get:  "#error BACK_PROBE_BED_POSITION must be less than FRONT_PROBE_BED_POSITION."

Relevant code from SanityCheck.h (line 177):

    ...
    #elif FRONT_PROBE_BED_POSITION > BACK_PROBE_BED_POSITION
          #error BACK_PROBE_BED_POSITION must be less than FRONT_PROBE_BED_POSITION.

The error statement contradicts the test.
Which (FRONT or BACK) should actually be larger?

grob6000, 2015-09-01:
You're right, that doesn't work, and would throw the error with default config. This is all fishy: isn't the back of the bed YMAX and the front of the bed YMIN?

My reasoning being, the following is an internally consistent right-hand-rule coordinate system (looking at the printer, disregarding what moves, but from the perspective of an object on the bed):
* XMIN is left, XMAX is right
* YMIN is front, YMAX is back
* ZMIN is on-bed and ZMAX is way-above-bed?

Where anything other than a rotation of this would start to reflect geometry?

So shouldn't this be:
```
#define LEFT_PROBE_BED_POSITION X_MIN_POS + 15
#define RIGHT_PROBE_BED_POSITION X_MAX_POS - 15
#define BACK_PROBE_BED_POSITION Y_MAX_POS - 15 // back is ymax
#define FRONT_PROBE_BED_POSITION Y_MIN_POS + 15 // front is ymin
```
and
```
#elif FRONT_PROBE_BED_POSITION > BACK_PROBE_BED_POSITION
      #error BACK_PROBE_BED_POSITION must be greater than FRONT_PROBE_BED_POSITION.
```

?

thinkyhead, 2015-09-03:
#127

thinkyhead, 2015-09-03:
Fixed in #127

closed 2015-09-03 by thinkyhead