MarlinFirmware/MarlinDev #123: Minor problem with "delta_calibration" menue on a 2004 LCD

o-lampe, 2015-09-02:
When I send a "Calibrate_X" command and then navigate back to the main screen, the X-position shown is wrong. ( X-21 instead if X-121) The first digit got chopped off.

Newest Dev-master download.
Cheers
-Olaf

o-lampe, 2015-09-02:
Same for calibrate_Z. Display reads Y140 instead of Y-140.
This time the - sign is chopped off.

thinkyhead, 2015-09-03:
I see that the delta calibration menu is meant to be customized before flashing. The default code is pasted below. What does your delta calibration code look like?
```cpp
static void lcd_delta_calibrate_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
  MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
  MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F8000 X-77.94 Y-45 Z0"));
  MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F8000 X77.94 Y-45 Z0"));
  MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F8000 X0 Y90 Z0"));
  MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_CENTER, PSTR("G0 F8000 X0 Y0 Z0"));
  END_MENU();
}
```

o-lampe, 2015-09-03:
@thinkyhead 
Mine looks the same, except, I adapted it to my bigger printable radius of 140mm:
#if ENABLED(DELTA_CALIBRATION_MENU)
  static void lcd_delta_calibrate_menu() {
    START_MENU();
    MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
    MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F5000 X-121.24 Y-70.0 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F5000 X121.24 Y-70.0 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F5000 X0 Y140.0 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_CENTER, PSTR("G0 F8000 X0 Y0 Z0"));
    END_MENU();
  }
#endif // DELTA_CALIBRATION_MENU

It´s the 3 digit numbers that are problematic for the 2004 LCD.
BTW: How do I mark code in github? [code][/code] doesn´t work...

Wackerbarth, 2015-09-03:
there is a "triple tick" delimiter. You can check out a description of Markdown by following the "Markdown supported" link which is just above the comment entry field.

thinkyhead, 2015-09-07:
> It´s the 3 digit numbers that are problematic for the 2004 LCD.

So this affects you whenever the coordinates are very large? Is your display only 16 or 18 characters wide?

o-lampe, 2015-09-07:
@thinkyhead  it´s a LCD 2004, so it`s 20 chars wide.

thinkyhead, 2015-09-14:
@o-lampe Check out #181 which will show XY with 4 digits instead of only 3.

closed 2016-05-04 by jbrazio