MarlinFirmware/MarlinDev #125: CoreXY Stepper Judder

chorca, 2015-09-02:
It appears that using the latest version of MarlinDev, the CoreXY functionality is not working correctly.
My stepper motors are shaking on deceleration from speed, noticable at 1000 accel but not 100.
Here's a video of it happening: https://youtu.be/ilGyefw6ImQ
Basically, acceleration is smooth as usual, but during deceleration, the stepper stops at one point, then resumes the deceleration until rest is reached.

Currently at commit 962120d1d23b6847e87c6160d69dde570f68e070

Wackerbarth, 2015-09-02:
First, let me thank you for the way that you are reporting the issue. You have given a reasonable description of the behavior and a reference to the exact version of the firmware that you are testing.
I only wish that everyone would be so precise.

The one thing that you failed to do is describe the G-code that was used to cause the motion shown.

From watching it, I suspect that it was some form of homing (G28) rather than a move to the origin (G1 X0 Y0).

If that is the case, this is not an error. When homing, the carriage is moved to the index position and then backed off and reproached (more slowly) to obtain a higher quality estimate of the triggering position.

In that case, what appears to be a stutter (sic Judder) is actually a sequence of three intentional movements.

Please confirm/clarify the commands being demonstrated.

chorca, 2015-09-02:
Apologies for the oversight. This is a G0 Y-100 command, being sent by OctoPrint to the printer. While the second part of the video shows motion close to the endstop, it doesn't actuate the endstop, just gets close to it. The first part of the video shows a move away from the endstop near the middle of the bed, I just wanted to show the motor and carriage as it gives a better visual representation of the issue.
I did some testing and noticed that if I disabled CoreXY in Configuration.h, a single motor moved as would be expected, but it did not suffer this issue. It appears that enabling CoreXY causes it to happen.

chorca, 2015-09-04:
Quick update on this issue. After some more testing with a very base pull of the latest version (not much configured other than CoreXY and some endstops), I've found that setting the X/Y Steps per mm to anything over 120 causes this issue. 120.01 will cause the studdering badly, while 120.00 has no issues.
I'm attempting to determine when this bug was introduced, but I'm on a tight timeline, so not sure how much i'll be able to get done.

chorca, 2015-09-04:
Spent all night locating a good previous revision and running a git bisect on the code.
According to bisect, 715104e4775125dc46919c6af68d24e77232ed48 is the commit which introduced the issue. Before that, I'm able to run the steppers over 120 steps per unit without issue, but after this i get bad studdering when this happens. Tested up to 160 steps per unit to make sure.

I believe this hasn't been noticed before as many CoreXY machines are running the older Allegro stepper drivers or others at 16x microstepping, which is around 80 steps per mm. With the new DRV8825s and others running at 32x microstepping, the standard 20-tooth pulley and 1.8-degree stepper now runs at 160 steps per mm, so the issue becomes apparent.

Wackerbarth, 2015-09-04:
Very good detective work. We can certainly look into exactly what changed at that point.

However, I do have one question: You are relating the problem to "steps per mm". I find it difficult to suspect that to be the correct units. Although it is possible that some calculation is overflowing a counter, I think that it is more likely to be an issue of "steps per second". In operation, they would be related in that if you double the micro-stepping, you require twice as many steps per second to maintain the same velocity.

Have you tried reducing the maximum travel speeds to see if that affects the threshold at which the problem is manifest?

oysteinkrog, 2015-09-04:
Please check if the issue remains at 16x microstepping.

Perhaps related to this?:
https://plus.google.com/+%C3%98ysteinKrog/posts/XCTdGGBbjvd

You can try to apply the suggested fix described here:
https://github.com/MarlinFirmware/MarlinDev/issues/35#issuecomment-136103737

Wackerbarth, 2015-09-04:
@oysteinkrog -- Your references are also to the same issue of the stepping rate which I raised. He definitely needs to check whether, or not, it is this rate issue which we suspect to be the real issue.

chorca, 2015-09-04:
This looks like the issue. Using the same steps per unit, I ran the speeds down on my moves until it smoothly accelerated. The sound is dramatically different as well between F7501 and F7500. The latter is a smooth, quiet move where the former has a harsher tone and studders upon decel.

I applied the fix suggested in the referenced timing bug and that solved the problem. I can step up to around F16000 with 160 steps per unit now with smooth motion.
If I go above that, there are some odd noises during accel and decel and it makes the same 'thunk' as it slows again, so I may just be hitting the limit of that fix.

For now, I don't mind going back to 16x microstepping if I need that speed, now that I know what the issue is and how it's triggered.

alexborro, 2015-09-04:
This issue has been deeply discussed.
It is related to how long the ISR take to be completed. I prefer the
approach to decrease the maximum step frequency from 10Khz to 9Khz.
The impact is minimum and I'm running my bots with this fix for months.

Cheers.

Alex.

thinkyhead, 2015-09-05:
Will the default settings for new installs of Marlin cause this issue generally for CoreXY? If so, we could tweak the configuration example, or add a warning comment to the relevant setting.

alexborro, 2015-09-05:
There is an workaround. You should enable endstop checking only during
homing. Endstop checking takes some precious cycles of the ISR.

And if you prefer, you can put an ifdef to reduce the max stepper frequency
for CoreXY bots.

Cheers.

Alex.

thinkyhead, 2015-09-07:
@alexborro Are you talking about the global setting `#define MAX_STEP_FREQUENCY 10000` (or `40000`) that's set in `Conditionals.h`? Then, with `COREXY` / `COREXZ` it would be set to something like `9000` (or `36000`)…. Or is there some other setting we would need to adjust?

oysteinkrog, 2015-09-07:
This problem is AFAIK not restricted to only CoreXY.

alexborro, 2015-09-07:
@thinkyhead , MAX_STEP_FREQUENCY do not affect this behavior.

I'm running an old Marlin (from April 2015) so my code may be a little different. Check this function:

FORCE_INLINE unsigned short calc_timer(unsigned short step_rate) {
  unsigned short timer;
  if (step_rate > MAX_STEP_FREQUENCY) step_rate = MAX_STEP_FREQUENCY;

  if (step_rate > 17000) { // If steprate > 17kHz >> step 4 times
    step_rate = (step_rate >> 2) & 0x3fff;
    step_loops = 4;
  }
  else if (step_rate > 8500) { // If steprate > 8.5kHz >> step 2 times
    step_rate = (step_rate >> 1) & 0x7fff;
    step_loops = 2;
  }

I have changed the threshold of "one step to double step" (from 10khz to 8.5khz) and "double step to quad step" (from 20khz to 17khz)... That way the ISR have a few more cycles to complete.

@oysteinkrog , the issue may happen on all kinematics, but CoreXY is more prone to be affected, because it has more endstop checks in ISR and usually CoreXY bots run way faster than other ones. My CoreXY has travel of 400mm/s... regular cartesian bots do not reach such speeds.

Cheers.

Alex.

oysteinkrog, 2015-09-07:
I reach this problem on a normal Prusa i3 with high travel speeds.

thinkyhead, 2015-09-14:
> Check this function…

@alexborro That seems awfully familiar. I seem to recall that @imitation pointed out these lines in one of his videos, editing them to fix performance on a certain machine that he was reviewing. Perhaps these numbers need to be made configurable, or even tunable at runtime…?

Wurstnase, 2015-09-15:
Don't let users play with that values. They won't see any effect until they start to cry. 

We should take care about features which need that much cycles. Maybe we can give an option for a_lot_of_features_enabled to reduce that. But reduce that for any user makes no sense.

thinkyhead, 2016-04-18:
I assume we fixed this bug in `RCBugFix`…? If so, we can close this issue. Code from `RCBugFix` will be getting migrated to `MarlinDev` (and reorganized) right after 1.1.0 release.

closed 2016-05-27 by jbrazio