MarlinFirmware/MarlinDev #126: Auto Bed Leveling fails when tool moves after homing.

jminardi, 2015-09-02:
The following script causes my printer to crash its axes:

```gcode
G28
G1 X-50
G29
```

I did a little probing to figure out why but didn't find the bug. There is an error thrown when `G29` is called without homing first, but after homing no error is thrown and it fails.

If there is anything I can do to help debug let me know. FWIW I tried this with and without grid leveling mode and both fail.

Here is the config file I am using: https://github.com/Voxel8/Marlin/blob/Integration/Marlin/Configuration.h

thinkyhead, 2015-09-03:
@jminardi Using the latest code, you can enable `DEBUG_LEVELING_FEATURE` and then use `M115 S16` to enable detailed logging of homing and leveling. From that we might get some clues as to what exactly is happening.

Wackerbarth, 2015-09-03:
@thinkyhead -- You got the wrong bit.   It's ``` M115 S32```

Wurstnase, 2015-09-15:
I guess it's something with CoreXY. I have the same issues but never spend much time on it. 

Btw. from where you get that copy of Marlin? There is a pneumatic part!?! What is that for?

jminardi, 2015-09-15:
We've added that in ourselves as our 3D printer pneumatically extrudes materials.

I will try to port all our changes to the latest code so I can get detailed logging.

Wurstnase, 2015-09-15:
How does this work? Are there any videos?

jminardi, 2015-09-15:
The only video we have right now is on our website: http://www.voxel8.co/

Wurstnase, 2015-10-02:
Found the problem!
Never use st_get_position_mm() to set the position on CoreXY. This is _not_ the position of the head!

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L1223
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/planner.cpp#L997

jminardi, 2015-10-07:
So what should be used instead of `st_get_position_mm()`?

Wurstnase, 2015-10-08:
No, anything from steps to mm won't work. You can send a M114. The former is the position of your head. The latter the position of your motor. After each move make a M114 and you will see what happens.

Wurstnase, 2015-10-09:
This could work:

replace this part in Marlin_main.cpp
```
        //vector_3 corrected_position = plan_get_position_mm();
        //corrected_position.debug("position before G29");
        vector_3 uncorrected_position = plan_get_position();
        //uncorrected_position.debug("position during G29");
        current_position[X_AXIS] = uncorrected_position.x;
        current_position[Y_AXIS] = uncorrected_position.y;
        current_position[Z_AXIS] = uncorrected_position.z;
        sync_plan_position();
```
 with 

```
        //vector_3 corrected_position = plan_get_position_mm();
        //corrected_position.debug("position before G29");
        vector_3 uncorrected_position = plan_get_position();
        //uncorrected_position.debug("position during G29");
        current_position[X_AXIS] = (uncorrected_position.x + uncorrected_position.y) / 2;
        current_position[Y_AXIS] = (uncorrected_position.x - uncorrected_position.y) / 2;
        current_position[Z_AXIS] = uncorrected_position.z;
        sync_plan_position();
```

acutally untested and needs later some condintionals.

Wurstnase, 2015-10-09:
Just tested... But this won't work. Marlin has a horrible mix of steps for motors and steps for position...

thinkyhead, 2016-04-18:
The latest code over at `RCBugFix` has full CoreXY support in the stepper / planner. That code will be migrated here shortly after the 1.1.0 release. So this issue can be closed.

closed 2016-04-18 by thinkyhead